raspberrypi/linux #1133: Please enable CONFIG_CIFS_DFS_UPCALL

strongly-typed, 2015-09-14:
# Short

When accessing a DFS share via CIFS, I get access errors with "Object is remote" like

	ls: cannot access /mnt/data/DIRECOTRY1: Object is remote

It was not obvious to me what the source of the error is but I was able to resolv the issue by setting the kernel config

	CONFIG_CIFS_DFS_UPCALL=y

Please also activate

	CONFIG_CIFS_ACL=y
	CONFIG_CIFS_SMB2=y
	CONFIG_CIFS_FSCACHE=y
	
as e.g. the Ubuntu 14.04 image does (2015-04-06-ubuntu-trusty.img).

# Long

I am using the official `2015-05-05-raspbian-wheezy.img` image, updated it with 

	apt-get update
	apt-get dist-upgrade -y
	rpi-update
	
to the latest kernel `4.1.6-v7+ #812` on an RPi 2 Model B.

Mounting a DFS via CIFS was possible without error, but accessing the mount point gives strange (at least to me) errors and data access is not possible.

	mkdir /mnt/data
	mount -t cifs -v -o username=USERNAME,domain=DOMAIN //MSDFS/data /mnt/data
	ls -l /mnt/data
	ls: cannot access /mnt/data/DIRECOTRY1: Object is remote
	ls: cannot access /mnt/data/DIRECOTRY2: Object is remote
	ls: cannot access /mnt/data/DIRECOTRY3: Object is remote
	ls: cannot access /mnt/data/DIRECOTRY4: Object is remote
	ls: cannot access /mnt/data/DIRECOTRY5: Object is remote
	
The `srcversion` of the `cifs` module is `EB92C1E5A3CCF10E0244518`.

I was able to solve my problem my activating

	CONFIG_CIFS_DFS_UPCALL=y

in the kernel config, recompiling the kernel modules 
and replacing `cifs.ko` with my own cross-compiled `cifs.ko`.

I was compiling `rpi-4.1.y` branch from `git://github.com/raspberrypi/linux.git` using the instructions from `http://elinux.org/Raspberry_Pi_Kernel_Compilation` and tools from `git://github.com/raspberrypi/tools.git`.

The server is accordingly to `smbclient`: `Windows Server 2008 R2 Standard 7601 Service Pack 1`

With this kernel configuration access to the DFS share is possible without problems. I can provide more testing if necessary. I tested 

	CONFIG_CIFS_DFS_UPCALL + CONFIG_CIFS_ACL + CONFIG_CIFS_SMB2 + CONFIG_CIFS_FSCACHE
	CONFIG_CIFS_DFS_UPCALL + CONFIG_CIFS_ACL + CONFIG_CIFS_SMB2
	CONFIG_CIFS_DFS_UPCALL + CONFIG_CIFS_ACL
	CONFIG_CIFS_DFS_UPCALL

Thanks in advance for your support.

popcornmix, 2015-09-15:
Sounds okay. Added to 4.1.y and 4.2.y trees. Will appear in next rpi-update kernel.

strongly-typed, 2015-09-18:
Works fine with 4.1.y. Well done, thanks. So I close the issue.

    root@debug:~# uname -a
    Linux debug 4.1.7-v7+ #815
    root@debug:~# modprobe configs
    root@debug:~# zcat /proc/config.gz | grep CIFS
    CONFIG_CIFS=m
    # CONFIG_CIFS_STATS is not set
    CONFIG_CIFS_WEAK_PW_HASH=y
    CONFIG_CIFS_UPCALL=y
    CONFIG_CIFS_XATTR=y
    CONFIG_CIFS_POSIX=y
    CONFIG_CIFS_ACL=y
    CONFIG_CIFS_DEBUG=y
    # CONFIG_CIFS_DEBUG2 is not set
    CONFIG_CIFS_DFS_UPCALL=y
    CONFIG_CIFS_SMB2=y
    CONFIG_CIFS_FSCACHE=y

closed 2015-09-18 by strongly-typed