raspberrypi/linux #1136: mkknlimg not compatible with busybox

Ypnose, 2015-09-16:
Good evening,
<code>mkknlimg</code> is not compatible with <code>busybox</code>. <code>grep</code> implementation lacks <code>-b</code> option. So, we are almost unable to mark the kernel, on distro like Alpine Linux, which uses <code>busybox</code> natively for the system tools.

pelwell, 2015-09-17:
Adding --dtok will force it to append the trailer, whatever the input file. It won't include a valid kernel version string, but that doesn't matter.

Ypnose, 2015-09-17:
Why not replace <code>-b</code> option by a "pure" Perl function?

pelwell, 2015-09-17:
If somebody in the Alpine Linux community wants to create a Pull Request I'm sure it will be accepted.

Ypnose, 2015-09-17:
The official <code>APKBUILD</code> for the <code>RPi</code> do not use <code>mkknlimg</code>, so I doubt it's gonna happen. It is not about Alpine Linux, it is about supporting non-coreutils implementations.
I probably do not need to say you busybox is quite common in embedded world. You could simply let it go, but it's definitely a wrong idea…

closed 2015-09-18 by Ypnose