raspberrypi/linux #1137: official usb wifi dongle disconnected frequently

CNMan, 2015-09-20:
i'm using raspberry pi 2 model b + official usb wifi dongle , but network disconnected frequently, i need unplug and replug usb wifi dongle or reboot the system. 

```
pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015 armv7l GNU/Linux
pi@raspberrypi ~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.6-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 8085af80, node_mem_map b9fa6000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9420->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b9f65000 s20608 r8192 d24448 u53248
[    0.000000] pcpu-alloc: s20608 r8192 d24448 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa01041 bcm2709.serial=0x3a914691 smsc95xx.macaddr=B8:27:EB:91:46:91 bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939516K/966656K available (5958K kernel code, 534K rwdata, 1644K rodata, 420K init, 757K bss, 18948K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff000000   (1080 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80774c6c   (7604 kB)
      .init : 0x80775000 - 0x807de000   ( 420 kB)
      .data : 0x807de000 - 0x80863af4   ( 535 kB)
       .bss : 0x80866000 - 0x8092378c   ( 758 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000010] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000032] Switching to timer-based delay loop, resolution 52ns
[    0.000320] Console: colour dummy device 80x30
[    0.002149] console [tty1] enabled
[    0.002214] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.002314] pid_max: default: 32768 minimum: 301
[    0.002684] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002751] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.004069] Initializing cgroup subsys blkio
[    0.004153] Initializing cgroup subsys memory
[    0.004219] Initializing cgroup subsys devices
[    0.004282] Initializing cgroup subsys freezer
[    0.004358] Initializing cgroup subsys net_cls
[    0.004471] CPU: Testing write buffer coherency: ok
[    0.004593] ftrace: allocating 20203 entries in 60 pages
[    0.053928] CPU0: update cpu_capacity 1024
[    0.054012] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.054059] [bcm2709_smp_prepare_cpus] enter
[    0.054224] Setting up static identity map for 0x8240 - 0x8274
[    0.113809] [bcm2709_boot_secondary] cpu:1 started (0) 17
[    0.114115] [bcm2709_secondary_init] enter cpu:1
[    0.114166] CPU1: update cpu_capacity 1024
[... 227 lines cut ...]
[    5.155979] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    5.965531] systemd-journald[107]: Received request to flush runtime journal from PID 1
[    6.674105] cfg80211: Calling CRDA to update world regulatory domain
[    6.709402] usbcore: registered new interface driver brcmfmac
[    6.736190] random: nonblocking pool is initialized
[    6.744765] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    6.753729] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.008836] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Apr  3 2014 04:43:32 version 6.10.198.66 (r467479) FWID 01-32bd010e
[    7.047751] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    7.151741] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
[    7.160009] brcmfmac: brcmf_add_if: ignore IF event
[    9.834941] cfg80211: Calling CRDA to update world regulatory domain
[   11.616455] Adding 1914876k swap on /var/swap.  Priority:-1 extents:26 across:4429820k SSFS
[   12.995088] cfg80211: Calling CRDA to update world regulatory domain
[   16.155050] cfg80211: Calling CRDA to update world regulatory domain
[   19.315159] cfg80211: Calling CRDA to update world regulatory domain
[   22.475125] cfg80211: Calling CRDA to update world regulatory domain
[   25.635141] cfg80211: Calling CRDA to update world regulatory domain
[   28.795177] cfg80211: Calling CRDA to update world regulatory domain
[   31.955266] cfg80211: Calling CRDA to update world regulatory domain
[   35.115264] cfg80211: Calling CRDA to update world regulatory domain
[   38.275318] cfg80211: Calling CRDA to update world regulatory domain
[   41.435335] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA

```

CNMan, 2015-09-20:
and i'm using this bash to detect whether a network can be connected every 15 minutes:

```
#!/bin/bash
export PATH=/usr/bin:/bin:/usr/local/sbin:/usr/sbin:/sbin
ping_count() {
  count=0
  `timeout 5 ping 192.168.0.1 | while read LINE; do
  {
        if [[ "${LINE}" =~ "64 bytes from" ]]; then
                let "count = $count + 1"
                echo "export count=$count"
        fi
  }
  done`
  echo $count
}
if [[ $(ping_count) < 1 ]]; then
        sudo ifconfig wlan0
        sudo ifconfig wlan0 down
        sleep 1
        sudo ifconfig wlan0 up
        sleep 1
        if [[ $(ping_count) < 1 ]]; then
                echo "Fatal error: wifi is down, rebooting now..."
                sudo reboot
        fi
fi
```
/etc/crontab
*/15 * * * *	root	/home/pi/network.sh >> /var/log/network.log 2>&1

network.log show as follows:
```
connect: Network is unreachable
wlan0     Link encap:Ethernet  HWaddr b4:ee:b4:87:3a:d1  
          inet addr:169.254.196.6  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::841a:64f5:157:a49/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2803840 errors:0 dropped:1183 overruns:0 frame:0
          TX packets:1427154 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4140153694 (3.8 GiB)  TX bytes:99419918 (94.8 MiB)

connect: Network is unreachable
Fatal error: wifi is down, rebooting now...
connect: Network is unreachable
wlan0     Link encap:Ethernet  HWaddr b4:ee:b4:87:3a:d1  
          inet addr:169.254.196.6  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::841a:64f5:157:a49/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5703 errors:0 dropped:2478 overruns:0 frame:0
          TX packets:1206 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:537084 (524.4 KiB)  TX bytes:89556 (87.4 KiB)

connect: Network is unreachable
Fatal error: wifi is down, rebooting now...
connect: Network is unreachable
wlan0     Link encap:Ethernet  HWaddr b4:ee:b4:87:3a:d1  
          inet addr:169.254.196.6  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::841a:64f5:157:a49/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1512 errors:0 dropped:499 overruns:0 frame:0
          TX packets:671 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:240020 (234.3 KiB)  TX bytes:69263 (67.6 KiB)

connect: Network is unreachable
Fatal error: wifi is down, rebooting now...
connect: Network is unreachable
wlan0     Link encap:Ethernet  HWaddr b4:ee:b4:87:3a:d1  
          inet addr:169.254.196.6  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::841a:64f5:157:a49/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8528 errors:0 dropped:4292 overruns:0 frame:0
          TX packets:2366 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:958279 (935.8 KiB)  TX bytes:210236 (205.3 KiB)

connect: Network is unreachable
Fatal error: wifi is down, rebooting now...
connect: Network is unreachable
wlan0     Link encap:Ethernet  HWaddr b4:ee:b4:87:3a:d1  
          inet addr:169.254.196.6  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::841a:64f5:157:a49/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8788 errors:0 dropped:520 overruns:0 frame:0
          TX packets:1755 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:721957 (705.0 KiB)  TX bytes:161960 (158.1 KiB)

connect: Network is unreachable
Fatal error: wifi is down, rebooting now...
connect: Network is unreachable
wlan0     Link encap:Ethernet  HWaddr b4:ee:b4:87:3a:d1  
          inet addr:169.254.196.6  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::841a:64f5:157:a49/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2767 errors:0 dropped:406 overruns:0 frame:0
          TX packets:1096 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:256961 (250.9 KiB)  TX bytes:192425 (187.9 KiB)

connect: Network is unreachable
Fatal error: wifi is down, rebooting now...
connect: Network is unreachable
wlan0     Link encap:Ethernet  HWaddr b4:ee:b4:87:3a:d1  
          inet addr:169.254.196.6  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::841a:64f5:157:a49/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:998 errors:0 dropped:468 overruns:0 frame:0
          TX packets:393 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:202668 (197.9 KiB)  TX bytes:70995 (69.3 KiB)

connect: Network is unreachable
Fatal error: wifi is down, rebooting now...
```

but the normal when connected should be:
```
pi@raspberrypi ~ $ ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr b4:ee:b4:87:3a:d1
          inet addr:192.168.0.105  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::b6ee:b4ff:fe87:3ad1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:375 errors:0 dropped:78 overruns:0 frame:0
          TX packets:207 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:47316 (46.2 KiB)  TX bytes:26477 (25.8 KiB)
```

How to solve this problem? THX

pelwell, 2015-09-20:
Whoever you got this script from must have failed to make it clear that the script may need to be modified to suit your network.

192.168.0.1 is an address in a range often used within private networks. Routers know this is a private address, and the attempt to access that address comes from something else in the same network, the router will report that the network is unreachable.

Your Pi has been assigned the address 169.254.196.6. You can find the address of your router using "route -n" and looking for the line that starts 0.0.0.0 that ends with an address like 169.254.x.y - use that address in place of 192.168.0.1. If your script-fu is strong you could even make the script find that address automatically.

CNMan, 2015-09-20:
@pelwell yes, my pi is connected to TP-LINK TL-WR842N,the address need assign to 192.168.0.105, why pi assigned it to 169.254.196.6?

pelwell, 2015-09-20:
Ah, 169.254.x.y is a self-assigned, link-local address - see [here](https://tools.ietf.org/html/rfc3927). Your must have some kind of configuration issue. I'll reopen until we understand it.

Ruffio, 2016-08-17:
@CNMan has your issue been resolved? If so, please close this issue. Thanks.

closed 2016-08-18 by CNMan