raspberrypi/linux #1139: rpisense-fb not allocating `/sys/graphics/fb*` device

mgreter, 2015-09-24:
Hi folks

I'm trying to use my sense hat with a raspberry pi 2 on a custom gentoo linux installation, but I cannot get the led matrix to work. What I got from reading the [python sourcecode] [2], it should allocate a framebuffer device for it, but I fail to get that device "allocated".

I'm using [raspberrypi-sources-4.1.9999.ebuild] [1] (currently at https://github.com/raspberrypi/linux/commit/d2b2388d05d8a97b0ba14fcf2b71f19f66bc4d02) with this config:

```
# cat .config | grep RPI
CONFIG_JOYSTICK_RPISENSE=m
CONFIG_TOUCHSCREEN_RPI_FT5406=m
CONFIG_MFD_RPISENSE_CORE=m
CONFIG_FB_RPISENSE=m
CONFIG_SND_BCM2708_SOC_RPI_DAC=m
CONFIG_SND_BCM2708_SOC_RPI_PROTO=m
CONFIG_LIRC_RPI=m
```

The module seems to load correctly
```
# modprobe rpisense-fb
# modprobe rpisense-js
# lsmod | grep rpi
rpisense_js             2431  0
rpisense_fb             3664  0
rpisense_core           2187  2 rpisense_fb,rpisense_js
fb_sys_fops             1149  1 rpisense_fb
syscopyarea             2949  1 rpisense_fb
sysfillrect             3339  1 rpisense_fb
sysimgblt               1961  1 rpisense_fb
```

I still only have one framebuffer device (I'd expect [`RPI-Sense FB`] [3]):

```
# cat /sys/class/graphics/fb*/name
BCM2708 FB
```

The sense hat is showing the "rainbow" after the raspberry pi is powered up and I can see the sensors and the "rpisense" interface (I guess it's the one at `0x46`):

```
# modprobe i2c-dev
# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- UU 1c -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- 46 -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

My findings for future reference:
```
0x1c Magnetometer LSM9DS1
0x46 rpi-sense (ATtiny88?)
0x5c Pressure LPS25H
0x5f Humidity HTS221
0x6a Accelerometer LSM9DS1
```

I also don't get any messages in `dmesg`? I even tried to hack an early `fb_info` message into `rpisense_fb_probe` of `drivers/video/fbdev/rpisense-fb.c`, but I still don't see anything in `dmesg`!?

Any pointers what I might be doing wrong? I first tried to connect the sense hat only with the necessary pins (since I need to attach additional i2c breakouts), but I have it now directly connected, so that should not be the problem (unless I did some damage I'm not aware of in the experimental process).

I also tried to play with `dtoverlay` in `/boot/config.txt` and I am confused what this option actually does? I know the difference between builtin modules, initrd and "runtime" modules. Do I always need to add `dtoverlay=rpi-sense` or only if it is built into the kernel? I have put `rpi-sense-overlay.dtb` into `/boot` and `/boot/overlays` and the line in `/boot/config.txt`, but still got no framebuffer device? Do I have to compile them myself (I can get avr-gcc built but will probably take a day or two to compile). Or should it be ok if I downloaded it from the firmware repository?

Hope I got the right Bugtracker for this!
Any help is welcome! Thank you!

[1]: http://data.gpo.zugaina.org/open-overlay/sys-kernel/raspberrypi-sources/raspberrypi-sources-4.1.9999.ebuild
[2]: https://github.com/RPi-Distro/python-sense-hat/blob/master/sense_hat/sense_hat.py#L166-L175
[3]: https://github.com/RPi-Distro/python-sense-hat/blob/master/sense_hat/sense_hat.py#L19

XECDesign, 2015-09-24:
Sounds like everything should work. You shouldn't even need the dtoverlay parameter, since that should be loaded automatically from the EEPROM. I suspect you don't have a dt-enabled kernel. I don't see anything in that ebuild that would pass it through mkknlimg and i2cdetect doesn't show 'UU' on 0x46, suggesting that the device isn't instantiated.

mgreter, 2015-09-25:
So any suggestions what I can do to get it "instantiated"? I would guess that i2cdetect should show `UU` once the device is used by a kernel module. The `mkknlimg` sounds interesting. Do I need such a step to create a kernel? I do think to have seen some instructions at some point regarding this.

Actually the ebuild only installs the sources into `/usr/src/` and I only added some additional filesystems (`btrfs`) via `make menuconfig` (it internally runs `make bcmrpi_defconfig` once). I started with some 4.1.17 kernel and updated today to 4.1.18 (commit id as mentioned in the original post), then ran `make olddefconfig`.

I'm also not familiar what you mean by "dt-enabled" kernel. If you can point me to the config entry I will gladly provide that information. Thanks!

XECDesign, 2015-09-25:
Method 4 here https://www.kernel.org/doc/Documentation/i2c/instantiating-devices

Yup, mkknlimg is required for device tree to work properly.

If you're using that defconfig all the necessary kernel options are there, there's just a bit that mkknlimg tags on due to... reasons.

Instead of `cp -a arch/arm/boot/Image /boot/kernel7.img`, try `scripts/mkknlimg arch/arm/boot/Image /boot/kernel7.img`

But I'd point out that bcmrpi_defconfig should be kernel.img (pi 1 kernel), not kernel7.img (pi 2 kernel). The correct defconfig for kernel7.img would bcm2709_defconfig.

mgreter, 2015-09-25:
Thanks! I will try that. I actually saw that error in the ebuild for `bcm2709_defconfig` once, and I thought I had incorporated it into my local ebuild. But I just checked and I still have `bcmrpi_defconfig`, so will rebuild a clean kernel with `bcm2709_defconfig` and apply my changes, then I'll try to use `mkknlimg`. Will report back this weekend! Really hope I can get this working! Thanks again!

mgreter, 2015-09-25:
Just read your link and remembered that command from my ds1307 rtc and actually
```
echo rpisense-fb 0x46 > /sys/class/i2c-adapter/i2c-1/new_device
dmesg: i2c i2c-1: new_device: Instantiated device rpisense-fb at 0x46
```

But I still have no `/sys/class/graphics/fb*` device ...

XECDesign, 2015-09-25:
Try rpisense-core instead of rpisense-fb

mgreter, 2015-09-25:
It seems to give the same results. I do get a new "sys/i2c" directory. I hope I may be able to add some debug statements to the c code, so I can see why it does not load the device now I see something in dmesg.

```bash
ls -la /sys/class/i2c-adapter/i2c-1/1-0046/
total 0
-r--r--r-- 1 root root 4096 Sep 25 03:33 modalias
-r--r--r-- 1 root root 4096 Sep 25 03:33 name
lrwxrwxrwx 1 root root    0 Sep 25 03:33 subsystem -> ../../../../../bus/i2c
-rw-r--r-- 1 root root 4096 Sep 25 03:32 uevent
```

mgreter, 2015-09-25:
Success! Got it working today. Seems I really was only missing `scripts/mkknlimg arch/arm/boot/Image /boot/kernel7.img`! Thanks @XECDesign for your help. 

```
# cat /sys/class/graphics/fb*/name
BCM2708 FB
RPi-Sense FB
```

closed 2015-09-25 by mgreter