raspberrypi/linux #1141: DTS file for Adafruit 13m TFT on Rasperry Pi 2 with Device Tree

kenrestivo, 2015-09-25:
Your call if you want to add it in.

https://github.com/kenrestivo/linux/blob/3d61a22b05cbf7ac2f6b620dcd45c2fb49fb9442/arch/arm/boot/dts/overlays/adafruit13m-overlay.dts

pelwell, 2015-09-25:
This one looks better - you've borrowed from a good source.

1) You can remove the credit about the ds1307 support - you've deleted it all.

2) I couldn't find a 13m display on the Adafruit site - is there another name for it?

3) If the display connects to GPIO pins via flying leads then it would be good to make those pins selectable via DT parameters.

4) All overlays should have an entry in the README file.

Ruffio, 2016-08-17:
@kenrestivo has your issue been resolved? If so, please close this issue. Thanks.

JamesH65, 2017-05-17:
@pelwell Did this ever get added?

pelwell, 2017-05-17:
No - there was no response to my feedback.

Closing.

closed 2017-05-17 by pelwell