raspberrypi/linux #1142: DTS file for 2 rotary controllers attached to Raspberry Pi 2 with Device Tree

kenrestivo, 2015-09-25:
Your call if you want this too:

https://github.com/kenrestivo/linux/blob/b74f2674fd49a1d58baf5dbe5f0fbe052b65f873/arch/arm/boot/dts/overlays/rotaries-overlay.dts

pelwell, 2015-09-25:
1) I don't think you should use attach your pinctrl reference (the pinctrl-names and pinctrl-0 properties) to the gpio node. It is supposed to be in the node that uses them, which in your case would be the two rotary nodes. It also makes more sense to split the pins into two groups, one for each encoder.

2) Why force people to use both channels? If  you set the status of the second channel to "disabled"  you can use a DT parameter to enable it.

3) The overlay would be more useful if people could choose which pins to use. A few more DT parameters would allow this.

There are plenty of examples in the overlays folder of how to use the Pi DT parameter mechanism.

kenrestivo, 2015-09-25:
All good points. This overlay was made for a specific application and served its purpose.

I'll make it more generic, and yes I'm familiar with DT parameters which could be applied here.

Thanks for the tip on pinctrl-* being valid in other nodes. Every example I'd seen had them separated out into a separate fragment; I thought that was required.

closed 2015-09-25 by kenrestivo