raspberrypi/linux #1143: Pulseaudio ALSA errors

kirgene, 2015-09-25:
I have pulseaudio installed and when I start playing audio I sometimes see following error messages in syslog:
```
ALSA woke us up to write new data to the device, but there was actually nothing to write!
Most likely this is a bug in the ALSA driver '(null)'. Please report this issue to the ALSA developers.
We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
```

webattitude, 2016-01-02:
+1
pulseaudio unusable in rtp mode : audio distribution, with usb audio dongle on server and hifiberry on ampli end.
I can get rtp working for a few seconds on a laptop with integrated sound device, till the raspi server show this error. BTW pulseaudio with hifiberry dac is also unable to play received sound, and show this error right away.

Ruffio, 2016-08-17:
@kirgene has your issue been resolved? If so, please close this issue. Thanks.

closed 2017-05-17 by JamesH65