raspberrypi/linux #1146: forgotten Password

HBMBR, 2015-09-28:
Hi - how would i reset a forgotten root password?

popcornmix, 2015-09-28:
You should use the forum for support. But:
``sudo passwd root``
should allow you to set it.

closed 2015-09-28 by popcornmix

HBMBR, 2015-09-28:
No no - after you have installed the system how would you reset the password then after you have set it

popcornmix, 2015-09-28:
Just try it. With "sudo" you can change root password without knowing it.

HBMBR, 2015-09-28:
from which prompt - i cannot log into the System

popcornmix, 2015-09-28:
Do you have the pi password? It will be user=pi password=raspberry if you haven't changed it.

HBMBR, 2015-09-28:
I changed all the passwords - 

on the desktop version you can reset it during the grub boot-loader screen.

popcornmix, 2015-09-28:
There are a few ways of doing this, but create a thread here: https://www.raspberrypi.org/forums/
This is not the right place for this - each message spams a large number of developers who follow the github repo.