raspberrypi/linux #1147: How to support the 7" touch display from raspberrypi org on a custom kernel?

DieterReuter, 2015-09-28:
We do use a custom built kernel 4.1.x for the Raspberry Pi and would like to support the new 7" touch display. Until now I do have the pure display supported but without the touch feature.

I just need a pointer what kernel config is necessary, and/or what DTS overlay file we have to include and maybe how to activate it. 

Thanks for any help or pointers!

popcornmix, 2015-09-28:
Add:
```
CONFIG_INPUT_TOUCHSCREEN=y
CONFIG_TOUCHSCREEN_RPI_FT5406=m
```
and make sure you ship rpi-ft5406-overlay.dtb

procount, 2015-10-14:
Would I be able to add this touchscreen driver to the Kernel used with NOOBS (3.18.7-rescue-v7)?
Would it be easy or a complicated backport (I have no experience of backporting)? Which files do I need to port - just: drivers/input/touchscreen/rpi-ft5406.c & Kconfig?
Or is the preferred method to use a later v4.1.x Kernel for NOOBS?

DieterReuter, 2015-10-14:
Thanks for the answers. I had all the right kernel setting still applied, but was missing to enable the Device Tree support on the kernel with the script ` mkknlimg`. Once this step was included in our build script, the 7" touch screen works just fine.

procount, 2015-10-14:
Hmmm. I'd forgotten DT support - I don't think that is in v3.18. Is this a showstopper, or is there a workaround?

pelwell, 2015-10-14:
3.18 is the first released Pi kernel that supported DT, so that in itself shouldn't be a problem.

procount, 2015-10-14:
OK down the rabbit hole then! :) Is this the correct way?
Step 1 - Enable DT support in buildroot kernel with Boot Options/ Flattended DeviceTree Support
Step 2 - Tag the kernel image as DT aware using mkknlimage
Step 3 - Test kernel DT is working
Step 4 - Add the RPI_FT5406 touchscreen into the kernel
Step 5 - cross fingers & test

pelwell, 2015-10-14:
The mach-bcm2708 support defines CONFIG_BCM2708_DT, which when set enables the options required for a DT-enabled kernel. It's a shorthand for USE_OF + ARCH_REQUIRE_GPIOLIB + PINCTRL + PINCTRL_BCM2835, but it also triggers the building of the DTB files.

Does buildroot not yet you use a defconfig such as bcmrpi_defconfig (Pi 1) or bcm2709_defconfig (Pi 2) instead?

The other steps look correct.

procount, 2015-10-14:
Thanks, I think that was the missing piece I needed. I'll give it a go.

Yes, the defconfig files are present, but NOOBS uses 2 preconfigured 'kernelconfig_recovery.armv6/7' files instead. Contrary to what I read on github in `device-tree.md#part-3-using-device-trees-on-raspberry-pi`, the NOOBS kernel does not appear to be DT enabled yet (probably because there was no need).

pelwell, 2015-10-14:
The device-tree guide isn't saying that the NOOBS recovery image is DT-enabled, but rather that shipping NOOBS images with Raspbian pre-installed now include a DT-capable Raspbian. Sorry for any confusion.

procount, 2015-10-15:
So, I built a clean version of NOOBS. I then set the BCM2709 Device Tree Support (I am building for Pi2) which also set Flattened Device Tree Support for me. I recompiled the kernel.
I booted it and got messages about missing DT for CPU1,2,3 & 4 (or something). 
So I used mkknlimg on recovery7.img. 
I copied the DTBs from a Raspbian Jessie installation with the Overlays folder onto the NOOBS partition.
When I booted, I got the following 3 errors:

of_dma_request_slave_channel: dma-names property of node '/soc/uart@7e201000' missing or empty
mmc-bcm2835 mmc-bcm2835.o: cannot request region
mmc-bcm2835 mmc-bcm2835.o: bcm2835_mmc_probe failed -16

The recovery program started, and then crashed.

What did I miss?

pelwell, 2015-10-16:
I've seen the first error before - we can ignore that for now. The other two don't make much sense to me - they appear to come from sdhci_pltfm_init, but the bcm2835-mmc driver doesn't call into that function (as far as I can tell).

If you are up for a bit of digging, can you edit drivers/mmc/host/sdhci-pltfm.c, locate sdchi_pltfm_init and insert `WARN(1, "here");` at the top, as shown below:
```
	struct resource *iomem;
	int ret;
	WARN(1, "here"); // <=====

	iomem = platform_get_resource(pdev, IORESOURCE_MEM, 0);
```
This should cause a stack backtrace to appear in the kernel log during booting, so we can work out what is calling into that function.

procount, 2015-10-16:
I'm willing to do any digging as I am using NOOBS as a way to learn more about linux development.
I added the WARN line above and recompiled. I can't find any kernel log - /var/log has nothing useful in it. Perhaps logging is disabled in the NOOBS kernel?  Where should it be set? I had a quick look at busybox-menuconfig and syslogd is enabled. klogd is also enabled and set to use the klogctl() interface.

pelwell, 2015-10-16:
I get my logging output from the serial port on pins 6,8&10 (GND,TXD,RXD). Put "console=ttyAMA0,115200 loglevel=8" in the NOOBS cmdline.txt (the one in the first VFAT partition).

procount, 2015-10-16:
Hmm. OK. that will take me a while to setup. Is there an alternative in the meantime?

pelwell, 2015-10-16:
I modified my recovery.cmdline to say:
```
qvt.cur_default=1 elevator=deadline console=ttyAMA0,115200 loglevel=8
```
If can get to a login console (Ctrl-Alt-F2 to switch to a different window), log in using "root" and "raspberry", then "dmesg" will get you the message log.

procount, 2015-10-16:
Doh! I was looking for something more complicated than that!

[    0.194225] ------------[ cut here ]------------
[    0.194281] WARNING: CPU: 0 PID: 1 at arch/arm/mach-bcm2709/armctrl.c:148 armctrl_xlate+0x190/0x27c()
[    0.194331] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 3.18.7-rescue-v7 #6
[    0.194363] Backtrace: 
[    0.194405] [<80011928>] (dump_backtrace) from [<80011ac8>] (show_stack+0x18/0x1c)
[    0.194443]  r6:00000000 r5:85c36008 r4:80432310 r3:00000000
[    0.194503] [<80011ab0>] (show_stack) from [<803352e8>] (dump_stack+0x98/0xd0)
[    0.194550] [<80335250>] (dump_stack) from [<8001d328>] (warn_slowpath_common+0x70/0x8c)
[    0.194588]  r6:00000094 r5:00000009 r4:00000000 r3:00000000
[    0.194646] [<8001d2b8>] (warn_slowpath_common) from [<8001d41c>] (warn_slowpath_null+0x24/0x2c)
[    0.194686]  r8:00000000 r7:86ca02e8 r6:86ca02e8 r5:85c37c68 r4:85c10000
[    0.194754] [<8001d3f8>] (warn_slowpath_null) from [<80018e10>] (armctrl_xlate+0x190/0x27c)
[    0.194808] [<80018c80>] (armctrl_xlate) from [<80058f64>] (irq_create_of_mapping+0x6c/0x118)
[    0.194863] [<80058ef8>] (irq_create_of_mapping) from [<802993a8>] (irq_of_parse_and_map+0x2c/0x34)
[    0.194904]  r5:85d426c0 r4:00000000
[    0.194949] [<8029937c>] (irq_of_parse_and_map) from [<802993d0>] (of_irq_to_resource+0x20/0xc0)
[    0.195002] [<802993b0>] (of_irq_to_resource) from [<80299590>] (of_irq_to_resource_table+0x44/0x54)
[    0.195042]  r8:00000000 r7:00000001 r6:86ca02e8 r5:85d426dc r4:00000000
[    0.195111] [<8029954c>] (of_irq_to_resource_table) from [<80297108>] (of_device_alloc+0xf0/0x164)
[    0.195151]  r7:00000001 r6:85d4c800 r5:86ca02e8 r4:00000000
[    0.195212] [<80297018>] (of_device_alloc) from [<802971cc>] (of_platform_device_create_pdata+0x50/0xa0)
[    0.195253]  r10:85d32c10 r9:00000000 r8:00000000 r7:86ca0338 r6:00000000 r5:85d32c10
[    0.195318]  r4:86ca02e8
[    0.195354] [<8029717c>] (of_platform_device_create_pdata) from [<80297300>] (of_platform_bus_create+0xe4/0x2f4)
[    0.195397]  r8:00000000 r7:803635a0 r6:00000001 r5:00000000 r4:86ca02e8 r3:85d32c10
[    0.195474] [<8029721c>] (of_platform_bus_create) from [<8029735c>] (of_platform_bus_create+0x140/0x2f4)
[    0.195515]  r10:00000000 r9:85d32c10 r8:86ca02e8 r7:803635a0 r6:00000001 r5:00000000
[    0.195579]  r4:86c9c40c
[    0.195616] [<8029721c>] (of_platform_bus_create) from [<80297664>] (of_platform_populate+0x64/0xa8)
[    0.195656]  r10:00000000 r9:00000001 r8:00000000 r7:00000000 r6:803635a0 r5:86c9ad8c
[    0.195720]  r4:86c9c40c
[    0.195755] [<80297600>] (of_platform_populate) from [<803e48a8>] (bcm2709_init+0x6c/0x220)
[    0.195793]  r9:85c36038 r8:803e0870 r7:8043e1c0 r6:8043e624 r5:803ff7b8 r4:8042a100
[    0.195869] [<803e483c>] (bcm2709_init) from [<803e0898>] (customize_machine+0x28/0x48)
[    0.195907]  r6:85d2f040 r5:803ff7b8 r4:803ff7b8 r3:803fba48
[    0.195967] [<803e0870>] (customize_machine) from [<803dde44>] (do_one_initcall+0x120/0x1c4)
[    0.196018] [<803ddd24>] (do_one_initcall) from [<803de02c>] (kernel_init_freeable+0x144/0x1e8)
[    0.196058]  r10:803fffec r9:000000b7 r8:803dd5e8 r7:8043e1c0 r6:8043e1c0 r5:803fffd4
[    0.196122]  r4:00000003
[    0.196159] [<803ddee8>] (kernel_init_freeable) from [<803319b8>] (kernel_init+0x10/0xf8)
[    0.196197]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:803319a8
[    0.196260]  r4:00000000
[    0.196296] [<803319a8>] (kernel_init) from [<8000e898>] (ret_from_fork+0x14/0x3c)
[    0.196333]  r4:00000000 r3:85c36000
[    0.196390] ---[ end trace fa8ecb34b63f021c ]---

pelwell, 2015-10-16:
Interesting. That's not the warning I was expecting, which is likely to be about 2 seconds later in the log. This warning looks to be because the dtb you have copied from the recent Raspbian Jessie image isn't compatible with the kernel. This shouldn't be the case with Device Tree, which should be an OS-neutral description of the hardware, but at the moment the two are evolving hand-in-hand and this kind of problem is likely if you mix and match. It's better to use the `dtbs` kernel make target to build the right DTB files for a particular kernel.

procount, 2015-10-16:
ok I'll give it a try.
In buildroot/output/build/linux-b6.... I tried `make dtbs`. 
What should I answer to this Question:

<pre>
Kernel compression mode
> 1. Gzip (KERNEL_GZIP)
  2. Bzip2 (KERNEL_BZIP2) (NEW)
  3. LZMA (KERNEL_LZMA)
  4. XZ (KERNEL_XZ)
  5. LZO (KERNEL_LZO)
  6. LZ4 (KERNEL_LZ4)
choice[1-6?]: 
</pre>

popcornmix, 2015-10-17:
Always include "ARCH=arm" on kernel make lines, or you will default to x86 config. E.g.
``make ARCH=arm dtbs``

procount, 2015-10-17:
Thanks.
Next problem- where to put the ft5406.dts file as there is no overlays folder in v3.18?
(Or would it just be easier to bump the NOOBS kernel to a version that includes this touchscreen driver already?)

procount, 2015-10-17:
Step 4: backporting from linux-b630db9dcb5c73367050a672823047b0466b16b9
Updated drivers/input/touchscreen/Kconfig to include rpi-ft5406 configuration.
Added   drivers/input/touchscreen/rpi-ft5406.c
Updated drivers/input/touchscreen/Makefile to include rpi-ft5406 target
Added   arch/arm/boot/dts/rpi-ft5406-pverlay.dts (is it supposed to be in dts/overlays?)
Added   'dtb-$(RPI_DT_OVERLAYS) += rpi-ft5406-overlay.dtb' to arch/arm/boot/dts/Makefile
ran ./make_arch_linux-menuconfig.sh armv7 and set
 CONFIG_INPUT_TOUCHSCREEN=y
 CONFIG_TOUCHSCREEN_RPI_FT5406=y
ran ./BUILDME.sh : rpi=ft5406.c fails.
Added   missing include/linux/platform_data/mailbox-bcm2708.h
ran ./BUILDME.sh : successful.

I couldn't find rpi-ft5406-overlay.dtb so,
make ARCH=arm dtbs and got:
<pre>
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
make[1]: `include/generated/mach-types.h' is up to date.
  CC      kernel/bounds.s
gcc: error: unrecognized argument in option ‘-mabi=aapcs-linux’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
gcc: error: unrecognized command line option ‘-mlittle-endian’
gcc: error: unrecognized command line option ‘-mapcs’
gcc: error: unrecognized command line option ‘-mno-sched-prolog’
gcc: error: unrecognized command line option ‘-mno-thumb-interwork’
gcc: error: unrecognized command line option ‘-mfpu=vfp’
make[1]: *** [kernel/bounds.s] Error 1
make: *** [prepare0] Error 2
</pre>
So, I copied the dtb from Jessie again (probably wrong!) and put it in /overlays of SD card.
NOOBS booted fine, no errors, but no touchscreen either.

procount, 2015-10-20:
I guessed that the above errors were because I was using the wrong gcc (x86 on Ubuntu) so I tried:
`make ARCH=arm CROSS_COMPILE="/home/kevin/noobs_dev/buildroot/output/host/usr/bin/arm-buildroot-linux-uclibcgnueabi-" dtbs` and this compiled the dtbs, finally. I wonder if this will work now....

procount, 2015-10-20:
Nope. Still no touchdriver.
rpi-ft5406.c is being compiled and the DTBs have now been produced. mkknlimg used to tag img.
DMESG shows "Machine Model: Raspberry Pi 2 Model B Rev 1.1" but there is no message about "No ATAGs?"
There is a /proc/device-tree but there is no rpi-ft5406 file listed as I expected.
Added dtoverlay=rpi-ft5406 to config.txt but it did nothing.
I don't seem to have vcdbg to debug further.
Any ideas?

pelwell, 2015-10-21:
Copy the vcdbg from a normal image - you could put it in the vfat partition to make things easier - and then use it to look for messages from the firmware.

procount, 2015-10-21:
Yes. Thought of that overnight so tried copying it to both vfat and tmp and all I get is 
"-sh: ./vcdbg: not found."
Tried with & without ./ prefix.
vcdbg is 63583 bytes.
Type vcdbg says
"Vcdbg is a tracked alias for /usr/bin/vcdbg"
But that says "not found" too.
V. Odd.

procount, 2015-10-21:
It works ok in Jessie where I can see the line:
"Loaded overlay 'rpi-ft5406'"
But not in my noobs kernel.

pelwell, 2015-10-21:
That "not found" error can also be caused by missing shared-library dependencies. Running it on Jessie under strace shows it loading the following .so's:

    /usr/lib/arm-linux-gnueabihf/libarmmem.so
    /opt/vc/lib/libdebug_sym.so
    /opt/vc/lib/libvcos.so
    /lib/arm-linux-gnueabihf/libpthread.so.0
    /lib/arm-linux-gnueabihf/libdl.so.2
    /lib/arm-linux-gnueabihf/librt.so.1
    /lib/arm-linux-gnueabihf/libc.so.6

I would expect most of those to be available in NOOBS, but perhaps not libvcos.so and libdebug_sym.so. Make sure that all are somewhere in the library path.

procount, 2015-10-21:
Thanks. I copied them all over from Jessie into the vfat partition in preparation, so I'll try it out tonight & report back.

procount, 2015-10-21:
I don't know how, but I finally got vcdbg to run after putting ld-linux-armhf.so in /lib. Woo-Hoo!, But by this time I was debugging on my lapdock instead of the touchscreen for better visibility. I was even able to run strace on it. Nevertheless, config.txt had dtoverlay=rpi-ft5406 so I hoped it was trying to load the overlay. In the log it mentions "Failed to resolve overlay 'rpi-ft5406;" although it exists as /overlays/rpi-ft5406-overlay.dtb as per Jessie. Is this the right location for v3.18, or should it be in the root?

I'm going to have to reboot using the touchscreen and remember how I got vcdbg working again in order to get any more logs, so that will have to wait til tomorrow. I saved the strace and vcdbg output if it is of any use without the touchscreen - but I'll try and do it again with the touchscreen attached tomorrow.

pelwell, 2015-10-21:
From memory a failure to resolve isn't a failure to load. The overlays live in the "overlays" directory of the VFAT directory, so one level down from bootcode.bin.

It is more likely that there is a symbol absent from the .dtb file that the overlay is referring to. Seeing the vcdbg output would be helpful, but set dtdebug=1 for the next run.

procount, 2015-10-22:
Fortunately, my config.txt was already set as:
```
dtoverlay=rpi-ft5406
dtdebug=1
#device_tree=bcm2709-rpi-2-b.dtb
```
Remember that I didn't have the touchscreen attached, just my lapdock screen, so I removed the beginning of the log that just referred to HDMI
```
001087.406: *** Restart logging

001088.844: Read command line from file 'recovery.cmdline'
quiet vt.cur_default=1 elevator=deadline forcetrigger installvnc ssh

001093.254: Loading 'recovery7.img' from SD card
001222.556: Loading 'recovery.rfs' from SD card
004767.968: Kernel trailer DTOK property says yes
004767.990: Kernel trailer DDTK property says no
004768.253: Loading 'bcm2709-rpi-2-b.dtb' from SD card
004775.054: dtdebug: /aliases:i2c_vc=i2c0
004776.744: dtdebug: /__symbols__:i2c_vc=i2c0
004778.256: dtdebug: /__overrides__:i2c_vc=i2c0
004780.999: dtdebug: /__overrides__:i2c_vc_baudrate=i2c0_baudrate
004782.120: dtdebug: /aliases:i2c=i2c1
004783.828: dtdebug: /__symbols__:i2c=i2c1
004785.406: dtdebug: /__overrides__:i2c=i2c1
004786.578: dtdebug: /aliases:i2c_arm=i2c1
004788.353: dtdebug: /__symbols__:i2c_arm=i2c1
004789.968: dtdebug: /__overrides__:i2c_arm=i2c1
004792.837: dtdebug: /__overrides__:i2c_baudrate=i2c1_baudrate
004795.756: dtdebug: /__overrides__:i2c_arm_baudrate=i2c1_baudrate
004795.773: dtparam: pwr_led_gpio=35
004796.721: dtdebug: Found override pwr_led_gpio
004796.757: dtdebug:   override pwr_led_gpio: cell target gpios @ offset 4 (size 4)
004798.830: dtparam: arm_freq=900000000
004799.704: dtdebug: Found override arm_freq
004799.746: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
004802.355: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
004805.066: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
004807.827: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
004810.654: dtparam: cache_line_size=64
004811.625: dtdebug: /__overrides__ has no cache_line_size property
004811.638: Unknown dtparam 'cache_line_size' - ignored
004846.951: dterror: can't find symbol 'firmware'
004846.964: Failed to resolve overlay 'rpi-ft5406'
004847.158: dtdebug: delete_node(/hat)
004851.729: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
006185.876: vchiq_core: vchiq_init_state: slot_zero = 0xc6400000, is_master = 1
006189.702: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
006193.717: gpioman: gpioman_get_pin_num: pin LEDS_RUNNING not defined
006193.742: gpioman: gpioman_get_pin_num: pin LEDS_NAND_ACTIVITY not defined
006193.764: gpioman: gpioman_get_pin_num: pin LEDS_USB_ACTIVITY not defined
006193.785: gpioman: gpioman_get_pin_num: pin LEDS_FATAL_ERROR not defined
006193.811: gpioman: gpioman_get_pin_num: pin LEDS_APP_OK not defined
006193.835: gpioman: gpioman_get_pin_num: pin LEDS_APP_FAILED not defined
006193.856: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_AUTH not defined
006193.880: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_UNAUTH not defined
006193.903: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_ON not defined
006193.926: gpioman: gpioman_get_pin_num: pin LEDS_DVI_ON not defined
006193.949: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_HPD_UP not defined
006193.970: gpioman: gpioman_get_pin_num: pin LEDS_REMOTE_CONTROL not defined
006193.991: gpioman: gpioman_get_pin_num: pin LEDS_ARM_CONTROLLED not defined
```
So this shows I am missing the `firmware` symbol. 
I am now trying to copy the firmware branch from a later kernel into the soc branch of bcm2708.dtsi. Then dtc it to get the dtb. I assume this will be sufficient?

pelwell, 2015-10-22:
You'll need the firmware driver (rpi-firmware) and the Device Tree files that use it. Commits e6a1ebfa363797e35902c3d466af7c92ef05a62f and 31139cf5452a835f8e160b7093b05dca12c6dd29 should get you the driver, and 0c90bfa500f9c896f42a8cb59f7037b2c78fc8e5 might be enough for the DT support.

procount, 2015-10-22:
I'm beginning to think I should just use a later kernel.... But then I don't know how many other problems that would bring to NOOBS considering the work Lurch had to do to bring it up to v3.18 in the first place. ;-)
Nevertheless, I shall continue down this rabbit hole - thanks for your continued support!

procount, 2015-10-22:
Sorry for noobish question, but I just want to check I will be doing this the right or easiest way.
Buildroot only downloads a zip of the required commit, so I would have to apply the diffs manually to this (ugh). But I have also separately cloned the linux git repos. So I propose to create a new local dev branch from v3.18, then `cherry-pick` the suggested commits onto this new branch. Does that sound ok?

procount, 2015-10-22:
And since this brings in the firmware API, then I should also pick up the following commits for the touchscreen driver::
54fe5cc1635246522e5a2fe44c984e86ad95ecb5, a23d1eb1ed79aa2b4e555dbf5b2d1acfce95600b, 
cff75c6ebfc151717adfc93f6efe7cb2309fc42f ?

pelwell, 2015-10-22:
Yes, your workflow looks good - it's what I would do.

And yes, you should probably include those commits as well.

procount, 2015-10-23:
@pelwell So far I have integrated the following commits:
54fe5cc1635246522e5a2fe44c984e86ad95ecb5 -initial touchscreen driver, a23d1eb1ed79aa2b4e555dbf5b2d1acfce95600b,  - driver fix
e6a1ebfa363797e35902c3d466af7c92ef05a62f - Add RPI firmware driver (pre-req for next)
31139cf5452a835f8e160b7093b05dca12c6dd29 - Support booting without Device Tree
And this all compiles ok, but it's not sufficient for testing yet.
When trying to integrate:
0c90bfa500f9c896f42a8cb59f7037b2c78fc8e5 - Switch to firmware driver
I'm struggling to know how to port bcm2709.c  as this commit brings in partial changes to the bcm2708_dmaengine_device aswell. Do I need to chase the commits to properly port dmaman to dmaengine or can I just comment it out?

procount, 2015-10-23:
Wel,, I commented out the dmaengine for now and it compiled. I'll come back to it if it doesn't work. 1 more to go.

pelwell, 2015-10-26:
You've got impressively far with this, but I can't help thinking that switching to 4.1 might have been easier.

procount, 2015-10-26:
You may be right, but I did have a quick excursion to build the latest kernel (4.3.2 or something - probably too cutting edge) and also the 0c90bfa (?) commit but I didn't get past the rainbow screen. I didn't investigate further (still need to adapt my debug cable). 

My current stage is that I have integrated all the above commits, but only partially for 0c90bfa500f9c896f42a8cb59f7037b2c78fc8e5 . For that I just added the minimal changes to add the rpi_firmware driver and left the other changes to dma & vcio as they were, so I'm hoping this mixed combination works. At least it compiled and built! I shall try it out later this evening. (I'm a kernel newbie so I'm not really sure of the implications of all of these changes)

It if doesn't work, could you suggest a commit ID for a version of 4.1 that *might* work? 

Could you also confirm that all of this device tree stuff is necessary to use the ft5406 driver? The original driver seems to predate the DT, but I maybe wrong on that.

pelwell, 2015-10-26:
The 4.1 kernel is the current shipping RPi kernel, therefore the best known and supported. I would suggest starting with the current HEAD, whatever that is.

The rpi-ft5406 touchscreen driver in that kernel will not work without device tree because it uses the firmware driver. An earlier version (bf09fbb31f813db47e4b64fddbc1ea1d089133c2) uses the old bcm_mailbox driver and would work without Device Tree, but I wouldn't recommend that route.

procount, 2015-10-27:
OK, so I've bumped the kernel to the latest 4.1 HEAD.
I'm building the kernel through the NOOBS Buildroot system that specifies 2 custom kernel config files: kernelconfig-recovery.armv6 and .armv7 (It builds the kernel twice, once for each processor, with a 'make linux-reconfigure' for each)
My 1st step was to just build the kernel with the current noobs config with no device tree just as a sanity check, but it failed with:
<pre>drivers/built-in.o: In function `bcm2835_smi_get':
:(.text+0x470c8): undefined reference to `of_find_device_by_node'</pre>
which suggests to me that some aspects of the kernel assume/require DT support to be included, or do not conditionally check for DT support being included???? Or I'm missing some further config options??

Q1. Can 4.1 be built without DT support, or is it now required?

I guess I should somehow update the noobs buildroot config files for the latest kernel additions.
Q2. Which is the correct defconfig file: bcm2709_defconfig, bcm2835_defconfig or bcmrpi_defconfig? And is there a bcm2708_defconfig for PI1?

I can't use the above defconfig directly because I need to add the changes from noobs (intfamfs etc)
Q3. . How do I update the noobs buildroot kernel config files to add default values for the new kernel options? I guess if I was JUST compiling the kernel I could do `make oldconfig` or something, but how do I do it from within buildroot? make linux-oldconfig? Noobs has its own `make_arch_linux_menuconfig.sh` script to edit the kernel config and I added make linux-update-defconfig to update the kernelconfig-recovery files with any changes I made,

pelwell, 2015-10-27:
A1. The current Raspberry Pi trees assume that Device Tree support is present in the kernel, but they detect at boot time whether or not to use it.

A2. Pi 1 -> `bcmrpi_defconfig`, Pi 2 -> `bcm2709_defconfig`. `bcm2835_defconfig` is a configuration using only upstream code which can be run on a Pi but may have some limitations. However, the gap is closing.

A3. I've not looked at how NOOBS generates its kernel config file, but hang on a moment and I'll find the repo... OK. Buildroot keeps its configurations in configs - one with DT enabled (`raspberrypi_dt_defconfig`) and one without. These buildroot configs include a path to a kernel config, which in the case of the DT kernel is `board/raspberrypi/linux-3.18-rpi-dt.defconfig`. I would suggest comparing that 3.18 defconfig to the regular kernel `bcmrpi_defconfig` from the 3.18 tree to see how it differs. You can then start with 4.1 `bcmrpi_defconfig` and apply the same changes, then generate a new `linux-4.1-rpi-dt.defconfig` using `make ... savedefconfig`.

procount, 2015-10-27:
Thanks a lot for your support.
A3 - Yes normally, but NOOBS uses BUILDME.sh that sets BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE to kernelconfig-recovery.armv6 or armv7 accordingly. I shall compare these config files for similarity and add options manually as I go as you suggest. The NOOBS kernel is quite cut-down, so I only want to add options that are necessary and keep the bloat down.
(I think this will take me a while longer...)

procount, 2015-10-27:
Sorry for the stupid question, but assuming I get ft5406 loaded and working and given that NOOBS is very stripped down, is that all I need to add for the NOOBS QT recovery program to start receiving mouse position and left button presses? I mean, is there another driver that converts the multi-touch events to these mouse events, or do I need to add one?

pelwell, 2015-10-27:
My (limited) understanding is that the calls to `input_report_abs` within the driver present the touch events to the input system - nothing else should be required.

procount, 2015-10-28:
I'm back to my incremental attempt, currently at 31139cf and looking to integrate 0c90bfa500f9c896f42a8cb59f7037b2c78fc8e5.
To get ft5406 to load, I need to add the rpi-firmware driver, which in turn seems to require  bcm2835-mbox.
So I think I additionally need to pull in bcm2835-mailbox.c as the mailbox currently seems to be provided by bcm2708-vcio.

What is the relationship between these 2 drivers? Do I need both?
Can I just upgrade the rpi-firmware to use bcm2835 and leave the rest using vcio?
I thought bcm2835 was for upstreaming and I could ignore it, but this seems like an exception.

pelwell, 2015-10-28:
bcm2708-vcio is the old driver that provides both kernel and user-space interfaces to the VPU mailbox interface. This has been replace by the bcm2835-mbox mailbox driver, the bcm2835/rpi-firmware driver that presents a kernel interface, and the bcm-vcio driver that presents a user-space interface. Both of the latter two use the underlying mailbox driver.

What you are seeing is the increasing use of upstreamed (bcm2835) drivers in the Raspberry Pi kernel. This is a good thing in the long run. So one driver has been replaced by three, but it is progress because two of them conform to standard APIs.

procount, 2015-10-28:
OK, but can the 2 co-exist? Or do I have to update everything to use the new mailbox drivers? (Or could I even downgrade rpi-firmware to use the old vcio driver?)

procount, 2015-10-28:
(A small diversion to avoid these drivers - I know you didn't advise it but...)

At my current commit of 31139cf I have the original ft5406 driver linked in and the kernel is compiled with device-tree enabled. But if I set `config.txt/device_tree=` then DT should be disabled, so there's no DTB looking for non-existent firmware nodes to restrict loading, and that version of the driver doesn't seem to require device-tree.
Once booted I can see rpi-ft5406 in  /sys/bus/platform/drivers so I suppose it is present
But the touchscreen is not working with the NOOBS recovery program.
I suppose there shoudl be a corresponding device (in /bus.platform/devices) but there isn't.
I added a printk message to the ft5406_probe() function and I don't see it in dmesg (I assume that's where it would go) so is there anything else I need to do to initiate this driver in lieu of device-tree?

pelwell, 2015-10-28:
The touchscreen driver is a platform driver. Without a platform device structure, whether from Device Tree or created by the board support code, it won't be probed. For months we have been refusing additions to the board support code, and removing bits that can be replaced with Device Tree except for core functionality. This has meant no touchscreen support without DT - it would be hypocritical to do anything else.

We are now in a position where we could switch over to using the upstream bcm2835 code, which would mark the end of non-DT compatibility. This is why I have been suggesting that you don't go down this route.

procount, 2015-10-28:
Thanks. I'm learning such a lot despite it looking like I'm not getting very far!

pelwell, 2015-10-28:
Careful - you'll find yourself co-opted to the wider Pi kernel team.

procount, 2015-10-28:
Not quite earned my l337 hax0r badge yet ;-)

procount, 2015-10-28:
Woo-hoo! I have a touchscreen cursor in NOOBS !
OK, so it's not DT enabled yet, but at least it works. I feel a lot happier now.
Also, I've got to work out why it's more sensitive than my finger - it seems to move twice as far. Maybe I have to check my screen resolution / scaling.
Then I can move onto the DT side of things again.
Thanks for your guiding hand @pelwell.

pelwell, 2015-10-29:
Nice work! I hope you realise that I'll be expecting a patch set at the end of this.

DieterReuter, 2015-10-29:
@procount great work!

procount, 2015-10-29:
Thanks.
Patch set? - Hehehe - lol!
I wouldn't want to put you in a hypocritical position! ;-)
Maybe if I ever get it working with DT and can get the scaling right I can provide a PR to Noobs.

I don't understand the scaling issue because the screen res already seems to be 800x480 that the driver expects (I don't think it can be anything different). Maybe it is something to do with the QT screen driver in the Noobs recovery program?

procount, 2015-11-03:
Official Touchscreen now working properly in NOOBS! Woo Hoo! 
No device tree support yet, but that should be the easy part, right?
What was that about a patch set? ;-)

procount, 2015-11-05:
@pelwell, @ghollingworth - Before I move onto the device tree version of the rpi-ft5406 touchscreen driver, I want to make sure I have integrated it properly as a platform driver in NOOBS. So are there any specific resources (or dma_mask) I should be setting for it in <pre>static struct resource rpi_ft5406_resources[]   ?</pre>It seems to work without, but not sure if that is correct.

[22 more replies]

closed 2016-08-17 by DieterReuter