raspberrypi/linux #1148: Please enable CONFIG_UHID

matlo, 2015-10-01:
Hello,
Please consider adding the uhid module in the next update.

pelwell, 2015-10-01:
Can you say what it does and why it would be useful?

matlo, 2015-10-01:
It allows connecting a userspace-managed device to the HID core driver.
I need it for the GIMX project.

pelwell, 2015-10-01:
I've got no objections - the option just causes a small, simple, self-contained module to be built. @popcornmix ?

popcornmix, 2015-10-01:
Sounds okay - will add.

popcornmix, 2015-10-02:
This should be in latest rpi-update firmware.

matlo, 2015-10-02:
I got the uhid module in the update, and tested it successfully.
Thanks. This is great for the GIMX users.

closed 2015-10-02 by matlo

popcornmix, 2015-10-02:
No problem. GIMX looks interesting. No (new enough) games console here, but it's an interesting use of a Pi.