raspberrypi/linux #1149: cfg80211: Calling CRDA to update world regulatory domain

tomdean1, 2015-10-01:
> lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release: 8.0
Codename: jessie
> uname -a
Linux raspberrypi 4.1.7+ #817 PREEMPT Sat Sep 19 15:25:36 BST 2015 armv6l GNU/Linux
> sudo ifconfig
eth0      Link encap:Ethernet  HWaddr b8:27:eb:00:be:8f  
          inet addr:192.168.2.12  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fe00:be8f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5167 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3962 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:671736 (655.9 KiB)  TX bytes:837105 (817.4 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
> sudo iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

I do not have a wireless adapter.  I use a cable connection.

CFG80211 is marked unstable on Debian.

I get frequent messages
cfg80211: Calling CRDA to update world regulatory domain

On boot, this takes considerable time.

popcornmix, 2015-10-02:
Is this a new problem?
Did this occur after updating to Jessie?

giddyhup, 2015-10-02:
I also get this on one of my (wheezy) Pis ( 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015). I do use wifi, though. wpa_supplicant.conf contains a line to set the country (country=XX).

emk2203, 2015-10-08:
Same problem here, on a RPi 2 after the update to Jessie. Runs with the latest rpi-update kernel. No wifi attached.

popcornmix, 2015-10-08:
Can you all post contents of ``/etc/network/interfaces``

giddyhup, 2015-10-08:
auto lo
    iface lo inet loopback
    
    auto eth0
    allow-hotplug eth0
    iface eth0 inet manual
    
    auto wlan0
    allow-hotplug wlan0
    iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

and the header from /etc/wpa_supplicant/wpa_supplicant.conf
 
    ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
    update_config=1
    country=XX

where XX is my country

tomdean1, 2015-10-10:
I have upgraded to Jessie on an RPi 2.  I can boot the original RPi, if 
necessary.

 > cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

 > cat /etc/network/interfaces
# Please note that this file is written to be used with dhcpcd.
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'.

auto lo
iface lo inet loopback

auto eth0
allow-hotplug eth0
iface eth0 inet manual

auto wlan0
allow-hotplug wlan0
iface wlan0 inet manual
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

auto wlan1
allow-hotplug wlan1
iface wlan1 inet manual
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

 > sudo cat /etc/wpa_supplicant/wpa_supplicant.conf
[sudo] password for tomdean:
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

 >  dmesg | grep CRDA
[   22.293941] cfg80211: Calling CRDA to update world regulatory domain
[   25.453216] cfg80211: Calling CRDA to update world regulatory domain
[   28.613183] cfg80211: Calling CRDA to update world regulatory domain
[   31.773183] cfg80211: Calling CRDA to update world regulatory domain
[   34.933181] cfg80211: Calling CRDA to update world regulatory domain
[   38.093187] cfg80211: Calling CRDA to update world regulatory domain
[   41.253188] cfg80211: Calling CRDA to update world regulatory domain
[   44.413184] cfg80211: Calling CRDA to update world regulatory domain
[   47.573225] cfg80211: Calling CRDA to update world regulatory domain
[   50.733180] cfg80211: Calling CRDA to update world regulatory domain
[   53.893225] cfg80211: Calling CRDA to update world regulatory domain
[   57.053189] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA

Tom Dean

dgordon42, 2015-10-15:
It seems that Jessie loads a kernel module, `cfg80211`, when booting, and it tries to get `crda` to update it's world regulatory database. However, `crda` is not installed by the published Jessie image.
`sudo apt-get install crda`
will install `crda`, as well as the `iw`, and `wireless-regdb` packages.
Reboot, and you will see `cfg80211` update it's database in dmesg, no more error messages, and boot-up seems a little faster!

I think `crda`, `iw`, and `wireless-regdb` should be included in the published Jessie image.

Thanks.

tomdean1, 2015-10-15:
On 10/15/15 11:33, dgordon42 wrote:

>
> I think |crda|, |iw|, and |wireless-regdb| should be included in the
> published Jessie image.

My system is often booted without a net connection.  I need the 
flexibility to either have or not have a net connection.

Tom Dean

dgordon42, 2015-10-17:
Just booted my Pi without an internet connection. The `crda`, `iw` and `wireless-regdb` packages were installed as detailed above, and it booted normally, with no error messages from `cfg80211` or `crda`.

I think Jessie needs these packages, and should come with them installed. 

I don't have wired internet here to test, only wireless.

Dave.

emk2203, 2015-10-18:
crda and the others are not installed. Contents of /etc/network/interfaces:

pi@EMK-RPi2 ~ $ cat /etc/network/interfaces
auto lo

iface lo inet loopback
iface eth0 inet dhcp

allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp
pi@EMK-RPi2 ~ $ dpkg -l crda
dpkg-query: no packages found matching crda
pi@EMK-RPi2 ~ $ apt-cache search crda
crda - wireless Central Regulatory Domain Agent
wireless-regdb - wireless regulatory database
pi@EMK-RPi2 ~ $ dmesg | grep crda
<nothing found>

16:40 Uhr:

dgordon42, 2015-10-18:
@emk2203, Are you running Wheezy or Jessie?
Post the output of `cat /etc/os-release`
Thanks,
Dave.

giddyhup, 2015-10-18:
After installing crda the regulatory domain can now be set (I do it via .`iw reg set XX &` in /etc/rc.local).

tomdean1, 2015-10-18:
> cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

If I remove wlan from /etc/network/interfaces I do not see CRDA in dmesg.

Tom Dean

emk2203, 2015-10-19:
@tomdean1: Jessie at the time of posting (updated to stretch now).

20:01 Uhr:

Ruffio, 2016-07-27:
X-ref: https://github.com/raspberrypi/firmware/issues/630

Ruffio, 2016-08-17:
@tomdean1 has your issue been resolved? If so, please close this issue. Thanks.

JamesH65, 2017-05-17:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

Note, there have been changes to the network drivers that may help. Update to the latest kernel to get them.

closed 2017-05-17 by JamesH65