raspberrypi/linux #1150: bcm2708-dmaengine and bcm2835-dma

notro, 2015-10-01:
I need a break from a long running project, so I have started to look at the possibility for us to move to bcm2835-dma.

I see 2 possibilities:
1. Move the legacy DMA API code to bcm2708_fb since it's the only user.
2. Export bcm_dmaman_probe() and bcm_dmaman_remove() from bcm2708-dmaengine so bcm2835-dma also can provide the legacy API.

In addition bcm2835-dma will need scatter/gather support, for which I have a patch that never received attention from upstream maintainers.

I prefer number one, because that means we can drop bcm2708-dmaengine.
What do you think?

popcornmix, 2015-10-01:
I think either would be useful, so I'm happy for you to go with your favoured one.
If you have some time there is a 4.3 tree pushed.

pi1 support looks good, but pi2 has a reliability issue (panics in memcpy sometimes after running for a while). There is now an upstream pi2 interrupt controller that we could probably use (and possibly some other upstreamed bits).

pelwell, 2015-10-01:
I prefer 1 as well, because it means we don't have to pollute the pristine upstream code with our horrible downstream code. ;-)

notro, 2015-10-01:
Oh well I didn't think this one through, looking at the legacy code it needs reg/irq from the dma device.
So option 2 it is. I plan to use bcm2835-dma with Device Tree and bcm2708-dmaengine for ATAGS booting, since I keep the legacy API in bcm2708-dmaengine. So bcm2835-dma just have to call bcm_dmaman_probe() to enable the API.

@pelwell Hopefully this won't be too horrible even though it's coming from the bikeshed :laughing:

closed 2015-12-09 by notro

pucgenie, 2021-02-16:
So that's why compiling a kernel with compiled-in bcm2835-dma and modularized bcm2708-dmaengine fails to compile. Just tested:
`DMA_BCM2835 [=y]`
`DMA_BCM2708 [=m]`
...
`  CC [M]  net/sunrpc/auth_gss/gss_rpc_xdr.o
  CC [M]  net/sunrpc/auth_gss/trace.o
  LD [M]  net/sunrpc/auth_gss/auth_rpcgss.o
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.kallsyms1
ld: Unexpected GOT/PLT entries detected!
ld: Unexpected run-time procedure linkages detected!
ld: drivers/dma/bcm2835-dma.o: in function `bcm2835_dma_remove':
bcm2835-dma.c:(.text+0x150): undefined reference to `bcm_dmaman_remove'
make[3]: *** [Makefile:1170: vmlinux] Error 1
make[2]: *** [debian/rules:7: build-arch] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
make[1]: *** [scripts/Makefile.package:83: bindeb-pkg] Error 2
make: *** [Makefile:1536: bindeb-pkg] Error 2`

And then:
`DMA_BCM2835 [=y]`
`DMA_BCM2708 [=y]`
...
`  HDRINST usr/include/asm/sembuf.h
  HDRINST usr/include/asm/ioctls.h
  INSTALL debian/linux-libc-dev/usr/include
dpkg-deb: building package 'linux-libc-dev' in '../linux-libc-dev_5.10.14-v8bierpi+-5_arm64.deb'.
dpkg-deb: building package 'linux-image-5.10.14-v8bierpi+' in '../linux-image-5.10.14-v8bierpi+_5.10.14-v8bierpi+-5_arm64.deb'.
 dpkg-genbuildinfo --build=binary
 dpkg-genchanges --build=binary >../linux-5.10.14-v8bierpi+_5.10.14-v8bierpi+-5_arm64.changes
dpkg-genchanges: info: binary-only upload (no source code included)
 dpkg-source --after-build .
dpkg-buildpackage: info: binary-only upload (no source included)`

I believe fixing this by moving the code to a new common module would be too much hassle for almost no gain, however I might try to create a pull request for that...