raspberrypi/linux #1151: Config options for supporting openlabs 802.15.4 radio?

katiska, 2015-10-02:
Would it be possible to include the config options required by the [openlabs 802.15.4 radio](http://openlabs.co/OSHW/Raspberry-Pi-802.15.4-radio)?

This allows the RasPi to be used as a 6LoWPAN border router using [linux-wpan](http://wpan.cakelab.org/).

The required kernel options and .dts: [(source)](http://openlabs.co/blog/archives/1-6LoWPAN-kernel-on-a-Raspberry-Pi)

    CONFIG_IEEE802154
    CONFIG_IEEE802154_6LOWPAN
    CONFIG_MAC802154
    CONFIG_IEEE802154_DRIVERS
    CONFIG_IEEE802154_AT86RF230
    
    &spi {
        status = "okay";
        at86rf233@0 {
            compatible = "atmel,at86rf233";
            reg = <0>;
            interrupts = <23 1>;
            interrupt-parent = <&gpio>;
            reset-gpio = <&gpio 24 1>;
            sleep-tpio = <&gpio 25 1>;
            spi-max-frequency = <500000>;
        };
    };

popcornmix, 2015-10-04:
I've added the config options to latest rpi-update.
@pelwell how do you want to do the overlay?

pelwell, 2015-10-04:
I'll write one tomorrow and attach the .dtb for testing.

pelwell, 2015-10-05:
A .tgz containing the source and .dtb file for the overlay can be found [here](https://drive.google.com/file/d/0B8VsfKAD4-NOXzNrSVZTXzNvQU0/view?usp=sharing). The extract from the README say:
```
Name:   at86rf233
Info:   Configures the Atmel AT86RF233 802.15.4 low-power WPAN transceiver,
        connected to spi0.0
Load:   dtoverlay=at86rf233,<param>=<val>
Params: interrupt                GPIO used for INT (default 23)
        reset                    GPIO used for Reset (default 24)
        sleep                    GPIO used for Sleep (default 25)
        speed                    SPI bus speed in Hz (default 750000)
        trim                     Fine tuning of the internal capacitance
                                 arrays (0=+0pF, 15=+4.5pF, default 15)
```
There have been some slight changes - SPI clock speed now 750KHz, the use of a level-sensitive interrupt, and the addition of the trim parameter - based on information found online.

Unless I hear of a problem this will appear in the next release.

katiska, 2015-10-05:
That's great! Thank you for the quick response.

Would you, @alexaring (linux-wpan) or @benemorius (openlabs), have any comments on this?

pelwell, 2015-10-06:
I've pushed the commit containing the overlay. Let me know if you experience any problems, and close  the issue if it works.

alexaring, 2015-10-10:
Hi,
sorry for the delay. I had the experience that some of the openlabs module can't run at the highest 7.5 Mhz clockrate. I set it to 6 Mhz and then it's working fine again, otherwise I had some incorrect data on the spi bus.

Otherwise it's looking good. Thanks.

Please refer: http://lists.infradead.org/pipermail/linux-rpi-kernel/2015-May/001685.html

pelwell, 2015-10-12:
Thanks for the feedback, @alexaring - I've reduced the default clock frequency to 6MHz,

@katiska OK to close?

katiska, 2015-10-12:
I'm still hitting that data corruption issue at 6 MHz. In my case it seems that the frequency needs to be lowered down to 4 MHz to make it work properly.

alexaring, 2015-10-12:
Okay, make it so. I am still fine with 4 Mhz to be sure _all_ openlabs module will working properly.

Did you test it with 4 Mhz and everything was working fine then?

katiska, 2015-10-12:
Well, I haven't really managed to set up a proper WSN system yet. Just by looking at the periodic RPL DIS packets sent by a CC2650 SensorTag (Contiki), Wireshark showed a lot of corrupt packets at 6 MHz and 5 MHz. At 4 Mhz all the packets I saw were OK. I need to do more testing with proper configuration and traffic.

njarvis, 2015-10-16:
On Raspberry PI 2 with the openlabs 802.15.4 radio, I had to reduce the speed to 2MHz.

In my /boot/config.txt:

dtoverlay=at86rf233,speed=2000000

pelwell, 2015-10-16:
Any advance on 2MHz? It would be nice to establish a safe maximum.

njarvis, 2015-10-16:
OK, did a further binary chop, and it is also stable at 3Mhz.  Definitely does not work for me at 4MHz (I retested).  

My setup is three Raspberry PI 2s; one radio in monitor and the other two running 6lowpan and showing IPv6 traffic successfully passing.

alexaring, 2015-10-16:
I am fine with 3 Mhz if this seems to be working correct as a maximum safe value.
This is already under the half of possible maximum clock rate. We should not go lower than this value.

pelwell, 2016-02-19:
OK to close?

njarvis, 2016-02-19:
Fine by me.

closed 2016-02-19 by pelwell