raspberrypi/linux #1154: bcm2835-gpiomem doesn't build on ARCH_BCM2835

notro, 2015-10-04:
```
drivers/char/broadcom/bcm2835-gpiomem.c: In function ‘bcm2835_gpiomem_probe’:
drivers/char/broadcom/bcm2835-gpiomem.c:201:26: error: ‘GPIO_BASE’ undeclared (first use in this function)
   inst->gpio_regs_phys = GPIO_BASE;
```
Maybe non-DT support just can be dropped?
```c
	if (node) {
		ioresource = platform_get_resource(pdev, IORESOURCE_MEM, 0);
		inst->gpio_regs_phys = ioresource->start;
	} else {
		inst->gpio_regs_phys = GPIO_BASE;
	}
```

pelwell, 2015-10-06:
Yes, with hindsight it was a mistake to try to make this driver non-DT compatible. I have a patch to remove the non-DT compatibility, and to properly handle the missing resource.

pelwell, 2015-10-06:
Fix for rpi-4.1.y and rpi-4.2.y.

closed 2015-10-06 by notro