raspberrypi/linux #1156: spi-bcm2835 driver doesn't handle CS status properly

rlbartle, 2015-10-05:
Normally at the end of an spi_message the CS status should be set to off, unless the cs_change of the last spi_transfer is set.  This doesn't happen, it just always stays on.

Also, at the end of an spi_transfer the cs_change needs to be checked and the CS status set accordingly.

Here is a link to the changes I made to support my driver: https://github.com/rlbartle/linux/commit/d66f45a4f2f56dfdfef9a210ba5b9b3d78260988
These changes cause CS to go off after every spi_message, it should check the cs_change of the last spi_transfer to make sure it is desired.  Also, only polling spi_transfers check cs_change, whereas interrupt driven and DMA driven transfers should do so as well.

Here is my [driver](https://github.com/rlbartle/axpi-fpga)

popcornmix, 2015-10-05:
ping @msperl

msperl, 2015-10-06:
No, that assumption about CS_change is wrong - it is all about allowing a cs_change in the middle of a spi-message (essentially de- asserting CS for a short period before continuing the next transfer of the spi message)

When a spi-message is finished the Chipselects will always de-assert, as you would otherwise get confusing situations when you got 2 distinct devices on the bus driven by different drivers - with that API-change you propose you would:
A) get data sent to both devices when spi-messages were interleaved
B) such a change would need to go into the spi-framework not an individual driver that lets the framework handle the Chipselects management.
C) you would need to modify ALL drivers (in and outside of the kernel) that use spi, so that they set cs_change for the last transfer, as this is not a requirement right now

I guess you need to rethink your design so that the whole transfer is done as one and not several spi_messages.

The only solution for what you want to do would be grabbing the whole bus via spi_lock/spi_unlock and then handle chip-selects in your own driver...

Note that the use of is_dma_mapped is depreciated, even if it is not officially communicated by the spi maintainer (in the source-code, except for in one email)

msperl, 2015-10-06:
See the details of the framework in drivers/spi/spi.c and there the framework implementation in spi_transfer_one_message

msperl, 2015-10-06:
Actually: rereading that code in spi.c: if you set cs_change on the last spi_transfer of a spi_message, then it does not get asserted!
So that is what you need to fix, as you set it on every transfer...

rlbartle, 2015-10-06:
You misunderstand my code.  There are a sequence of spi_transfers in my message which need to deactivate the chip select.  That is why the cs_change is used, but the master driver/framework didn't seem to do respond at all to any permutation of cs_change, whether on intermediate transfers or the last transfer.  The last transfer in all messages I send does not have cs_change set, so I expect the CS to deactivate but it doesn't.
That is why I have added these hacks in this driver to trigger it for the cases I need.  I am aware it is not implemented appropriately, the code is simply there to demonstrate the requirement and shortfall that I came across.  I'm not sure how I would implement the functionality using the framework when it doesn't seem to respond to cs_change at all.

msperl, 2015-10-06:
It is only this driver of yours that requires this hack, so there must be something in your driver.

Is it possible to test your code without having the corresponding hardware to see what happens with my logic analyzer?

If so, then how can I trigger such transfers?

msperl, 2015-10-06:
As a note while looking at your code more in depth

Why do you run:
list_add_tail(&t->transfer_list, &t2->transfer_list);

When you really should be using spi_message_add_tail.

The way I read it you essentially insert t in front of t2 in the list of spi transfers (assuming that t2 is the only one item in the list - so there are 2 items in the linked list: spi-message transfers and t2. So with the add of t1 at the end of the list relative to the assumed list-head t2  and not spi-message, you insert between spi message list-head and t2), which introduces issues and reorders the list of transfers, which makes the code not very readable.

On top you set cs_change during initialization phase only which will result in possibly unexpected behavior in conjunction with the above list

I would recommend that you dump your spi message before your call spi_async and see that it is what you expect it to be...

Something like:
Dev_err(&spi->dev,"spi message start\n");
List_for_each_entry(t, m->transfers,transfer_list) {
  Dev_err(&spi->dev, "msg: %pK %i\n", t, t->cs_change)
}
Dev_err(&spi->dev,"spi message end\n");

If cs_change is set for the last then that is your issue!
Please also check on the expected sequence of transfers from a pointer perspective - I guess they are not in the expected order when it prints out, resulting in cs_change being set for the last transfer.

Finally if the issue persists, then please create a simple sample code with 2 transfers that are set up in the direct code path that also shows the issue.

rlbartle, 2015-10-06:
In the test directory in my repository is some userspace code to interface with the driver.

Thanks for your tips on my driver code - I did find the list api rather confusing, so no surprise I have implemented it a rough/unorderly fashion

rlbartle, 2015-10-06:
Just tested and cs_change is 0 for the last transfer of each message, and 1 for every other transfer.  That's what I expected.

msperl, 2015-10-07:
Avoiding the complexity of your code I have created a simple test driver that triggers 4 spi_messages:
* with 1 transfer
* with 2 transfers, first with cs_change=1
* with 3 transfers, first and second with cs_change=1
* with 3 transfers, first, second and third with cs_change=1 (so CS should stay asserted at the end of the spi_message)

```
#include <linux/delay.h>
#include <linux/kernel.h>
#include <linux/module.h>
#include <linux/spi/spi.h>

#define TRANSFERS 3
static int test_probe(struct spi_device *spi)
{
	struct spi_message m;
	struct spi_transfer t[TRANSFERS];
	int i;

	/* clear structures */
	memset(&m, 0, sizeof(m));
	memset(&t, 0, sizeof(t));

	/* now set up transfer pointers */
	for(i = 0; i < TRANSFERS ; i++) {
		t[i].tx_buf = &t[i];
		t[i].len = sizeof(struct spi_transfer);
	}

	/* run with a single transfer first */
	spi_message_init(&m);
	spi_message_add_tail(&t[0],&m);
	spi_sync(spi,&m);

	udelay(100);

	/* run with two transfers */
	t[0].cs_change = 1;
	spi_message_add_tail(&t[1],&m);
	spi_sync(spi,&m);

	udelay(100);

	/* run with three transfers */
	t[1].cs_change = 1;
	spi_message_add_tail(&t[2],&m);
	spi_sync(spi,&m);

	udelay(100);

	/* run with three transfers plus cs_change set on the last
	 * this will keep cs asserted as it is intended to work...
	 */
	t[2].cs_change = 1;
	spi_sync(spi,&m);

	return 0;
}

static const struct of_device_id test_of_match[] = {
	{
		/* you may want to change this string to match your device-tree */
		.compatible	= "shiji-led,apa102",
	},
	{ }
};
MODULE_DEVICE_TABLE(of, test_of_match);

static struct spi_driver test_driver = {
	.driver = {
		.name = "test driver",
		.owner = THIS_MODULE,
		.of_match_table = test_of_match,
	},
	.probe = test_probe,
};
module_spi_driver(test_driver);

MODULE_AUTHOR("Martin Sperl <kernel@martin.sperl.org>");
MODULE_DESCRIPTION("test driver via SPI");
MODULE_LICENSE("GPL");
```
And this produces the following:
[image]
First row is SCK
Second row is MOSI
Third row is CS

And you see the first 3 spi-messages all pull cs high at the end and have another cs high in between transfers as well.

Only the last message with cs_change=1 for the last transfer shows that cs stays low after the transfer, but this is expected and per design.

So if your behavior is different, then I guess it is the code of yours - if you think otherwise please try to produce it based on the above example...

Note that the example shows how one would typically write an spi driver from an initialization perspective (including mapping to the device-tree) - you may want to adopt that!
Registering the char device should probably use dynamic allocation of Major/minor numbers (0,0) and would get done inside the test_probe function.

rlbartle, 2015-10-07:
Ok you've convinced me, it must be a coding issue on mine.

closed 2015-10-07 by rlbartle