raspberrypi/linux #1158: ALSA: large pcm delay jitter

wm4, 2015-10-09:
I'm seeing jitter as large as 10ms in the snd_pcm_delay function. I'm quite confused as to what it's coming from. It doesn't seem to be caused by audio underruns or too high CPU usage. Is it known whether the ALSA driver handles this function correctly?

Here's a test program: http://sprunge.us/gGGj

popcornmix, 2015-10-12:
The alsa driver works by calling into the gpu.
Messages like open, write, close are sent to gpu.
The gpu sends back periodic messages to signal how many samples have been played out. These messages are sent every 10ms (and cause the snd_pcm_period_elapsed function to update).

This does mean your test code will jump every 10ms. Does this cause a problem? The architecture of the alsa driver remoting requests to the GPU means means very low latency audio is not possible. Assuming latency is set to something like 100ms, does the 10ms of jitter cause any problem?

wm4, 2015-10-13:
Looked at the driver code again. So it seems that bcm2835_playback_fifo_irq indeed advances the playback position once every time the GPU gets a VC_AUDIO_MSG_TYPE_WRITE message, and the message comes only all 10ms. There's nothing else going on, so for 10ms, the playback state is not updated at all.

Yes, this is kind of a problem, because it messes with A/V sync in video playback. 10ms deviation is just a bit much.

I've made a fix here: https://github.com/wm4/linux/commit/856436659ef01cda29ba5d06e559bec025f219d6

I have no idea what I'm doing, but, well, it gives me the expected results. I tried to look at how other hardware deals with this. But they all seem to be able to query the playback state somehow, usually by reading a register. I haven't found a single driver yet which interpolates the position manually using a timer. There doesn't seem to be generic code for interpolation either.

Does the OMX audio output API (as used by kodi) also have this problem?

popcornmix, 2015-10-14:
OMX doesn't have this issue. It directly queries the GPU when it needs the delay value (and gets back a sample accurate result, although with a small latency delay for the IPC comms). That's obviously a better scheme  than GPU sending unsolicited updates, and arm asynchronously polling it.

Thanks for interpolation code - was talking to @pelwell yesterday about whether that was a sensible approach. Can the code path that interpolates be called when audio is paused? I'm not sure if you will need to disable it when audio isn't playing normally.

I'll test your PR again, and see if I spot any regressions.

wm4, 2015-10-14:
Would it be better to use OMX in the kernel? I'm also pondering whether implementing this as a userspace ALSA plugin wouldn't be a better idea.

>Can the code path that interpolates be called when audio is paused?

From what I know, the driver does not support pausing at all, so it can't be a problem.

popcornmix, 2015-10-14:
start and stop can be called multiple times in a single stream and they are effectively resume and pause playback:
https://github.com/raspberrypi/linux/blob/rpi-4.1.y/sound/arm/bcm2835-vchiq.c#L113
https://github.com/raspberrypi/linux/blob/rpi-4.1.y/sound/arm/bcm2835-vchiq.c#L134

I don't think we need OMX in kernel driver. We could add a poll message to audioserver and remove the "complete" callbacks. Just working out where the poll call needs to go? Just in snd_bcm2835_pcm_pointer? Does snd_pcm_period_elapsed still need to be called?

wm4, 2015-10-14:
>Just working out where the poll call needs to go? Just in snd_bcm2835_pcm_pointer? Does snd_pcm_period_elapsed still need to be called?

Very unsure. It seems a driver can just return a value without doing anything else, but snd_pcm_period_elapsed should probably still be called somewhere. Can take a closer look later.

popcornmix, 2015-10-14:
If you can find a way of plumbing it in, this test firmware:
https://dl.dropboxusercontent.com/u/3669512/temp/firmware_latency.zip

adds a new audio_server function:
http://pastebin.com/s4AdDK0m

which returns (in VC_AUDIO_RESULT_T success field) the latency in samples of the audio sink.
It might be worth removing the audio complete callbacks handling and using this latency directly.

wm4, 2015-10-15:
Thanks a lot. This looks ideal. What I worry about is that this would have to send a message to the GPU and wait for the reply (if I understand this right), and I'm not fully sure if it will be possible to do this wait in this context. But maybe it's no issue.

Anyway, I think I don't have time to test this any time soon, since the current less elegant fix works for me.

popcornmix, 2015-10-15:
Yes, any communication with GPU will need a wait for response, and so blocking behaviour, so can't be called from atomic/interrupt context. Should be okay from calls from userland alsa library, but I'm not too sure what context different alsa driver functions are called from. Not very well documented...

wm4, 2015-10-15:
What I'm mostly worried about is that ``snd_pcm_delay()`` itself wraps all further calls inside of a ``snd_pcm_stream_lock_irq(substream);``. This call at acquires a lock of some kind, and depending on other things, disable IRQs on the current CPU. Might cause trouble.

popcornmix, 2015-10-15:
Not sure if we can set substream->pcm->nonatomic. Sounds like that might be valid if we remove the complete callbacks.

wm4, 2015-10-19:
Just another thought: could the firmware write the buffer state to a fixed shared memory location? This would imply that atomic memory accesses between GPU and CPU work.

Ruffio, 2016-08-17:
@wm4 has your issue been resolved? If so, please close this issue. Thanks.

JamesH65, 2017-05-17:
@pelwell @popcornmix @wm4 Is there likely to be anything else happening on this?

pelwell, 2017-05-17:
No.

closed 2017-05-17 by pelwell

wm4, 2017-05-17:
Would still be nice if this stuff could somehow be upstreamed, if it isn't already (I have no overview over this anymore). At Plex we're probably still using large patches to get useful ALSA support.

popcornmix, 2017-05-17:
Actually audio driver is upstreamed (to staging):
https://github.com/raspberrypi/linux/tree/rpi-4.12.y/drivers/staging/vc04_services/bcm2835-audio