raspberrypi/linux #1159: USB-Harddrive gets fully powered down

lduesing, 2015-10-09:
in 4.1.10 my USB connected harddrive gets fully powered down from time to time, only a reset of the raspberry helps. Due to being there my OS my raspberry freezes complete. Interestingly only the HDD connected via "JM20329 SATA Bridge" is powered down. A "Western Digital Technologies, Inc. Elements 2TB" does not show this issue as far as I can see.

popcornmix, 2015-10-10:
Are you reporting this as a bug or just a comment?
I think if you power off the harddrive that contains the OS then you are in undefined behaviour territory and I doubt we can do anything to fix that.

pelwell, 2015-10-10:
The bug, if indeed that is what it is, is that the powering down is spontaneous and unrequested.

lduesing, 2015-10-11:
What happens is, that the drive powers down by usb-command. And it only happens in newer kernels. I tried 4.1.9 and 4.1.10 which both failed, whereas 4.1.6 is definitely stable. (all via rpi-update <SHA>)

popcornmix, 2015-10-11:
What do you mean by "powers down"?
You can't power the drive down from a usb command. Are you talking about a "spin down"?
Can you post a dmesg log after the failure.
Can you describe how the drive is connected and powered? Sounds more like an inadequate power supply for the drive.

lduesing, 2015-10-12:
Hardrive gets resetted, and put into sleep mode.  (Same as if I unmount it via Windows)
I cannot get a dmesg log, as OS cannot load anything, due to harddrive being down.
Drive is directly connected to raspberry, and powered by original power-supply. (And is running since 13111 hours without any troubles). Harddrive is running on 4.1.6 now since 57 hours without troubles. In 4.1.10 problem occured after 30 minutes to 2 hours.

popcornmix, 2015-10-12:
If you could identify the exact firmware update (and so which 4.1.x kernel version) that introduced the issue, we could have a look through the commits and see if anything stands out.

Ruffio, 2016-08-17:
@lduesing has your issue been resolved? If so, please close this issue. Thanks.

P33M, 2017-05-04:
Closing USB issues as OP has not updated the thread. Further comments by the OP stating that the issue is still present on latest rpi-update kernel will lead to a review.

closed 2017-05-04 by P33M