raspberrypi/linux #1161: [Pi2] rcu_preempt detected stalls

NormalRa, 2015-10-11:
Hello,

My Raspberry Pi 2 doesn't like to work hard; any compilation I have started results in the device freezing/stalling/becoming unresponsive after around ten minutes into the build. Otherwise it functions perfectly fine for longer periods of time (I haven't actually tested if this issue occurs regardless of load by leaving the device unattended for several days — perhaps the chance increases as does cpu usage?). Running Kodi, browsing the web, using P2P software, or transferring large amounts of data also do not stall the device.
Because I've been experiencing this problem for quite a while now, I'm assuming it is in no way related to  74328c7da3924f856cb986509e505b414939060d (by Eric Dumazet).

I'm running Arch Linux with the latest 4.1.10 kernel (with the aforementioned commit). Due to the nature of the distribution all packages have their build "recipes" readily available, and [so does the kernel](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-raspberrypi/PKGBUILD). Therefore AUFS and BFQ patchsets are the only customizations.

A small debug dump excerpt from the console characterising every stall, copied by hand (in case of any errors and for a detailed trace please consult [this image](http://i.imgur.com/E5Ht3Zv.jpg)):
```
INFO: rcu_preempt detected stalls on CPUs/tasks: { 0} (detected by 1, t=2102 jiffies, g=50012, c=50011, q=6543)
[<8077bd70>] (__schedule) from [<80a7ffa0>] (init_thread_union+0x1fa0/0x2000)
rcu_preempt kthread starved for 2102 jiffies!
```

From the taskdump in [that image](http://i.imgur.com/E5Ht3Zv.jpg) it may be observed that I'm using btrfs which is actually interesting because I've wondered if changing the underlying filesystem from ext4 would help. Unfortunately no.

Both PWR (red) and ACT (green) lights during the stall do not blink and constantly shine.

It's also worth mentioning that by using distcc and a non-ARM machine I've managed to successfully compile the kernel for (and from) the Pi. From that I can only conclude that yes indeed, my Raspberry Pi certainly doesn't like to work hard but it does like to boss around.

Any insight into this matter would be greatly appreciated :)

Nerdenator, 2015-10-11:
When you have compile errors, does that include anything in the way of GCC error text leading up to the freeze, or is it just a complete and immediate freeze-out?

NormalRa, 2015-10-11:
I don't have any compilation errors at all. This stall renders the system completely unusable: it does not react to any input in that state, nor is it able to recover from it.

popcornmix, 2015-10-12:
We can't provide much help with the Arch kernel.
Can you try a new raspbian jessie install and see if that has stalls during the same workload?

WarheadsSE, 2015-10-12:
The ALARM kernel has been bumped to 74328c7da3924f856cb986509e505b414939060d

NormalRa, 2015-10-12:
@popcornmix Absolutely! I downloaded and flashed the latest Raspbian image from the official websites and the promptly followed [the official kernel building guide](https://www.raspberrypi.org/documentation/linux/kernel/building.md) because why not.

The issue unfortunately [persists](http://i.imgur.com/qQRkvJ3.jpg).

@WarheadsSE As mentioned in my original post, I have a feeling this is a separate issue.

popcornmix, 2015-10-12:
Is raspbian image also using btrfs?

NormalRa, 2015-10-12:
Nope, I haven't customized anything beyond increasing the log verbosity.

popcornmix, 2015-10-12:
Can you capture the full backtrace for raspbian?
Do you have any USB (or other) peripherals attached?
Is sdcard used for raspbian and the location of the kernel build?

NormalRa, 2015-10-12:
The only USB device use is the official Wi-Fi dongle, but I've used an ordinary USB mouse to set-up the system from the graphical interface, which I then disconnected. Ethernet cable is plugged into my main machine. An HDMI cable is plugged into a monitor which allowed me to acknowledge the debug messages.
The sdcard is used for both Raspbian and the build. My workdir is in pi users' home directory.

I'll report back when I capture the full backtrace :)

popcornmix, 2015-10-13:
It would be useful to determine what is unique about your system that causes the panics.
Can you confirm if you get the panics with the wifi dongle and/or usb mouse removed?

NormalRa, 2015-10-14:
@popcornmix [Here's the backtrace](http://i.imgur.com/xDvTJK9.jpg), sorry for the delay.
This stall occurred without any USB peripherals which leaves ethernet & HDMI plugged in.

The only remaining piece of the puzzle would be the microSD card which is a Samsung product with 16GB of storage. Either the card is problematic or perhaps some of Pi's circuitry was damaged during transport :/

popcornmix, 2015-10-14:
It does seem to be an sdcard issue.
Can you try adding ``dtoverlay=sdhost`` to config.txt which will use a new sdcard driver.
Also can you try with a different sdcard?

NormalRa, 2015-10-14:
Very interesting! I've set `dtoverlay=sdhost`:
```
$ dmesg | grep sdhost
[    4.835809] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
```

The system intermittently reset only two times for now, one of which was definitely my fault (I had set a hung to be treated as a kernel panic and the kernel to reboot after a panic — oops!). The other one, I'm hoping, is related to some bad weather over here. Coincidentally, on both occasions, my monitor was off so no confirmation. Oh dear.

I've had a more-or-less reliable way of triggering the stalls (besides compiling) and it hasn't caused any issues so far. The night is still young though; gcc is being built as I type this so we'll see.

NormalRa, 2015-10-16:
The system hasn't stalled for two days now, that is enough to consider it a proper solution. Thanks @popcornmix

closed 2015-10-16 by NormalRa