raspberrypi/linux #1162: Jessie: apt-get freezes while/after unpacking vim-runtime

lhartvik, 2015-10-12:
Model: Raspberry Pi 2 B+
Image: Jessie (September 2015)
1. Created a new raspbian on a 16 Gb memory card.
2. Booted the raspberry pi 2, quit raspi-config(no overclocking).
3. sudo apt-get update
4. sudo apt-get install vim
The raspberry pi freezes on "Unpacking vim-runtime". 

I did this twice yesterday. Then tried the same on Wheezy(May 2015), and no issues. 
apt-get also works a lot faster on Wheezy.

XECDesign, 2015-10-12:
Tried it and it worked fine. Takes a suspiciously long amount of time, but it doesn't hang. If this can be considered a bug, it should probably go to debian or dpkg maintainers, since it's not a linux kernel issue.

PeterPablo, 2015-10-12:
Even though it might be another issue, I'd like to note that I had to **disable IPv6** on my freshly installed Raspbian Jessie, as `apt-get` would either not resolve the repositories at all or it would take a very long time. Possibly this is an issue with my local network not being (fully) capable of IPv6, but I certainly did not face this problem in Raspbian Wheezy.

SharkWipf, 2016-01-26:
Facing this issue with vim as well, while installing vim-runtime the Raspberry Pi becomes completely unresponsive, both over ssh and locally. It also stops replying to any pings. 
I don't think this can be considered an upstream issue, a package installation shouldn't be able to cause the entire Raspberry Pi to lock up, regardless what kind of package it is. 
I've also noticed the same behavior on other packages during an apt-get upgrade on a clean system.

Software: 
- Everything downloaded and installed today, latest Raspbian from the RasPi site.
- Filesystem expanded
- No overclocks whatsoever
- Vanilla config.txt
- Used several power supplies including ones sold specifically for the RasPi.
- Default amount of swap Raspbian creates, nothing running except ssh (and apt). No GUI.

Hardware: 
- Brand new RasPi 2 B, arrived today
- Kingston Class 10 MicroSD card
- All unnecessary peripherals removed, no monitors or USB devices

At the time of writing I've waited about 30 minutes for the RasPi to finish, with no success.

michaelstoops, 2016-02-08:
I am experiencing a similar issue, with additional symptoms.

Steps to reproduce are the same as lhartvik's original post, but after rebooting, the /etc/sudoers file is messed up. I can't sudo for anything, can't even reboot without CTRL+ALT+DEL or unplug.

I'm using the Raspbian Lite 2015-11-21 image, have seen it manifest a number of times on different hardware, and on the "full" Raspbian image as well.

Ruffio, 2016-08-17:
@lhartvik has your issue been resolved? If so, please close this issue. Thanks.

Ignush, 2016-10-29:
I have the same problem, same goes for installing gnome-user-guide has anyone solved the issue?

XECDesign, 2016-10-29:
This is just how dpkg works when the package has a lot of files. Many resources online point to eatmydata or enabling the unsafe IO option to speed it up, but having profiled dpkg, most of the time is spent in the hashing function and nothing else makes much of a difference.

closed 2017-03-29 by lhartvik

semenmiroshnichenko, 2017-11-09:
I faced to exactly same problem, but on installing python packages, on unpacking step. Solved with using more powerful power supply (2 amps).