raspberrypi/linux #1171: mkknlimg: Problem on Gentoo with xz-compressed Kernels

MatthiasKrause, 2015-10-19:
I've Compressed my RPi2-Kernel (4.1.10-v7+) with XZ on Gentoo. It seems that mkknlimg can't detect that the Kernel is xz-compressed and copy's the kernel in Passthrough Mode. After changing the Compress-methode to gzip it's working.

If i'am right (my Perl-Knowledge is very minimal), the Script uses "unxz --single-stream" to try decompressing the Kernel. The Parameter "--single-stream" doesn't exist on unxz (xz-utils v5.0.8), which is the actually Stable Version on Gentoo. After Upgrading to testing (v5.2.2) it's working now.

Unfortunately the Script says nothing about that Problem, just goes in Passthrough. So It would be nice if the Script would contain a Version-Check of unxz.

Thanks :)

Greeting from Germany
Matthias

pelwell, 2015-10-21:
We don't officially support any distribution other than Raspbian, but if you want to submit a Pull Request with a suitable check we are likely to accept it.

Ruffio, 2016-08-17:
@mkrsn has your issue been resolved? If so, please close this issue. Thanks.

mkrsn, 2016-08-18:
Yes, it was solved by gentoo.

closed 2016-08-18 by mkrsn