raspberrypi/linux #1174: Jessie - apt-get upgrade breaks wifi

MrEngman, 2015-10-24:
Installed Jessie, 2015-09-24-raspbian-jessie, on a new SD card. On a Pi 2 B after running apt-get update, apt-get upgrade the wifi refuses to start up. I have tried 8192cu, 8188eu and 8812au wifi adapters. kernel is `4.1.7-v7+ #817`.

This behaviour has been reported on the forum in several posts - e.g. https://www.raspberrypi.org/forums/viewtopic.php?uid=10221&f=66&t=124069&start=0

With the X gui running, using vncserver on the Pi and TightVNC viewer on a win 7 laptop, selecting the network icon states "no wireless interfaces found", so it is not possible to configure the wifi, but running command `sudo iwlist wlan0 scan` in a terminal window shows a list of locally accessible wifi networks. 

If the wifi has already been configured then it is necessary to run the command `sudo ifup wlan0` to get the wifi to start up.

apt-get upgrade adds packages as follows
```
	The following packages have been kept back:
	  raspberrypi-ui-mods
	The following packages will be upgraded:
	  dhcpcd5 gir1.2-gtk-3.0 libgtk-3-0 libgtk-3-bin libgtk-3-common libsasl2-2 libsasl2-modules libsasl2-modules-db raspberrypi-net-mods raspi-config rc-gui
	11 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
```
apt-get upgrade causes file /etc/network/interfaces to be updated to 
```
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d

auto lo
iface lo inet loopback

iface eth0 inet manual

iface wlan0 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf

iface wlan1 inet manual
    wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```
The wifi sections no longer have auto/allow-hotplug lines included. Altering the file to include `allow-hotplug wlan0` and  `allow-hotplug wlan1` and the wifi starts to work as normal and accessing the network icon shows the list of locally accessible wifi networks again.

XECDesign, 2015-10-24:
Confirmed that the allow-hotplug lines are needed for the interfaces to show up in that menu. Fixing ASAP.

XECDesign, 2015-10-24:
Pushed the fixed package.

closed 2015-10-24 by XECDesign