raspberrypi/linux #1175: Set necessary clocks automatically in pwm-bcm2835.c

gohai, 2015-10-25:
Right now (Raspbian Jessie) it is necessary to make use of the audio output, e.g. by running `omxplayer example.mp3`, before the PMW functionality can be used. With clocks-audio queued for 4.4, and [RPi Upstreaming](http://elinux.org/RPi_Upstreaming), would it be possible to remove this requirement and setup the clocks directly from the driver?

(#1100 added the DT overlays for PWM)

popcornmix, 2015-10-26:
No support in current kernel, but it may be possible in later kernels.

gohai, 2015-11-05:
Seems this is being worked on (yay): http://lists.infradead.org/pipermail/linux-rpi-kernel/2015-November/002499.html

Ruffio, 2016-08-17:
@gohai has your issue been resolved? If so, please close this issue. Thanks.

gohai, 2016-08-18:
@Ruffio No, this is still an issue on latest released Raspbian.

Try this: set `dtoverlay=pwm` in `/boot/config.txt` and reboot. Then:

    $ echo "0" > /sys/class/pwm/pwmchip0/export
    $ echo "1000000" > /sys/class/pwm/pwmchip0/pwm0/period
    $ echo "500000" > /sys/class/pwm/pwmchip0/pwm0/duty_cycle
    $ echo "1" > /sys/class/pwm/pwmchip0/pwm0/enable

Expecting a 1 kHz period with a 50% duty cycle on GPIO18, but no:

[image]

CC @repk

repk, 2017-01-02:
Hi @gohai ,

Sorry to dig an old issue.

Have you solved this issue now ? It appears that this repository does not have an appropriate overlay for my PWM clock patches to work.

You can find a proper overlay in https://github.com/raspberrypi/linux/issues/1533 (it uses GPIO19 but you can modify fragment@0 to set the appropriate function on GPIO18).

gohai, 2017-01-02:
@repk Thanks for your reply!

This is still unresolved for me, but I will re-test shortly. Is the [pwm-overlay.dts](https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/boot/dts/overlays/pwm-overlay.dts) from the raspberrypi kernel not sufficient? (Based on the time this was last modified I presume that I was testing this version).

repk, 2017-01-02:
No the current bcm2708 devicetree create a fixed clock named "pwm" which prevent the real pwm clock to be initialized. To workaround this, you can create a overlay that renames this fixed clock such as the one in #1533 , the pwm-overlay.dts does not do that.

gohai, 2017-01-03:
@repk  I just finished building the latest downstream 4.4 kernel with the overlay you mentioned attached. But before I tried booting into it, I gave the stock kernel (and overlay) a try: and it worked.

My previous test was on an earlier Raspbian distribution, this now is the September release with kernel 4.4.21-v7+. Perhaps Raspbian's new window manager initializes the audio subsystem?

I am inclined to close this bug - @repk, this is not an issue with upstream kernels anyway, I presume?

repk, 2017-01-03:
Ok nice to know that, and yes this should not be an issue with upstream kernels that use different device tree.

gohai, 2017-01-04:
@repk I'll double check this in the coming days with the latest vanilla Raspbian and a logic analyzer, and will close the issue then. Great that upstream is unaffected in any way!

gohai, 2017-01-08:
@repk This is confirmed fixed (tested with vanilla Raspbian, which has 4.4.34-v7+ kernel, and no other overlays besides `pwm`).

closed 2017-01-08 by gohai

shibenyong, 2017-07-17:
May I ask what is the default duty_cycle? Is there any way to change default value for boot?