raspberrypi/linux #1179: Switching on baclight

pssc, 2015-10-30:
[  165.533763] rpi-backlight rpi_backlight: Backlight change failed
[  167.085803] rpi-backlight rpi_backlight: Backlight change failed
[  168.281433] rpi-backlight rpi_backlight: Backlight change failed
[  169.628353] rpi-backlight rpi_backlight: Backlight change failed
[  169.742678] rpi-backlight rpi_backlight: Backlight change failed

Linux screenpi 4.1.12-v7+ #824 SMP PREEMPT Wed Oct 28 16:46:35 GMT 2015 armv7l GNU/Linux

I have an application controling the backlight via the 

/sys/class/backlight/rpi_backlight/bl_power

Powering off writing 4 into the file seems reliable however power on fails.

I can sometime prod it into life via echoing more stuff into the file manually but its not reliable.

'''
       val = level == 0 and "4" or "0"
        if self:getSettings()["backlight"] and (not only or only == "backlight") then
                dirNodeSetValue("/sys/class/backlight","bl_power",val,skipbacklight)
        end
end

function dirNodeSetValue(dir,node,val,skip)
        for entry in lfs.dir(dir) do repeat
                local entrydir = dir.."/"..entry

                if skip and skip[entry] then
                        break
                end
                local entrymode = lfs.attributes(entrydir, "mode")
                log:debug("dirNodeSetValue: ",entrydir," is ", entrymode)
                if entry:match("^%.") or (entrymode ~= "directory") then
                        break
                end

                local fblank = entrydir.."/"..node
                local mode = lfs.attributes(fblank,"mode")
                log:debug("dirNodeSetValue: ",fblank," mode is ", mode)
                if mode == "file" then
                        local fh = io.open(fblank,"w")
                        if fh then
                                log:info("dirNodeSetValue: ",fblank," to ",val)
                                fh:write(val)
                                fh:flush()
                                fh:close()
                        end
                end
                break -- oddness as we are in a double loop
        until true end
end
'''

Code works on gpio_backlight for fbtft displays (adafruit28rt).

P33M, 2015-10-31:
There's a known bug with the firmware glue switching the DSI backlight. A fix is in the patch queue.

pssc, 2015-10-31:
Ok thanks, will retest then as soon as its pushed out...

popcornmix, 2015-11-01:
Firmware is pushed. Please test after running rpi-update.

pssc, 2015-11-01:
@popcornmix one on off cycle then off and back to:-
```
[   94.898890] rpi-backlight rpi_backlight: Backlight change failed
[   95.883632] rpi-backlight rpi_backlight: Backlight change failed
[   98.683035] rpi-backlight rpi_backlight: Backlight change failed
[  102.183071] rpi-backlight rpi_backlight: Backlight change failed
[  102.465310] rpi-backlight rpi_backlight: Backlight change failed
[  104.654344] rpi-backlight rpi_backlight: Backlight change failed
[  114.677035] rpi-backlight rpi_backlight: Backlight change failed
```

P33M, 2015-11-02:
You're switching the backlight at a fast interval. I can see two entries in the log that are less than 500ms apart.

Switching the backlight on and off that quickly will hit the ratelimit built into the firmware.

Is the issue still present if your do .. until loop incorporates a ~1 second delay?

pssc, 2015-11-02:
@P33M I am triggering on a key press (activating soft off),  I left the install overnight and tired triggering again no joy, I am also toggling the frame buffer blanking? at the same time. Even if it skips a setting it should fix after a pause and timer reset, is that rate limit  there for any real reason? and as said this code path works fine for backlights provided via gpio_backlight and fbtft,

I will go back and generate an a application  log and corresponding kernel log a am fairly sure that I only wrote once in a second to the sysfs bl power file but i didn't save the logs.

Phill.

P33M, 2015-11-02:
Setting framebuffer blanking as well as backlight power is unnecessary - blanking the fb will turn off the backlight.

pssc, 2015-11-02:
@P33M ok yes that's probably whats causing the backlight setting in rapid succession though it really should recover gracefully as left overnight and just switching on failed? i am not sure if toggling the blanking on a fbtft frame buffer toggles the bl(and not all display have bl control) which is why the codes in there for the blanking and of course blanking the HDMI output before calling vcgencmd display_power to cause HDMI screen power saving.

P33M, 2015-11-02:
Part of the problem is that it appears the linux-side expects a backlight change to never fail - the backlight update function can return a value, but this value is ignored by the framebuffer update.
c.f. http://lxr.free-electrons.com/source/drivers/video/backlight/backlight.c#L40
Therefore there's the possibility that the backlight state and framebuffer blank/unblank could get out of sync until the next time the backlight is set.

I'll think about this a bit more.

pssc, 2015-11-02:
@P33M something doesn't smell right in my case I am setting blank and power off then unblank and poweron, so it shouldn't matter that the second op failed as it would be set to the same state by either call...
I am doing a complete open close for each file toggle and not keeping the descriptor open.

P33M, 2015-11-02:
Hm. That's right actually - even with the limit you should still be seeing the backlight actually switch state.

Can you post the full script you use so I can replicate the behaviour?

pssc, 2015-11-02:
@P33M is available but theirs quite an overhead as this is an applet embedded in a build of squeeze play of the Pi debs in https://github.com/pssc/squeezeplay-dist its alpha and if you have don't have a Logitech  music server you will have to abort the setup to connect to server by pressing home (q) toggles soft power.  Probably best if I try and document my tests tonight with dmesg and application log output, I cant test from here and I can disable blank and/or backlight control will a simple option to test how thats affecting the interaction....

In summary before firmware update:-
Backlight powered off and I could sometimes prod back into life having cated several random values into the bl sysfs file possibly with several attempts at this.
After update:-
Backlight on / off ok then off and suck off but again just sometimes some toggling would bring it back on.

Re toggled off last night wouldn't switch on left for 7hrs tried to power on again in the morning no joy

pssc, 2015-11-03:
@P33M ok disabling the back light control and just toggling the backlight in the app works so its some artefact of setting both bl and blaking

pssc, 2015-11-03:
@P33M also first toggle blanks screen but doesnt toggle the backlight...

P33M, 2015-11-05:
I've installed squeezeplay and can see the issue.

pssc, 2015-11-05:
@P33M oh good  it not just not me, by default it will toggle both, there should be a menu option under setup advanced SqueezePi to toggle Bl control/ Blanking also add applet.Linux to info in the logging config so see what’s going on and sqeezeplay.ui to warn to reduce noise.  Feed back appreciated as your probably about the 3rd user of that package...

P33M, 2015-11-05:
For some reason, pressing Q (or waiting for a backlight wakeup) causes two writes to the mailbox interface in quick succession. The first always has a bl value of 0, the second is what the backlight is supposed to be, which is why switch-off works but not switch-on.

There is an obvious fix to make to the firmware - don't bother ratelimiting unless the backlight actually changes. This solves this particular case.

P33M, 2015-11-05:
Can you test the start.elf from this link? Replace start.elf in /boot

https://www.dropbox.com/s/jvyfhqij1vyb9ll/start.elf?dl=0

pssc, 2015-11-05:
@P33M ware you still getting two writes event with one of the two options disabled? and did you see on the first toggle after reboot to blanking only that it didn't switch off? , say if the values the same not to bother with the rate limiting would seem to me reasonable.

pssc, 2015-11-05:
@P33M start.elf will do, have to wait till about 11 this eve as the PI here at work doesn’t have a Official Pi LCD screen ;-)

pssc, 2015-11-05:
@P33M ok seems to be good with both blanking and bl control enabled, however still seeing first toggle with blanking only not affect the back light. initial sate odd somewhere?

P33M, 2015-11-06:
If I reset the Pi when SqueezePi is running and the backlight is off, on the next boot I either get an infinite loop of backlight calls to the firmware or a call to turn the backlight on, but the framebuffer remains blanked.

The internal state of the driver and firmware remains consistent.

pssc, 2015-11-06:
ok that’s  different to the issue I am seeing which is when i reboot the pi the back-light on and console is unblanked and squeezepi is on (console blanks bl remains on and I only have blanking control enabled).  I will try replicate that at home.

You could see if the what the application issues by enabling logging(Settings->Advanced->logging) at info of the applet.Linux module stuff should appear is syslog.

Are you running  with bl and blanking control? (Settings->Advanced->SqueezePi) and is your instance connected to a server?

pssc, 2015-11-06:
FYI I only issue bl/ blacking contol on soft power on/off events so struggling to figure how that loops occurring in the code unless your not connected to a server and thats causing the loop somehow.

popcornmix, 2015-11-07:
@P33M's firmware fix in in latest rpi-update firmware.

pssc, 2015-11-10:
@P33M not been able to replicate your bug yet, and hints for your setup, my backlight always powers on on boot.

pssc, 2015-11-10:
ok Squeezplay play aside

latest fw/kernel power pi
```
pssc@lothlorien:255~$ ssh testwipi
pi@testwipi's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Tue Nov 10 22:23:14 2015
pi@screenpi ~ $ sudo -s
root@screenpi:/home/pi# echo 1 >/sys/class/graphics/fb0/blank 
```
Console blanks but bl remains on
Prod console so it unblanks.
```
root@screenpi:/home/pi# echo 1 >/sys/class/graphics/fb0/blank
```
Console blanks and bl goes off, this points to an initial state/set problem

Snyder1977, 2015-11-27:
I have a similar backlight problem.  

@pssc: Do you have a USB-DAC or Hifiberry Amp connected? If yes, is the problem still there when you disconnect the DAC and you are using the default ALSA output?

pssc, 2015-11-30:
@Snyder1977 Nope no dac connected and experienced stability issues initially less so with more recent firmware via rpi-update, now have a pi2's with pi lcd screen with no dac and a Pi-DAC+

maru-sama, 2016-01-20:
I see the same issue as @pssc after an initial boot. If the framebuffer blanks the first time the backlight stays on. After the framebuffer turns on again, subsequent blanks of it also turn off the backlight. So it really is only the first blank which does not go through.
In my specific case I am autostarting a Kivy GUI in the console right after the boot. Since the backlight does not turn off the first time the console blanks the Kivy app continues to show. What I really want is that the screen blanks after 5 minutes. Currently the workaround for me is to blank and unblank the framebuffer one time during boot in rc.local. After that I see the expected behaviour. The backlight goes of after 5 minutes (the app being still active) and turns back on after I touch the screen. @popcornmix: Is there a way I can see whats happening for the backlight the first time the console blanks?

maru-sama, 2016-01-20:
Ok I played around with this some more this evening and found an easier workaround. Putting the following line in rc.local is enough for the screen to work the first time as well.

echo 0 >   /sys/class/graphics/fb0/blank

So instead of turning it off and on again, we just tell it that it is indeed on. This seems to sync the fb and the touchscreen and blanking works.

Ruffio, 2016-08-17:
@pssc has your issue been resolved? If so, please close this issue. Thanks.

Neovy, 2016-11-29:
This issue seems to have reappeared in the newest version of `raspberrypi-bootloader` or one of its dependencies. I use a Pi 2B with a v1.0 screen and writing to `/sys/class/backlight/rpi_backlight/bl_power` results in `rpi-backlight rpi_backlight: Backlight change failed` in the kernel log.

pelwell, 2016-11-29:
> @P33M's firmware fix in in latest rpi-update firmware.

Have you run rpi-update?

Neovy, 2016-11-29:
Yes, I updated all packages and the firmware in a previous attempt, but it breaks as soon as I update `raspberrypi-bootloader`.

z1c0, 2016-11-29:
Confirmed: issue reappears after latest firmware update

XECDesign, 2016-11-29:
There will be an update to the package going out today, but rpi-update should also fix the problem if it runs AFTER the raspberrypi-bootloader upgrade.

pelwell, 2016-11-29:
> Confirmed: issue reappears after latest firmware update

Updated how?

rpi-update is the leading edge of firmware updates - the raspberrypi-bootloader will lag behind that somewhat.

maru-sama, 2016-11-29:
I can reproduce this with the latest version installed via rpi-update. blanking the screen works but re-enabling it results in 

[ 4439.433528] rpi-backlight rpi_backlight: Backlight change failed

maru-sama, 2016-11-30:
Ok 4.4.32 works for me while i get the Error after some time with the current Version.

popcornmix, 2016-11-30:
Can you test this firmware: https://www.dropbox.com/s/m8vvbpgx2x047ox/firmware_bl.zip?dl=0
and report if it solved the issue.

maru-sama, 2016-11-30:
I did an rpi-update to the latest version and then update the firmware files. So far I cannot reproduce the issue, but I will leave it running like that over night and will test it again tomorrow.

maru-sama, 2016-12-01:
Ok i Just tested it again. This Firmware fixes the issue for me.

maru-sama, 2016-12-03:
Hello popcornmix. When while this Change make it into the  Main Kernel or the One available via rpi-update?

jorisvervuurt, 2016-12-03:
A few days ago, I received my original touchscreen.
When I got everything setup (clean Raspbian installation with everything updated), I noticed the screen going blank and then after a while no longer waking back up when touching it.

Then I noticed this issue, which seemed to be exactly the same problem I had.
Running rpi-update didn't fix it, but it does look like the files from the Dropbox link seem to fix it (fingers crossed...).

I'll leave it on over night and come back here to confirm whether or not it has really fixed the issue.

popcornmix, 2016-12-03:
This fix should be in latest rpi-update firmware.

roblan, 2016-12-03:
@popcornmix - some time ago (beofre pixel?) I could change rpi_backlight/brightness very quickly.
Now when I try to change it frequently I get "rpi-backlight rpi_backlight: Backlight change failed".
I wanted to create smooth backlight transition, but now it seems to be impossible?

jorisvervuurt, 2016-12-04:
@popcornmix I can confirm that the firmware downloaded from your Dropbox link has fixed the wake-after-blanking issue. I suggest getting this fix in an apt-get upgrade as soon as possible, to prevent people from returning 'defective' touchscreens. Thanks!

popcornmix, 2016-12-04:
@roblan Backlight switching is rate limited to 500ms for hardware reasons.
@P33M may know the details but I believe something bad happens if you switch too quickly.

roblan, 2016-12-04:
@popcornmix do you know when was this change introduced? And where can I find info why?

popcornmix, 2016-12-04:
The original 500ms rate limit was added over a year ago.
But a bug to this code was introduced in last apt-get upgrade firmware.
You can run rpi-update to get a fix for the current bug, or wait for the fix to reach apt-get upgrade firmware.

gfxfr, 2016-12-06:
@popcornmix Do you know when it will reach apt-get?

Raynman77, 2016-12-15:
I have the same problem. It's really frustrating when you are trying to host a kiosk using this setup. Did not have this problem using Wheezy.  It's rendered this 7" Raspberry Pi Touchscreen completely useless.

jorisvervuurt, 2016-12-16:
@Raynman77 Try running `sudo rpi-update` and all should be fine. ;-)

Raynman77, 2016-12-17:
@jorisvervuurt Doesn't this update Raspbian nightly and unstable updates?

jorisvervuurt, 2016-12-19:
@Raynman77 It does indeed install the latest (development) firmware, which might be unstable. I haven't had any problems with it. @popcornmix When can we expect the fix to be in the latest `sudo apt-get upgrade`?

popcornmix, 2016-12-19:
That's up to @XECDesign 
I did recommend a firmware bump to him

Raynman77, 2016-12-19:
@jorisvervuurt Yes this updated release does fix the issue. Thanks for the heads up. Hopefully there weren't too many bugs in this release.

popcornmix, 2016-12-19:
The apt firmware is being bumped as we speak.
It will be the same firmware as in current ``rpi-update`` so hopefully that is good.

apjavel, 2017-01-22:
Hi, I have been struggling with the same issues as described by others here. I'm on pi3 and latest official image.
Did  a sudo rpi-update as suggested above and The problems went away. Thanks for the fix!

[16 more replies]

closed 2017-05-17 by P33M