raspberrypi/linux #1180: Provide udlfb (support for displaylink) in firmware

paines, 2015-10-30:
Any chance compiling kernel/modules width udlfb ? There is a bunch of us guys using the PI with USB displays. That would be awesome. Thanks!

pssc, 2015-11-10:
this was enabled at some point I know I asked for it and tested it when enabled but that was in the 3.X series

popcornmix, 2015-11-11:
Is ``CONFIG_FB_UDL=m`` sufficient?

paines, 2015-11-11:
@pssc 
When I fired up raspbian it workd just out of the bix. So you are right, it was enabled for 3.X. But after I firmware upgrade, which would pull kernel + overlays, it doesn't work anymore out of the box. One has to pull the sources, and recompile the whole thing.

paines, 2015-11-11:
@popcornmix 
Yes, that would be suffcient.

popcornmix, 2015-11-11:
Okay - added to linux tree. Will appear in next firmware update.

paines, 2015-11-11:
Great! Thank you so much

closed 2015-11-11 by paines

popcornmix, 2015-11-13:
rpi-update firmware now has this module