raspberrypi/linux #1182: vc4-kms-v3d-overlay not working?

gohai, 2015-11-01:
fec023fd3270cc035edcc66766f23c59773f68b2 added an overlay, presumably to test the new DRI driver.

Not sure if this is supposed to work already, but I haven't been able to get this to work on top of Raspbian Jesse, with the following added to config.txt and vc4-kms-v3d-overlay.dtb in /boot/overlays

    mask_gpu_interrupt0=0x400
    avoid_warnings=2
    disable_overscan=1
    dtparam=vc4-kms-v3d-overlay

After bootup, the screen remains at the gradient logo, and `dmesg` doesn't have anything whatsoever regarding `vc4`.

My kernel config is the rpi-4.2.y defconfig with the following [changes on-top](https://github.com/gohai/vc4-buildbot/commit/d8975eb526bb7b670fc74bdf28c8565bcc1b8756).

Please let me know if I am doing anything wrong :)

gohai, 2015-11-01:
Aww, I get the feeling this should probably be `dtoverlay` instead of `dtparam`.. testing.

gohai, 2015-11-01:
Using `dtoverlay` gets me to the same gradient logo with the following in dmesg:

    pi@raspberrypi ~ $ dmesg | grep vc4
    [    1.103216] [drm:vc4_hdmi_bind] *ERROR* Failed to get pixel clock
    [    1.109380] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
    [    1.117928] vc4-drm soc:gpu@7e4c0000: master bind failed: -517
    [    1.913492] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
    [    1.927207] vc4-drm soc:gpu@7e4c0000: master bind failed: -517
    [    2.354824] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
    [    2.363399] vc4-drm soc:gpu@7e4c0000: master bind failed: -517
    [    2.858538] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
    [    2.867237] vc4-drm soc:gpu@7e4c0000: master bind failed: -517
    [    3.766317] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
    [    3.781769] vc4-drm soc:gpu@7e4c0000: master bind failed: -517
    [    3.790871] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
    [    3.801753] vc4-drm soc:gpu@7e4c0000: master bind failed: -517
    [    4.948087] vc4-drm soc:gpu@7e4c0000: failed to bind 3f902000.hdmi (ops vc4_hdmi_ops): -517
    [    4.948232] vc4-drm soc:gpu@7e4c0000: master bind failed: -517

popcornmix, 2015-11-01:
There is nothing ready for testing yet. You will need numerous changes to userland packages to do anything useful with the new kernel driver.

My understanding is that .config changes aren't required, and mask_gpu_interrupt0 is implicit when dtoverlay=vc4-kms-v3d-overlay is used.

Make sure you have updated all dtb files as dtb's from an older kernel can cause issues.

gohai, 2015-11-01:
Thanks @popcornmix - I'll make sure I'll update all dtb files and report back.

I wrote a [script](https://github.com/gohai/vc4-buildbot/blob/master/BuildRaspbianVc4.py) that builds latest Mesa & Xorg for the Raspberry Pi. This worked before with a version of Eric's earlier downstream-kernel tree - currently just trying to get it back to work with Raspbian Jesse & 4.2.

gohai, 2015-11-02:
Tried again, this time with all overlays rebuilt. Same error as above.

If you want to give it a try: [http://sukzessiv.net/~gohai/vc4-buildbot/build/20151101-2034-vc4-image.zip](http://sukzessiv.net/~gohai/vc4-buildbot/build/20151101-2034-vc4-image.zip)

That's rpi-4.2.y with defconfigs and [those .config changes](https://github.com/gohai/vc4-buildbot/commit/d8975eb526bb7b670fc74bdf28c8565bcc1b8756) on top. Kernel was marked as supporting DT. Otherwise: stock `/boot` from Raspbian Jesse.

I tested on a Raspberry Pi 2.

popcornmix, 2015-11-02:
Do you get a framebuffer without any .config changes, but with 4.2 and vc4-kms-v3d-overlay enabled?

gohai, 2015-11-02:
Do you mean .config changes or config.txt changes - without the .config changes the driver wouldn't even be build I presume?

I tried those things on top of my previous comment:

* `dtoverlay=vc4-kms-v3d` instead of `dtoverlay=vc4-kms-v3d-overlay`
* removing all changes to config.txt besides the `dtoverlay` line
* testing again on HDMI-enabled monitor instead of the touchscreen

But no joy :(

popcornmix, 2015-11-02:
I was under the impression that bcm2709_defconfig was sufficient, but perhaps @pelwell can confirm.

pelwell, 2015-11-02:
I have had it booting to a framebuffer with a current kernel + overlay; now it boots, but no framebuffer. I'm investigating.

pelwell, 2015-11-02:
Ah, I was building with a custom config that, as well as setting the DRM and DRM_VC4 config options, sets the CMA allocation to 64MB. That's not something that can be done from Device Tree (which is processed far too late), and I'm not sure if we'd want to set aside that much space for all kernels (although it is usable for movable pages).

gohai, 2015-11-02:
@pelwell Can you send me the .config that works for you or look at mine
linked above. Not sure what I am missing.

pelwell, 2015-11-02:
My working configuration just added three extra settings to the standard _defconfig:

CONFIG_DRM=y
CONFIG_DRM_VC4=y
CONFIG_CMA_SEL_MBYTES=64

@popcornmix reminded me that I can actually set the CMA size from the command line, and that can be appended to from a DT overlay.

If I could get this working with DRM and DRM_VC4 built as modules then I would push my changes, but I'm currently seeing strange timer problems.

pelwell, 2015-11-03:
It seems that all of the problems I was seeing yesterday were due to allowing the old (i.e. current) X server to start. With boot-to-GUI disabled, and provided you don't run startx manually, the Pi will boot with an ARM-hosted framebuffer using the KMS driver if you include `dtoverlay=vc4-kms-v3d` in config.txt; no other settings are necessary.

I'm not claiming this is a particularly useful state in itself, but it is a milestone. I've pushed the required commit that enables the building of the drivers and increases the size of the CMA area.

gohai, 2015-11-06:
@pelwell 
I got it to work on a Pi 2 with a .config that matched yours: https://github.com/gohai/vc4-buildbot/commit/f77ed101e4151cc2654fbda28db669252e228e2b.

However, when I start *without* having also `mask_gpu_interrupt0=0x400` in the config.txt, the boot process stops for me with the message "A start job is running for LSB: Switch to ondemand cpu governor (unless shift key is pressed)". This is not a hard hang, but will wait forever (regardless if I hold shift pressed or not). So it seems this is needed for vc4 after all?

A second comment: 64 MB CMA is not sufficient to get into X on a 1080p display. I got the message `vc4-drm soc:gpu@7e4c0000: failed to allocate buffer with size 1048576`. I suggest upping this to 256 MB, at least on the Pi 2. Unfortunately the command line argument in the overlay takes precedence over what I put in cmdline.txt, so there's currently no way to change this on the fly.

Lastly, just an note for anyone else testing this: vc4 will always just display the gradient on any attached (touchscreen) display - so right now it seems to be necessary to have a HDMI monitor attached.

I'll be uploading an image with the up-to-date userspace code and post the link when it's available.

pelwell, 2015-11-06:
Hmm. The automatic masking of the gpu interrupts based on the contents of the Device Tree, which is enabled for v3d, hdmi, pixelvalve and hvs, first appeared in the October 21st firmware. When did you last rpi-update?

gohai, 2015-11-06:
Oh, never. I am doing all my experiments with stock Raspbian Jesse. Thanks for the pointer!
(What about the avoid_warnings one?)

Thanks.

gohai, 2015-11-06:
@pelwell What about making the firmware use the existing `gpu_mem` setting in config.txt to set the size of the CMA area, if you detect v3d etc in the device tree?

pelwell, 2015-11-06:
If it is possible you might get a low_voltage warning then you should still set avoid_warnings=2.

> What about making the firmware use the existing gpu_mem setting in config.txt to set the size of the CMA area, if you detect v3d etc in the device tree?

That wouldn't work because this CMA area (standard Linux CMA area used for allocating contiguous buffers, rather than the vc_cma area which used to implement a dynamic memory split between ARM and GPU) is in the ARM space, not the GPU space.

Instead, the overlay sets the size of the CMA area. The current recommendation from Eric is 256MB, aligned to a 256MB boundary - this is what the latest overlay uses (cma=256M@512M), but this was only changed yesterday and the update hasn't appeared in a firmware release yet.

gohai, 2015-11-06:
@pelwell I agree that those are not the same, but wouldn't you use them in a similar way? (When running out of graphics memory with the VC IV driver one would want to increase the vc_cma area - while with the vc4 driver one would want to increase the cma area. Both obviously take away from the available system memory.)

According to Eric, the vc_cma area isn't relevant to the vc4 driver at all ("*Nope, not relevant.  We don't call out to the VPU for any memory allocations any more.*"), that's why I suggested setting the cma command line argument based on gpu_mem and the current heuristic that takes into account which board the firmware is running on.

This way, one could also instruct users to (again) increase gpu_mem if they run into allocation failures. Hope this makes sense..

pelwell, 2015-11-06:
The gpu_mem config parameter determines where ARM memory ends and the VPU memory starts; its value varies according to whether you need the camera, etc. The cma kernel parameter determines how much of the ARM memory is set aside for contiguous allocations (or movable pages).

Please describe a suitable heuristic for setting one based on the other.

gohai, 2015-11-06:
What I *thought* could be helpful (pseudo-code):

    if (dt_using_vc4) {
      if (gpu_mem_set_in_config.txt) {
        cma = gpu_mem_set_in_config.txt;
      } else {
        cma = // pick per-board default: 256 MB for Pi 2, 128 MB for Model B, 96 MB for Model A
      }
      vc_cma = 16 MB; // or whatever is the absolute minimum the firmware needs
    } else {
      // cma kernel command line parameter is unset
      // vc_cma is set as it is currently
    }

... but I wasn't aware that vc_cma is used for anything besides the binary graphics driver's allocations, so that might be moot.

Thanks for your reply.

pelwell, 2015-11-06:
I'm against making a change like this in principle because it is less flexible than the current scheme and puts into firmware code (hard to change) what is currently in configuration files (easy to change).

I'm against this change in practice because CMA memory and GPU memory are completely independent, and we don't support vc_cma at all.

However, I am for people making suggestions for improvements, so thanks for trying.

gohai, 2015-11-06:
@pelwell Sorry, perhaps I wasn't so clear - with vc_cma in the pseudo-code above I meant the amount of memory that is set aside for the (binary) graphics driver.

My main concern with the current solution of using a device-tree file to set the cma command line parameter is that this doesn't allow the user an easy way to increase the amount of video memory with the vc4 driver from the default. It is also setting aside probably too much memory on the memory-constrained Raspberry Pi 1.

gohai, 2015-11-06:
@pelwell Since the cma parameter from the DT seems to take precedence over anything in `cmdline.txt`, currently the only way of changing the size of the CMA area would be to change and re-compile the overlay file, as far as I can see.

pelwell, 2015-11-06:
If that is your concern, and I think it is a valid one, then I'd rather change the order of application so that user settings take precedence.

gohai, 2015-11-07:
@pelwell If that's feasible, that'd be great to have.

@popcornmix @pelwell 
You find a working image with vc4 and latest userspace at http://sukzessiv.net/~gohai/vc4-buildbot/build/20151106-1147-vc4-image.zip. I'd be happy to provide continuous builds if I had some place to upload them to :)

I'll make this track rpi-4.2.y and its defconfigs.. Currently this still has `mask_gpu_interrupt0=0x400` in the config.txt, but I'll get rid of that and rather pull in the latest firmware.

Feel free to close this issue. Regards!

gohai, 2015-11-09:
@popcornmix @pelwell 
20151108-2031 and later on http://sukzessiv.net/~gohai/vc4-buildbot/build/ use the updated firmware and no `mask_gpu_interrupt0`. Confirmed working on the Pi 2 w/ HDMI. (Doesn't boot up on the Pi 1, but I assume this is because of the 256 MB for CMA.)

pelwell, 2015-11-09:
Thanks for the feedback. I have a patch to the firmware that shuffles the initialisation order to apply cmdline.txt settings after overlays have been merged. It appears to work, so if it passes scrutiny and there are no gotchas then it will be in the next release.

closed 2015-11-09 by pelwell

pelwell, 2015-11-12:
Yesterday's firmware release included the patch to apply the settings from cmdline.txt last, so you should be able to override the CMA setting.

gohai, 2015-11-12:
Awesome, thanks @pelwell!