raspberrypi/linux #1184: Pi v1 model B+ Ethernet dying

Xenopathic, 2015-11-02:
I'm running Arch Linux ARM on my Raspberry Pi model B+, which packages the 4.1.10 kernel, and I'm seeing an issue with Ethernet dying after some time (I can't say exactly how long, but it's weeks or more).

```
Nov 01 00:07:49 zeus.lan.mccorkell.me.uk kernel: ------------[ cut here ]------------
Nov 01 00:07:49 zeus.lan.mccorkell.me.uk kernel: WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:303 dev_watchdog+0x2dc/0x300()
Nov 01 00:07:49 zeus.lan.mccorkell.me.uk kernel: NETDEV WATCHDOG: eth0 (smsc95xx): transmit queue 0 timed out
Nov 01 00:07:49 zeus.lan.mccorkell.me.uk kernel: Modules linked in: <snip> 
Nov 01 00:07:49 zeus.lan.mccorkell.me.uk kernel: CPU: 0 PID: 0 Comm: swapper Not tainted 4.1.10-2-ARCH #1
Nov 01 00:07:49 zeus.lan.mccorkell.me.uk kernel: Hardware name: BCM2708
Nov 01 00:07:49 zeus.lan.mccorkell.me.uk kernel: [<c001496c>] (unwind_backtrace) from [<c0012834>] (show_stack+0x10/0x14)
Nov 01 00:07:49 zeus.lan.mccorkell.me.uk kernel: [<c0012834>] (show_stack) from [<c0027258>] (warn_slowpath_common+0x78/0xb0)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c0027258>] (warn_slowpath_common) from [<c00272c0>] (warn_slowpath_fmt+0x30/0x40)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c00272c0>] (warn_slowpath_fmt) from [<c06fb0a8>] (dev_watchdog+0x2dc/0x300)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c06fb0a8>] (dev_watchdog) from [<c006f434>] (call_timer_fn+0x30/0x1a4)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c006f434>] (call_timer_fn) from [<c006f894>] (run_timer_softirq+0x208/0x36c)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c006f894>] (run_timer_softirq) from [<c002a7c8>] (__do_softirq+0xe8/0x348)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c002a7c8>] (__do_softirq) from [<c002ad04>] (irq_exit+0xb0/0xfc)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c002ad04>] (irq_exit) from [<c0061544>] (__handle_domain_irq+0x88/0x100)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c0061544>] (__handle_domain_irq) from [<c07b3854>] (__irq_svc+0x34/0xc8)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c07b3854>] (__irq_svc) from [<c000fa20>] (arch_cpu_idle+0x28/0x30)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c000fa20>] (arch_cpu_idle) from [<c0055070>] (cpu_startup_entry+0x1bc/0x260)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: [<c0055070>] (cpu_startup_entry) from [<c0a48c08>] (start_kernel+0x344/0x3b0)
Nov 01 00:07:50 zeus.lan.mccorkell.me.uk kernel: ---[ end trace fb073588b6601421 ]---
```

Once this happens, it seems as though the system still receives packets, but can no longer send any packets out. A reboot is needed to fix the issue.

popcornmix, 2015-11-02:
Arch do build their own kernels with their own kernel options, so it's not something we directly support.
If you can provoke the same failure from a raspbian install (ideally find a way to make the failure happen more quickly) then we can help more.

First, the obvious: If you are overclocking then disable that and confirm the problem still occurs.

Xenopathic, 2015-11-02:
It may well be the overclocking, but I find it strange that this sort of issue can happen at all. Is there any way to reset the hardware if this problem is detected? As it stands at the moment, the Pi can get into a state where it is impossible to recover from without physical access.

popcornmix, 2015-11-02:
overclocking is not guaranteed and can crash in unpredictable ways.
The only was to recover from it is to enabled the watchdog and reboot.
But really you need to back off the overclock if you are seeing instability.
Disable completely now, and if it is stable, you can gradually increase it.

Xenopathic, 2015-11-02:
@popcornmix It was on the modest overclock, I've set it back to defaults. Now we wait...

Ruffio, 2016-08-17:
@Xenopathic has your issue been resolved? If so, please close this issue. Thanks.

closed 2016-09-14 by Xenopathic