raspberrypi/linux #1185: Jessie: Configuration > Localisation > Keyboard setting ignored

wardjk, 2015-11-03:
Some list indexes are ignored and not saved. Go to Raspberry Pi Configuration > Localisation > Set Keyboard > Select Country = United States > Variant = English (US) > OK > OK. Open same setting again and find that Variant = Spanish (Latin America) is selected instead. I first thought this bug was caused by list index 0 being disallowed but found other Variants in the list that were not saved and the selection went back to Spanish (Latin America) (list index 1). Some selections mid-way in the list can be saved but not the first and last.

popcornmix, 2015-11-03:
Can you report this here: https://github.com/raspberrypi/rc_gui/issues
as it's not a linux kernel issue.

closed 2015-11-03 by popcornmix

wardjk, 2015-11-03:
Done. I started with https://www.raspbian.org/RaspbianBugs and followed the link for "filed with Raspberry Pi" which lead me here (https://github.com/raspberrypi/linux/issues).