raspberrypi/linux #1188: openarena package does not work on RPi 2 running Raspbian jessie

spectrumero, 2015-11-04:
The openarena package in the Raspbian jessie distro does not run.

It does not run either under X11 nor on the frame buffer.

Crash log from X11:
ioq3 1.36+u20140802+gca9eebb-2+b1/Raspbian linux-arm Nov  7 2014
----- FS_Startup -----
Current search path:
/home/pi/.openarena/baseoa
/usr/lib/openarena/baseoa
/usr/lib/openarena/baseoa/pak6-patch088.pk3 (711 files)
/usr/lib/openarena/baseoa/pak6-patch085.pk3 (559 files)
/usr/lib/openarena/baseoa/pak6-misc.pk3 (229 files)
/usr/lib/openarena/baseoa/pak5-TA.pk3 (139 files)
/usr/lib/openarena/baseoa/pak4-textures.pk3 (1753 files)
/usr/lib/openarena/baseoa/pak2-players.pk3 (669 files)
/usr/lib/openarena/baseoa/pak2-players-mature.pk3 (231 files)
/usr/lib/openarena/baseoa/pak1-maps.pk3 (100 files)
/usr/lib/openarena/baseoa/pak0.pk3 (1042 files)

----------------------
5433 files in pk3 files
execing default.cfg
execing q3config.cfg
couldn't exec autoexec.cfg
Hunk_Clear: reset the hunk ok
----- Client Initialization -----
Couldn't read q3history.
----- Initializing Renderer ----
Trying to load "renderer_opengl1_arm.so" from "/usr/lib/ioquake3"...
-------------------------------
QKEY found.
----- Client Initialization Complete -----
----- R_Init -----
SDL using driver "x11"
Initializing OpenGL display
Estimated display aspect: 1.333
...setting mode 6: 1024 768
Available modes: '1024x768'
Couldn't get a visual
...WARNING: could not set the given mode (6)
Initializing OpenGL display
...setting mode 6: 1024 768
Available modes: '1024x768'
Couldn't get a visual
...WARNING: could not set the given mode (6)
Setting r_mode 6 failed, falling back on r_mode 3
Initializing OpenGL display
...setting mode 3: 640 480
Available modes: '1024x768'
Couldn't get a visual
...WARNING: could not set the given mode (3)
----- Client Shutdown (Client fatal crashed: GLimp_Init() - could not load OpenGL subsystem) -----
RE_Shutdown( 1 )
Hunk_Clear: reset the hunk ok
-----------------------
GLimp_Init() - could not load OpenGL subsystem

Crash log under frame buffer (run as root):
ioq3 1.36+u20140802+gca9eebb-2+b1/Raspbian linux-arm Nov  7 2014
----- FS_Startup -----
Current search path:
/root/.openarena/baseoa
/usr/lib/openarena/baseoa
/usr/lib/openarena/baseoa/pak6-patch088.pk3 (711 files)
/usr/lib/openarena/baseoa/pak6-patch085.pk3 (559 files)
/usr/lib/openarena/baseoa/pak6-misc.pk3 (229 files)
/usr/lib/openarena/baseoa/pak5-TA.pk3 (139 files)
/usr/lib/openarena/baseoa/pak4-textures.pk3 (1753 files)
/usr/lib/openarena/baseoa/pak2-players.pk3 (669 files)
/usr/lib/openarena/baseoa/pak2-players-mature.pk3 (231 files)
/usr/lib/openarena/baseoa/pak1-maps.pk3 (100 files)
/usr/lib/openarena/baseoa/pak0.pk3 (1042 files)

----------------------
5433 files in pk3 files
execing default.cfg
couldn't exec q3config.cfg
couldn't exec autoexec.cfg
Hunk_Clear: reset the hunk ok
----- Client Initialization -----
Couldn't read q3history.
----- Initializing Renderer ----
Trying to load "renderer_opengl1_arm.so" from "/usr/lib/ioquake3"...
-------------------------------
QKEY found.
----- Client Initialization Complete -----
----- R_Init -----
SDL using driver "fbcon"
Initializing OpenGL display
Estimated display aspect: 1.333
...setting mode 3: 640 480
Available modes: '640x480 768x576 800x600 1024x768 1024x768 1152x864 1280x960 1600x1200 1280x1024 1024x1024'
Couldn't get a visual
...WARNING: could not set the given mode (3)
Initializing OpenGL display
...setting mode 3: 640 480
Available modes: '640x480 768x576 800x600 1024x768 1024x768 1152x864 1280x960 1600x1200 1280x1024 1024x1024'
Couldn't get a visual
...WARNING: could not set the given mode (3)
----- Client Shutdown (Client fatal crashed: GLimp_Init() - could not load OpenGL subsystem) -----
RE_Shutdown( 1 )
Hunk_Clear: reset the hunk ok
-----------------------
GLimp_Init() - could not load OpenGL subsystem

popcornmix, 2015-11-04:
Yes, it's written for desktop GL, not openGL supported by the Pi.

closed 2015-11-04 by popcornmix

spectrumero, 2015-11-04:
Openarena has apparently run on the Pi according to various YouTube videos - is the package just compiled wrong, or perhaps configured incorrectly? Openarena was also included in the Pi store so presumably it has been properly ported to OpenGL ES.

XECDesign, 2015-11-04:
Raspbian is a recompile of debian for amrv6hf, for the most part, as is. The foundation does not currently provide a package of ioquake3 or openarena.

spectrumero, 2015-11-04:
If that's the case could it be removed from the repository, to prevent someone else going through a couple of hours of frustration thinking that since it's in the Raspbian repo it's built for OpenGL-ES, and is widely known to have been ported to the Pi?

XECDesign, 2015-11-04:
Raspbian, the core repo itself, isn't maintained or managed by the raspberry pi foundation.