raspberrypi/linux #1189: Please add CONFIG_MCP320X=m to standard config

tgaryet, 2015-11-04:
mcp320x.c is the driver for a set of Microchip A2D converters:

10 Bit converters
   MCP3001
   MCP3002
   MCP3004
   MCP3008

12 bit converters
   MCP3201
   MCP3202
   MCP3204
   MCP3208

I am creating a HAT module that contains the MCP3004 A2D converter attached to the SPI. Since the HAT device tree would reference this driver, it would be advantageous to have it in the standard configuration.

Please add:
CONFIG_MPC320X=m

to the standard configuration.

popcornmix, 2015-11-04:
Sounds okay.

tgaryet, 2015-11-04:
Thank you.  

I am unfamiliar with how I get access to this when it is done.

pelwell, 2015-11-05:
I've pushed the patch that enables this option. This will cause future builds to include the required driver.

The driver will appear first in the next rpi-update release, available via:
```
sudo rpi-update
```
Some time later it will appear in a Raspbian release, available via:
```
sudo apt-get update
sudo apt-get upgrade
```

tgaryet, 2015-11-05:
Thank you,
Terry

popcornmix, 2015-11-07:
This is now in latest rpi-update firmware. Please test and close if happy.

tgaryet, 2015-11-07:
Tested using:

Linux raspberrypi 4.1.12+ #825 PREEMPT Fri Nov 6 17:53:21 GMT 2015 armv6l

iio device exists...

pi@raspberrypi /sys/bus/iio/devices/iio:device0 $pi@raspberrypi /sys/bus/iio/devices/iio:device0 $ ls in_voltage*_raw
in_voltage0_raw  in_voltage0-voltage1_raw  in_voltage1_raw  in_voltage2_raw  in_voltage2-voltage3_raw  in_voltage3_raw

A2D data reported...

pi@raspberrypi /sys/bus/iio/devices/iio:device0 $ cat in_voltage*_raw
1
0
1
0
0
0

Looks like it works. 
Thank You!

closed 2015-11-07 by tgaryet