raspberrypi/linux #1190: kernel 4.2.5 does not boot in DT mode

hennichodernich, 2015-11-05:
Hi,

I have compiled a 4.2.5 kernel that boots fine if it is bootes as a "raw" zImage, but not if tagged with mkknlimg. In that case, the rainbow square screen freezes and on the serial console I only see "uncompressing linux... done booting the kernel".

The first 25 lines of dmesg for a successful boot read
`[    0.000000] Booting Linux on physical CPU 0x0`
`[    0.000000] Initializing cgroup subsys cpuset`
`[    0.000000] Initializing cgroup subsys cpu`
`[    0.000000] Initializing cgroup subsys cpuacct`
`[    0.000000] Linux version 4.2.5-hnch+ (paul@oslo) (gcc version 4.7.1 20120402 (prerelease) (crosstool-NG 1.15.2) ) #4 PREEMPT Thu Nov 5 10:37:31 CET 2015`
`[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d`
`[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache`
`[    0.000000] Machine: BCM2708`
`[    0.000000] cma: Reserved 8 MiB at 0x1b800000`
`[    0.000000] Memory policy: Data cache writeback`
`[    0.000000] On node 0 totalpages: 114688`
`[    0.000000] free_area_init_node: node 0, pgdat c089f74c, node_mem_map db403000`
`[    0.000000]   Normal zone: 1008 pages used for memmap`
`[    0.000000]   Normal zone: 0 pages reserved`
`[    0.000000]   Normal zone: 114688 pages, LIFO batch:31`
`[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768`
`[    0.000000] pcpu-alloc: [0] 0 `
`[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 113680`
`[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1200 bcm2708.boardrev=0xe bcm2708.serial=0x9df9661c smsc95xx.macaddr=B8:27:EB:F9:66:1C bcm2708_fb.fbswap=1 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait`
`[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)`
`[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)`
`[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)`
`[    0.000000] Memory: 436468K/458752K available (6195K kernel code, 496K rwdata, 1780K rodata, 360K init, 724K bss, 14092K reserved, 8192K cma-reserved)`
`[    0.000000] Virtual kernel memory layout:`
`[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)`

The kernel config is taken from raspbian, using "make olddefconfig".

Can anyone point me into the correct direction?

best regards
Henning

pelwell, 2015-11-05:
What are you trying to achieve? Have you followed the kernel building instructions [here](https://www.raspberrypi.org/documentation/linux/kernel/building.md)?

If you can upload the failing kernel image somewhere (dropbox, google drive etc.) I can take a look.

hennichodernich, 2015-11-05:
Thanks for the hint, I did not regenerate the dtbs, since I was using an older tutorial and that worked for a 4.1.y kernel.
Now it's working, thank you very much!

closed 2015-11-05 by hennichodernich