raspberrypi/linux #1198: The "memory" submodule of cgroups is not enabled (?)

Terminus-IMRC, 2015-11-14:
I can't use the "memory" submodule of cgroups on Raspberry Pi.

```/proc/config.gz``` says:
```
...
CONFIG_CGROUPS=y
...
CONFIG_MEMCG=y
# CONFIG_MEMCG_SWAP is not set
# CONFIG_MEMCG_KMEM is not set
...
```
which indicates that the "memory" submodule is enabled.
(c.f. https://www.kernel.org/doc/Documentation/cgroups/memory.txt )

But ```/proc/cgroups``` says:
```
#subsys_name    hierarchy       num_cgroups     enabled
...
memory  0       1       0
...
```
which indicates that the "memory" submodule is not enabled.

Also, mounting ```/sys/fs/cgroup/memory``` fails:
```
# mount -t tmpfs none /sys/fs/cgroup -o rw # Add "remount" option if needed
# mkdir -p /sys/fs/cgroup/memory
# mount -t cgroup none /sys/fs/cgroup/memory -o memory
mount: special device none does not exist
#
```

Therefore, the "memory" submodule is actually not enabled.

Do you have any idea why is it?

Thanks,
Terminus-IMRC.

popcornmix, 2015-11-14:
See: https://github.com/raspberrypi/linux/commit/853fe067aeff8a0be4027b63f50dd0478103f723 and https://www.raspberrypi.org/blog/web-browser-beta/
Enable with ``cgroup_enable=memory`` in cmdline.txt.

Terminus-IMRC, 2015-11-14:
Now it works! Thanks!
And sorry for the lack of my searches.

closed 2015-11-14 by Terminus-IMRC

barlock, 2018-02-01:
For those brought here by google, its now 

```
cgroup_memory=1
```

[source](https://www.reddit.com/r/raspberry_pi/comments/7avo0w/cant_enable_memory_cgroup_on_raspberry_pi_3/)

pelwell, 2018-02-01:
As of 01 Dec 2017 `cgroup_enable=memory` works again, but `cgroup_memory=1` also works.

hugalafutro, 2021-04-02:
Well here I am in 2021 and for whatever esoteric reason `cgroup_memory=1` doesn't work, but `cgroup_enable=memory` does.