raspberrypi/linux #1201: can't install dnsutils

CNMan, 2015-11-15:
```
pi@raspberrypi ~ $ sudo apt-get install dnsutils
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 dnsutils : Depends: libisc95 (= 1:9.9.5.dfsg-12) but 1:9.9.5.dfsg-12+b1 is to be installed
            Depends: liblwres90 (= 1:9.9.5.dfsg-12) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
os:raspbian testing

dlech, 2015-11-15:
This repository is for the linux kernel only, so this is not a valid issue.

Also, I see you are using raspbian testing. This is the nature of using testing. Things like the break and it is expected that users of testing learn how to resolve them. If you do want to learn, I suggest using `aptitude` and `apt-cache` tools to investigate dependencies. If you don't want to learn, I suggest sticking to stable instead of testing.

In this particular case however, since the dependency is strictly `=` and not `>=`, you will probably just have to wait until the dnsutils package is rebuilt against the new library in testing.

CNMan, 2015-11-15:
@dlech thanks very much!

closed 2015-11-15 by CNMan