raspberrypi/linux #1202: Input: xpad - add support for NEW Xbox One controllers

drsternau, 2015-11-16:
When I plug in the new Xbox One Controller (this one has a audio jack) my system crashes. 
The issue seems to be the unrecognized id of the controller. 
As Erik Lundgren pointed out, this issue can be addressed by adding 

{ 0x045e, 0x02dd, "Microsoft X-Box One pad (Covert Forces)", 0, XTYPE_XBOXONE }, 

as line 128.

It this possible

Ferroin, 2015-11-16:
This is a known issue in upstream Linux, and a patch has been posted there.

lurch, 2016-01-18:
I suspect if you provide a link to the (accepted?) upstream patch, @popcornmix will be happy to pull it into this repo.

popcornmix, 2016-01-18:
@Ferroin can you point at the upstream commit? I'm not seeing it on 4.4 for example.
In general if it is accepted upstream and we can cherry-pick it, we are much more likely to add to the kernel (as the patch will be dropped at some future kernel bump).
If it's not accepted upstream, then we are less likely to pick it.

Ferroin, 2016-01-18:
I don't know the exact commit, and I don't recall when it got merged, but I'm pretty sure something ended up in 4.3 to fix it.

popcornmix, 2016-01-18:
Okay it seems to be in 4.4:
https://github.com/raspberrypi/linux/blob/rpi-4.4.y/drivers/input/joystick/xpad.c#L128

we are moving to this kernel in the near future and it can be tested early with:
```sudo BRANCH=next rpi-update```

Ruffio, 2016-08-17:
@drsternau has your issue been resolved? If so, please close this issue. Thanks.

drsternau, 2016-08-18:
@Ruffio Sorry! Yeah, the issue has been resolved.

closed 2016-08-18 by drsternau