raspberrypi/linux #1205: boot failure with sdhost overlay

ali1234, 2015-11-20:
I tried to boot with both the sdio and sdhost overlays on Raspberry Pi 2. In both cases the SD hardware was correctly detected and the kernel was able to determine the partitions on the SD card, but then everything failed with IRQ timeouts. I don't have anything connected to the SD pins on the GPIO header, but this should not matter when only using the sdhost overlay, should it?

My kernel is about two weeks old, but I don't see any related changes. I'll try with the latest one and provide full logs if this is not expected behavior.

Logs snippets. During a normal boot I see this:

```
[    3.062641] sdhci: Secure Digital Host Controller Interface driver
[    3.070371] sdhci: Copyright(c) Pierre Ossman
[    3.076739] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
[    3.084146] mmc-bcm2835 3f300000.mmc: DMA channels allocated
[    3.128112] sdhci-pltfm: SDHCI platform and OF driver helper
[    3.140439] ledtrig-cpu: registered to indicate activity on CPUs

[    3.206327] mmc0: host does not support reading read-only switch, assuming write-enable
[    3.206475] mmc0: new SD card at address 99d6
[    3.207106] mmcblk0: mmc0:99d6 SU02G 1.89 GiB 
[    3.211098]  mmcblk0: p1 p2
```

When booting with sdhost or sdio I see this:

```
[    3.062679] sdhci: Secure Digital Host Controller Interface driver
[    3.070408] sdhci: Copyright(c) Pierre Ossman
[    3.168041] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
[    3.177423] mmc-bcm2835 3f300000.mmc: mmc_debug:0 mmc_debug2:0
[    3.188017] mmc-bcm2835 3f300000.mmc: DMA channels allocated
[    3.228100] sdhci-pltfm: SDHCI platform and OF driver helper
[    3.235970] ledtrig-cpu: registered to indicate activity on CPUs
[    3.238092] Indeed it is in host mode hprt0 = 00021501
[    3.245971] mmc0: host does not support reading read-only switch, assuming write-enable
[    3.246085] mmc0: new SD card at address 99d6
[    3.246746] mmcblk0: mmc0:99d6 SU02G 1.89 GiB 
[    3.248301]  mmcblk0: p1 p2

[   13.947935] mmc0: timeout waiting for hardware interrupt.
[   13.955232] mmc0: cmd op 25 arg 0x37a18000 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[   13.968303] mmc0: data blocks 8 blksz 200 - err 0
[   13.974824] mmc0:>stop op 12 arg 0x0 flags 0x49d - resp 00000000 00000000 00000000 00000000, err 0
[   13.987422] mmc0: =========== REGISTER DUMP ===========
[   13.994504] mmc0: SDCMD  0x0000080c
[   13.999848] mmc0: SDARG  0x00000000
[   14.005154] mmc0: SDTOUT 0x00bebc20
[   14.010437] mmc0: SDCDIV 0x00000008
[   14.015650] mmc0: SDRSP0 0x00000c00
[   14.020785] mmc0: SDRSP1 0x00000c0e
[   14.025861] mmc0: SDRSP2 0x7fffffff
[   14.030875] mmc0: SDRSP3 0x0002400c
[   14.035817] mmc0: SDHSTS 0x00000000
[   14.040701] mmc0: SDVDD  0x00000001
[   14.045543] mmc0: SDEDM  0x00010801
[   14.050307] mmc0: SDHCFG 0x0000040e
[   14.055010] mmc0: SDHBCT 0x00000006
[   14.059623] mmc0: SDHBLC 0x00000001
[   14.064169] mmc0: ===========================================
```
... this repeats forever and the root filesystem never comes up.

ali1234, 2015-11-20:
Now tested with latest kernel. Here is the full kernel log:

```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.13-v7+ (al@al-desktop) (gcc version 4.8.3 20140106 (prerelease) (crosstool-NG linaro-1.13.1-4.8-2014.01 - Linaro GCC 2013.11) ) #3 SMP PREEMPT Fri Nov 20 06:43:32 GMT 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B
[    0.000000] cma: Reserved 8 MiB at 0x3a000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] [bcm2709_smp_init_cpus] enter (9420->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @ba9e3000 s20608 r8192 d24448 u53248
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa01041 bcm2709.serial=0x3c5da33a smsc95xx.macaddr=B8:27:EB:5D:A3:3A bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 933556K/966656K available (5967K kernel code, 534K rwdata, 1652K rodata, 420K init, 757K bss, 24908K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff000000   (1080 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80778f64   (7620 kB)
[    0.000000]       .init : 0x80779000 - 0x807e2000   ( 420 kB)
[    0.000000]       .data : 0x807e2000 - 0x80867b6c   ( 535 kB)
[    0.000000]        .bss : 0x8086a000 - 0x8092779c   ( 758 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000011] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000032] Switching to timer-based delay loop, resolution 52ns
[    0.000306] Console: colour dummy device 80x30
[    0.002072] console [tty1] enabled
[    0.002136] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.002237] pid_max: default: 32768 minimum: 301
[    0.002607] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002673] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003995] Initializing cgroup subsys blkio
[    0.004077] Initializing cgroup subsys memory
[    0.004144] Initializing cgroup subsys devices
[    0.004203] Initializing cgroup subsys freezer
[    0.004279] Initializing cgroup subsys net_cls
[    0.004385] CPU: Testing write buffer coherency: ok
[    0.004507] ftrace: allocating 20235 entries in 60 pages
[    0.053989] CPU0: update cpu_capacity 1024
[    0.054071] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.054118] [bcm2709_smp_prepare_cpus] enter
[    0.054281] Setting up static identity map for 0x8240 - 0x8274
[    0.113872] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.114302] [bcm2709_secondary_init] enter cpu:1
[    0.114352] CPU1: update cpu_capacity 1024
[    0.114360] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.133849] [bcm2709_boot_secondary] cpu:2 started (0) 17
[    0.134200] [bcm2709_secondary_init] enter cpu:2
[    0.134229] CPU2: update cpu_capacity 1024
[    0.134237] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.153894] [bcm2709_boot_secondary] cpu:3 started (0) 16
[    0.154148] [bcm2709_secondary_init] enter cpu:3
[    0.154176] CPU3: update cpu_capacity 1024
[    0.154184] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[... 319 lines cut ...]
[   95.219547] mmc0: SDVDD  0x00000001
[   95.224276] mmc0: SDEDM  0x00010800
[   95.228957] mmc0: SDHCFG 0x0000040a
[   95.233592] mmc0: SDHBCT 0x00000000
[   95.238183] mmc0: SDHBLC 0x00000000
[   95.242688] mmc0: ===========================================
[  105.267985] mmc0: timeout waiting for hardware interrupt.
[  105.274465] mmc0:>cmd op 52 arg 0x80000c08 flags 0x195 - resp 00000000 00000000 00000000 00000000, err 0
[  105.286011] mmc0: =========== REGISTER DUMP ===========
[  105.292249] mmc0: SDCMD  0x00004034
[  105.296703] mmc0: SDARG  0x80000c08
[... cut ...] 00
[  105.305548] mmc0: SDCDIV 0x0000026f
[  105.309977] mmc0: SDRSP0 0xffffffff
[  105.314406] mmc0: SDRSP1 0x0000ff7f
[  105.318832] mmc0: SDRSP2 0x0fffffff
[  105.323253] mmc0: SDRSP3 0x00000000
[  105.327679] mmc0: SDHSTS 0x00000040
[  105.332111] mmc0: SDVDD  0x00000001
[  105.336545] mmc0: SDEDM  0x00010800
[  105.340982] mmc0: SDHCFG 0x0000040a
[  105.345421] mmc0: SDHBCT 0x00000000
[  105.349856] mmc0: SDHBLC 0x00000000
[  105.354281] mmc0: ===========================================
```

pelwell, 2015-11-20:
That doesn't match my experience or understanding.

If your description is correct, I can see an anomaly in your logs - the sdhost overlay should disable the bcm2835-mmc driver, but I can see that both are loaded. A failure to disable the mmc driver (and interface) could leave sdhost driver running but unable to access the SDCard interface since the pins are still mapped to the mmc interface.

Please retest with a stock kernel.

=====

I initially read your problem as being trying to run both at the same time. On re-reading I can see that this may not be the case, but here's my analysis anyway.

The sdhost overlay does three things:
1) adds a device node for the sdhost driver
2) maps the SDCard pins (48-53) to the sdhost interface
3) disables the mmc driver

The sdio overlay does all of those, but also:
4) maps pins 22-27 to the mmc interface
5) re-enables the mmc driver

The sdio overlay actually includes the sdhost overlay so all 5 steps are present, but the end result is effectively 1+2+4.

As you may have worked out by now, order of application is important where overlays overlap.
```
dtoverlay=sdhost
dtoverlay=sdio
```
is equivalent to
```
dtoverlay=sdio
```
but
```
dtoverlay=sdio
dtoverlay=sdhost
```
should be equivalent to
```
dtoverlay=sdhost
```

This overlap is hinted at in the README, although it isn't spelled out:
```
Name:   sdhost
Info:   Selects the bcm2835-sdhost SD/MMC driver, optionally with overclock
```
vs
```
Name:   sdio
Info:   Selects the bcm2835-sdhost SD/MMC driver, optionally with overclock,
        and enables SDIO via GPIOs 22-27.
```
I have just tried all four configurations and, without actually trying to use the GPIO pins for sdio, my experience is sdhost+sdio=sdio, and sdio+sdhost=sdhost.

ali1234, 2015-11-20:
The last log I posted is from a stock kernel which I built this morning according to the cross compiling instructions at https://www.raspberrypi.org/documentation/linux/kernel/building.md
and using commit f2e0b81c8843be005f519a9499ab1c559ad91026

When I use a kernel fetched by rpi-update the problem does not happen, but this gives me an extremely out of date kernel version. Here is the log when using such a kernel:

config.txt
```
[pi2]
kernel=kernel7.img
initramfs initrd.img-3.18.0-trunk-rpi2 followkernel
dtparam=i2c_arm=on
dtoverlay=sdhost
```

kernel log:
```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.18.13-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #790 SMP PREEMPT Wed May 20 19:56:14 BST 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B
[    0.000000] cma: Reserved 8 MiB at 0x3a000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] [bcm2709_smp_init_cpus] enter (8620->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 10 pages/cpu @ba9f3000 s11456 r8192 d21312 u40960
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239776
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa01041 bcm2709.serial=0x3c5da33a smsc95xx.macaddr=B8:27:EB:5D:A3:3A bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 934864K/966656K available (5739K kernel code, 397K rwdata, 1756K rodata, 384K init, 763K bss, 31792K reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xffe00000   (2048 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff000000   (1080 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80759fd4   (7496 kB)
[    0.000000]       .init : 0x8075a000 - 0x807ba000   ( 384 kB)
[    0.000000]       .data : 0x807ba000 - 0x8081d6bc   ( 398 kB)
[    0.000000]        .bss : 0x8081d6bc - 0x808dc494   ( 764 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (virt).
[    0.000015] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 3579139424256ns
[    0.000038] Switching to timer-based delay loop, resolution 52ns
[    0.000335] Console: colour dummy device 80x30
[    0.001795] console [tty1] enabled
[    0.001844] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001927] pid_max: default: 32768 minimum: 301
[    0.002333] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002396] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003656] Initializing cgroup subsys memory
[    0.003744] Initializing cgroup subsys devices
[    0.003816] Initializing cgroup subsys freezer
[    0.003870] Initializing cgroup subsys net_cls
[    0.003919] Initializing cgroup subsys blkio
[    0.004032] CPU: Testing write buffer coherency: ok
[    0.004147] ftrace: allocating 19681 entries in 58 pages
[    0.052910] CPU0: update cpu_capacity 1024
[    0.052990] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.053031] [bcm2709_smp_prepare_cpus] enter
[    0.053183] Setting up static identity map for 0x52c838 - 0x52c86c
[    0.112810] [bcm2709_boot_secondary] cpu:1 started (0) 17
[    0.113127] CPU1: Booted secondary processor
[... 220 lines cut ...]
[    6.108874] systemd[1]: Started udev Coldplug all Devices.
[    6.119795] systemd[1]: Time has been changed
[    6.317542] systemd[1]: Starting Remount Root and Kernel File Systems...
[    6.333480] systemd[1]: Mounting Configuration File System...
[    6.348463] systemd[1]: Mounted FUSE Control File System.
[    6.356542] systemd[1]: Starting Apply Kernel Variables...
[    6.360974] EXT4-fs (mmcblk0p2): re-mounted. Opts: errors=remount-ro
[    6.379572] systemd[1]: Starting Create Static Device Nodes in /dev...
[    6.395423] systemd[1]: Starting Syslog Socket.
[    6.404972] systemd[1]: Listening on Syslog Socket.
[    6.412020] systemd[1]: Starting Journal Service...
[    6.428140] systemd[1]: Started Journal Service.
[    6.631047] systemd-udevd[170]: starting version 215
[    7.087447] bcm2708_i2c 3f804000.i2c: BSC1 Controller at 0x3f804000 (irq 79) (baudrate 100000)
[... cut ...]
[    7.423855] systemd-journald[165]: Received request to flush runtime journal from PID 1
[    8.020992] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    8.030490] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    9.648955] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[    9.661338] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

Raspbian GNU/Linux 8 al-jessie ttyAMA0

al-jessie login: [   17.066480] random: nonblocking pool is initialized
```

pelwell, 2015-11-20:
For some reason, you are not running the latest rpi-update kernel. If you have a spare SD card, download and install the latest Raspbian, run rpi-update, then try loading the overlay. This will establish a baseline.

ali1234, 2015-11-20:
Just realised this is because I did "rpi-update next" instead of "rpi-update". Retesting now.

ali1234, 2015-11-20:
With rpi-update kernel:

```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.13-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3a000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] [bcm2709_smp_init_cpus] enter (9420->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @ba9e3000 s20608 r8192 d24448 u53248
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa01041 bcm2709.serial=0x3c5da33a smsc95xx.macaddr=B8:27:EB:5D:A3:3A bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 elevator=deadline root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 933556K/966656K available (5967K kernel code, 534K rwdata, 1652K rodata, 420K init, 757K bss, 24908K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xbb800000 - 0xff000000   (1080 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80778f64   (7620 kB)
[    0.000000]       .init : 0x80779000 - 0x807e2000   ( 420 kB)
[    0.000000]       .data : 0x807e2000 - 0x80867b6c   ( 535 kB)
[    0.000000]        .bss : 0x8086a000 - 0x8092779c   ( 758 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000011] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000033] Switching to timer-based delay loop, resolution 52ns
[    0.000317] Console: colour dummy device 80x30
[    0.002120] console [tty1] enabled
[    0.002185] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.002286] pid_max: default: 32768 minimum: 301
[    0.002662] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002727] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.004035] Initializing cgroup subsys blkio
[    0.004115] Initializing cgroup subsys memory
[    0.004183] Initializing cgroup subsys devices
[    0.004245] Initializing cgroup subsys freezer
[    0.004321] Initializing cgroup subsys net_cls
[    0.004432] CPU: Testing write buffer coherency: ok
[    0.004548] ftrace: allocating 20235 entries in 60 pages
[    0.054223] CPU0: update cpu_capacity 1024
[    0.054307] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.054353] [bcm2709_smp_prepare_cpus] enter
[    0.054517] Setting up static identity map for 0x8240 - 0x8274
[    0.114094] [bcm2709_boot_secondary] cpu:1 started (0) 17
[    0.114530] [bcm2709_secondary_init] enter cpu:1
[    0.114583] CPU1: update cpu_capacity 1024
[    0.114592] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.134065] [bcm2709_boot_secondary] cpu:2 started (0) 18
[    0.134421] [bcm2709_secondary_init] enter cpu:2
[    0.134454] CPU2: update cpu_capacity 1024
[    0.134462] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.154109] [bcm2709_boot_secondary] cpu:3 started (0) 17
[    0.154367] [bcm2709_secondary_init] enter cpu:3
[    0.154395] CPU3: update cpu_capacity 1024
[    0.154403] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[... 202 lines cut ...]
[    5.611358] systemd[1]: Time has been changed
[    5.809014] systemd[1]: Mounting Configuration File System...
[    5.823031] systemd[1]: Mounted FUSE Control File System.
[    5.831051] systemd[1]: Starting Apply Kernel Variables...
[    5.845636] systemd[1]: Starting Remount Root and Kernel File Systems...
[    5.862040] systemd[1]: Starting Create Static Device Nodes in /dev...
[    5.878115] systemd[1]: Starting Syslog Socket.
[    5.887817] systemd[1]: Listening on Syslog Socket.
[    5.894191] EXT4-fs (mmcblk0p2): re-mounted. Opts: errors=remount-ro
[    5.903119] systemd[1]: Starting Journal Service...
[... cut ...]
[    6.806266] systemd-journald[176]: Received request to flush runtime journal from PID 1
[    7.395978] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    7.405482] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.999773] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    9.008667] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1

Raspbian GNU/Linux 8 al-jessie ttyAMA0

al-jessie login: [   16.679311] random: nonblocking pool is initialized
```

pelwell, 2015-11-20:
Now you just have to work out why your kernel is different. Did you update the .dtbs?

ali1234, 2015-11-20:
Well, after reinstalling my own kernel, now it works too. I did install all dtbs and modules, so it must have been a firmware mismatch?

How exactly am I supposed to get a matching firmware for a custom built kernel?

ali1234, 2015-11-20:
Seems like the answer is `SKIP_KERNEL=1 rpi-update` to get the latest firmware without the kernel.

closed 2015-11-20 by ali1234