raspberrypi/linux #1207: Kernel panic with driver dm9601

jmceara, 2015-11-20:
Hi. driver 'dm9601' has a bug that causes kernel panic and system crash when you try to download or send/receiver any packet bigger than a 'ping'. This patch is simple and solve this issue. Already tested and works. Can you please commit to office kernel tree? Tks.

[dm9601-bug.patch.txt](https://github.com/raspberrypi/linux/files/40324/dm9601-bug.patch.txt)

popcornmix, 2015-11-20:
This needs to be reported upstream. If verified as the correct solution we can pull it in.
https://www.kernel.org/pub/linux/docs/lkml/reporting-bugs.html
https://www.kernel.org/doc/Documentation/SubmittingPatches

Ruffio, 2016-08-17:
@jmceara has your issue been resolved? If so, please close this issue. Thanks.

closed 2017-05-18 by JamesH65