raspberrypi/linux #1208: Backlight toggle function for official raspberry pi DSI display

Snyder1977, 2015-11-25:
The function to turn on/off the backlight of my pi is unstable. 

I usually run the following commands:

To turn the Display on:
'sudo bash -c "echo 0 > /sys/class/backlight/rpi_backlight/bl_power"'

To turn the Display off:
'sudo bash -c "echo 1 > /sys/class/backlight/rpi_backlight/bl_power"'

Sometimes it is working fine for me and sometimes it crashes the pi. The pi crashes and reboots immediatley after the execution of one of the above mentioned commands.

I don´t know what the reason might be? Do you have any ideas how to fix it?

For further details please have a look at this thread:

https://www.raspberrypi.org/forums/viewtopic.php?f=108&t=120968&sid=3914e9536acc702e0d27d490e2f6cc8e&start=75

popcornmix, 2015-11-26:
What is output of ``vcgencmd version``

Ping @P33M

Snyder1977, 2015-11-26:
Here is the output:

    pi@boomboxII ~/script $ vcgencmd version
Nov 18 2015 15:43:45 
Copyright (c) 2012 Broadcom
version 0c9af7cde38e223c95c3404f468db8fc5004495c (clean) (release)

Snyder1977, 2015-11-30:
I figured out why the pi was crashing sometimes and sometimes not.  The issue seemed to be related to power supply. My 7" display was powered via the GPIO pins of my PI2. The GPIO jumper cables did not supply the display with enough voltage. It was more a problem of the cable than of the power supply device.

Although I had a decent 6A power supply, my pi crashed very often when I switched off the backlight.
Since I am powering the display with a separate power supply, switching on/off the backlight is working reliable.

We can close this issue.

closed 2015-11-30 by Snyder1977