raspberrypi/linux #1210: Cannot load vc4 driver

bobodeng520, 2015-11-30:
My pi is rpi B Rev 2(2015-09-24-raspbian-jessie), with latest rpi-4.2.y kernel:
1) Update firmware;
2)Using gohai build script: https://github.com/gohai/vc4-buildbot/blob/master/BuildRaspbianVc4.py linux build part;
3)add:
avoid_warnings=2
dtoverlay=vc4-kms-v3d
to config.txt;

I cannot load vc4 driver, there is nothing vc4 message in dmesg;

And if I use modprobe vc4, seem it load it, but the dmesg has only these message:
[drm] Initialized drm 1.1.0 20060810

Still cannot use vc4 driver;

pelwell, 2015-11-30:
What does:
```
sudo vcdbg log msg |& grep -v gpioman | grep -iv hdmi
```
show?

bobodeng520, 2015-11-30:
It shows:

001119.975: *** Restart logging
001120.927: Read command line from file 'cmdline.txt'
dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
001121.709: Loading 'kernel.img' from SD card
001366.108: Kernel trailer DTOK property says yes
001366.369: Loading 'bcm2708-rpi-b.dtb' from SD card
001408.595: Failed to load overlay 'vc4-kms-v3d'
001408.643: dtparam: cache_line_size=32
003913.075: vchiq_core: vchiq_init_state: slot_zero = 0x5bc80000, is_master = 1

pelwell, 2015-11-30:
```
001408.595: Failed to load overlay 'vc4-kms-v3d'
```
There's your answer. Does vc4-kms-v3d-overlay.dtb exist in /boot/overlays/?

bobodeng520, 2015-11-30:
But if I copy the vc4-kms-v3d-overlay.dtb to /boot/overlays, it cannot boot:
```javascript
Uncompressing Linux... done, booting the kernel.
```

It will hang in there.

pelwell, 2015-11-30:
The VC4 KMS driver needs lots of RAM to work well, so the overlay is aimed at Pi 2 users with 1GB. I see you are on a Pi Model B, so you are probably running out of memory. You are unlikely to have a good user experience, but if you really want to try then add this to your `cmdline.txt` (you will need a recent firmware - Nov 11th or later):
```
cma=64M
```
This will override the 256M allocation requested by the overlay.

bobodeng520, 2015-11-30:
Now it can boot, but it still hang:
```javascript
cma: Failed to reserve 64MiB
....
...
Kernel command line: cma=256M@512M ....
...
...
systemd[1]: Mounting FUSE Control File System...
```

I found that my board is rpi B Rev 2, but it still load bcm2708-rpi-b.dtb, is this right?

wuyuehang, 2015-11-30:
hi @pelwell @bobodeng520 
are those dtb blobs used by kernel really loaded from SD?
i patch the __bcm2709-rpi-2-b.dts:7:	model = "Modifyme RPi 2";__ but when booting up, the kernel
stay still with this
```
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 256 MiB at 0x20000000
[    0.000000] Memory policy: Data cache writealloc
```

refer to source code, i think it should be updated to __Modifyme RPi 2__. do i miss something?
```c
717const void * __init of_flat_dt_match_machine(const void *default_match,
718		const void * (*get_next_compat)(const char * const**))
719{
720	const void *data = NULL;
721	const void *best_data = default_match;
722	const char *const *compat;
723	unsigned long dt_root;
724	unsigned int best_score = ~1, score = 0;
725
726	dt_root = of_get_flat_dt_root();
727	while ((data = get_next_compat(&compat))) {
728		score = of_flat_dt_match(dt_root, compat);
729		if (score > 0 && score < best_score) {
730			best_data = data;
731			best_score = score;
732		}
733	}
734	if (!best_data) {
735		const char *prop;
736		int size;
737
738		pr_err("\n unrecognized device tree list:\n[ ");
739
740		prop = of_get_flat_dt_prop(dt_root, "compatible", &size);
741		if (prop) {
742			while (size > 0) {
743				printk("'%s' ", prop);
744				size -= strlen(prop) + 1;
745				prop += strlen(prop) + 1;
746			}
747		}
748		printk("]\n\n");
749		return NULL;
750	}
751
752	pr_info("Machine model: %s\n", of_flat_dt_get_machine_name());
753
754	return best_data;
755}
```

pelwell, 2015-11-30:
@bobodeng520 Try decreasing your gpu_mem in your config.txt, or set it to something small like 32 (M). You might even get away with 16, but then you won't be able to use a camera.

@wuyuehang Yes 0 the .dtbs are loaded from the SD Card. The firmware overwrites the model property with a more specific string including the revision number.

wuyuehang, 2015-11-30:
> Yes 0 the .dtbs are loaded from the SD Card. The firmware overwrites the model property with a more specific string including the revision number.
ok i see, according to @pelwell 's explain, @bobodeng520dtb can your dtb blobs are overwriten by firmware too? seems cmdline arguments are cma=256M@512M?

pelwell, 2015-11-30:
@wueyuehang Thanks for your interest. I wrote most of the Device Tree support in the firmware, so I know that using the latest firmware settings in cmdline.txt will override anything added by DT overlays.

bobodeng520, 2015-11-30:
@pelwell @wuyuehang, However decrease memory to 32, 16, there still cannot reserve memory, the kernel command line will always be 256M@512M, and hang in the same place;
Can you help me out of this?

pelwell, 2015-11-30:
What does "vcgencmd version" show? I think your firmware may need an update.

pelwell, 2015-11-30:
And be sure to check the end of the kernel command - that is where entries from cmdline.txt will appear, and because they are processed after the earlier ones they will take precedence.

wuyuehang, 2015-11-30:
@bobodeng520 maybe revert firmware before the following commit, give a try, not sure whether it will help.
[firmware: arm_loader: Apply firmware DT settings after the overlays](https://github.com/raspberrypi/firmware/commit/f49cc9ea91496ccf3425d80279ed586ea6754784)
 
> I wrote most of the Device Tree support in the firmware, so I know that using the latest firmware settings in cmdline.txt will override anything added by DT overlays.

@pelwell i doesn't doubt the firmware overwriting behavior until saw this issue and modify its name. before that i'm guessing it's falling back to atags without dtb.

bobodeng520, 2015-12-01:
>What does "vcgencmd version" show? I think your firmware may need an update.

```
Nov 18 2015 15:43:45
Copyright (c) 2012 Broadcom
version 0c9af7cde38e223c95c3404f468db8fc5004495c (clean) (release)
```
@pelwell My firmware is up-to-date.

pelwell, 2015-12-01:
Reading back through the thread, it is clear that your addition to cmdline.txt is having an effect. Without it, the Pi doesn't get past "booting the kernel", but with it you get a failure later on. You will still see cma=256M@512M, but it will be before "cma=64M" which is the important thing.

Try decreasing the CMA size - "cma=32M" for example - until it boots.

bobodeng520, 2015-12-01:
@pelwell  I have try this:
After comment vc4-kms-v3d:
```
#dtoverlay=vc4-kms-v3d
```
The memory can allocated:
```
cma: Reserved 16 MiB at 0x1b000000
```
And the Kernel command line has nothing about cma=256M@512M,

But if open vc4-kms-v3d:
```
cma: Failed to reserve 16 MiB
```
And it will hang

pelwell, 2015-12-01:
What is your gpu_mem set to? Can you post the entire boot log?

bobodeng520, 2015-12-01:
After setting gpu_mem=16 in config.txt, cma=8M in cmdline.txt
the board can boot up, and load the vc4 driver;
But the net cannot work

pelwell, 2015-12-01:
Without a boot log I can't be sure of the RAM size of your Pi, but it is likely to be 256M. I think you are going to have to accept that the new VC4 driver isn't going to work for you unless you upgrade to a Pi 2, and even then it is still a work-in-progress.

bobodeng520, 2015-12-01:
@pelwell  Here is my boot log(gpu_mem=16, cma=8M):
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.2.6-2708+ (root@milesdeng) (gcc version 4.7.1 20120402 (prerelease) (crosstool-NG 1.15.2) ) #2 PREEMPT 5
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model B Rev 2
[    0.000000] cma: Failed to reserve 16 MiB
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 126976
[    0.000000] free_area_init_node: node 0, pgdat c08a37cc, node_mem_map deb9b000
[    0.000000]   Normal zone: 1116 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 126976 pages, LIFO batch:31
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 125860
[    0.000000] Kernel command line: cma=256M@512M dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=t
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 493328K/507904K available (6207K kernel code, 496K rwdata, 1784K rodata, 356K init, 724K bss, 14576K reserved, )
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xdf800000 - 0xff000000   ( 504 MB)
    lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc07d6180   (7993 kB)
      .init : 0xc07d7000 - 0xc0830000   ( 356 kB)
      .data : 0xc0830000 - 0xc08ac224   ( 497 kB)
       .bss : 0xc08ac224 - 0xc0961608   ( 725 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000]  Additional per-CPU info printed with stalls.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:522
[    0.000033] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000080] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000181] bcm2835: system timer (irq = 36)
[    0.000500] Console: colour dummy device 80x30
[    0.002240] console [tty1] enabled
[    0.002307] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060401] pid_max: default: 32768 minimum: 301
[    0.060789] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.060882] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061991] Initializing cgroup subsys blkio
[    0.062099] Initializing cgroup subsys memory
[    0.062186] Initializing cgroup subsys devices
[    0.062265] Initializing cgroup subsys freezer
[    0.062341] Initializing cgroup subsys net_cls
[    0.062474] CPU: Testing write buffer coherency: ok
[    0.062612] ftrace: allocating 20117 entries in 59 pages
[    0.173241] Setting up static identity map for 0x81c0 - 0x81f8
[    0.175962] devtmpfs: initialized
[    0.184848] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.185302] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.186186] pinctrl core: initialized pinctrl subsystem
[    0.187096] NET: Registered protocol family 16
[    0.192392] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.195430] bcm2708.uart_clock = 3000000
[    0.212086] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.212159] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.212352] Serial: AMBA PL011 UART driver
[    0.212850] 20201000.uart: ttyAMA0 at MMIO 0x20201000 (irq = 90, base_baud = 0) is a PL011 rev2
[    1.357984] console [ttyAMA0] enabled
[    1.362467] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[... 218 lines cut ...]
[    9.197052] [<c0398d1c>] (bus_for_each_dev) from [<c039a1f4>] (driver_attach+0x28/0x30)
[    9.197079] [<c039a1f4>] (driver_attach) from [<c0399f50>] (bus_add_driver+0xf8/0x208)
[    9.197106] [<c0399f50>] (bus_add_driver) from [<c039b124>] (driver_register+0x88/0x104)
[    9.197134] [<c039b124>] (driver_register) from [<c039bf64>] (__platform_driver_register+0x58/0x6c)
[    9.197210] [<c039bf64>] (__platform_driver_register) from [<bf1cc05c>] (vc4_drm_register+0x5c/0x74 [vc4])
[    9.197291] [<bf1cc05c>] (vc4_drm_register [vc4]) from [<c0009768>] (do_one_initcall+0x90/0x1e8)
[    9.197338] [<c0009768>] (do_one_initcall) from [<c008c004>] (do_init_module+0x6c/0x1e4)
[    9.197372] [<c008c004>] (do_init_module) from [<c008d9c0>] (load_module+0x1778/0x1e9c)
[    9.197406] [<c008d9c0>] (load_module) from [<c008e288>] (SyS_finit_module+0x74/0x80)
[    9.197442] [<c008e288>] (SyS_finit_module) from [<c000fbc0>] (ret_fast_syscall+0x0/0x54)
[    9.197460] ---[ end trace 0161c36c13415648 ]---
[    9.197623] vc4-drm soc:gpu@7e4c0000: bound 20902000.hdmi (ops vc4_hdmi_ops [vc4])
[... cut ...]
[    9.274781] vc4-drm soc:gpu@7e4c0000: failed to bind 20c00000.v3d (ops vc4_v3d_ops [vc4]): -22
[    9.297330] vc4-drm soc:gpu@7e4c0000: master bind failed: -22
[    9.297402] vc4-drm: probe of soc:gpu@7e4c0000 failed with error -22
[   10.877294] systemd-journald[89]: Received request to flush runtime journal from PID 1
[   11.846981] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   11.847847] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   17.062448] Adding 102396k swap on /var/swap.  Priority:-1 extents:6 across:573436k SSFS
```

pelwell, 2015-12-01:
From that log I can see that you have one of the newer 512MB B's, and that the initial 16MB CMA allocation is failing. At the moment this project is being developed for Pi 2 - although the intention is that it should run on a Pi 1 eventually, that isn't a priority at the moment.

ED6E0F17, 2015-12-01:
Your CMA allocation will be reduced to 32Mb less than GPU mem, so you need more GPU and CMA. I am using GPU 128, CMA 128 on a model A, and getting 96Mb CMA (still available to linux)

Ruffio, 2016-08-17:
@bobodeng520 has your issue been resolved? If so, please close this issue. Thanks.

closed 2017-05-18 by JamesH65