raspberrypi/linux #1211: UVC ioctl() freeze after device disconnection

flydutch, 2015-12-02:
Using last kernel provided by raspbian jessie repository:
```
Linux 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015
```
it happens that if a USB camera is disconnected while a thread is grabbing frames, using `ioctl(VIDIOC_DQBUF)`, the `ioctl()` never returns (the device is opened in blocking mode, i.e. without the `O_NONBLOCK` flag.

Looking at the UVC linux module the blocking `ioctl(VIDIOC_DQBUF)` should return:
 * when a buffer is available
 * when the streaming is turned off (`ioctl(VIDIOC_STREAMOFF)`).

The problem is that turning off the stream from another thread, when the USB disconnection is detected, doesn't work because the `ioctl(VIDIOC_STREAMOFF)` fails to shutdown the stream and return ENODEV error (i.e. device not available).

In other words seems that there is no way to return from the `ioctl(VIDIOC_DQBUFF)`, unmapping the mmapped buffers and closing the device file doesn't help.

pelwell, 2015-12-02:
Are you saying this used to work, i.e. is this a regression?

Either way, this sounds like a problem in the upstream kernel and should be reported there. For drivers/media/usb/uvc/*, the maintainer is Laurent Pinchart (<laurent.pinchart@ideasonboard.com>).

flydutch, 2015-12-02:
Ok, thank you.
(I don't know if it's a regression or a new bug).

closed 2015-12-02 by flydutch