raspberrypi/linux #1214: Raspberry Pi Zero does not reboot or shutdown

incyi, 2015-12-06:
The Raspberry Pi Zero V1.2 does not reboot ro shutdown. Only with -f (forced) works.
pi@raspi:~ $ uname -a
Linux raspi 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015 armv6l GNU/Linux

pi@raspi:~ $ sudo reboot
Failed to talk to init daemon.
pi@raspi:~ $ sudo shutdown -h now
Failed to talk to init daemon.
pi@raspi:~ $ sudo init 6
Couldn't find an alternative telinit implementation to spawn.

popcornmix, 2015-12-06:
Does it do this with a clean raspbian install?

incyi, 2015-12-06:
I will test this and return with feedback.

incyi, 2015-12-07:
So best option is to format the card and start again with a fresh image ?

popcornmix, 2015-12-07:
I suspect that will quickest. I don't know how your image is broken so I don't know what is involved in fixing it.

incyi, 2015-12-07:
Thanks. I start with a fresh image.

closed 2015-12-07 by incyi

heypete, 2016-09-06:
I am experiencing the same issue with a Raspberry Pi 3 Model B running jessie. uname -r returns 4.4.13-v7+ for the kernel. I've also updated to 4.4.19-v7+ using rpi-update, to no avail.

Reinstalling from a fresh image is possible, but a massive pain due to the resulting downtime and need to reconfigure everything. Reinstalling also would not resolve the root cause of whatever caused this issue to originally occur.

Any recommendations on diagnosing the issue? I'd be happy to provide logs, if desired. Just let me know what is needed.

heypete, 2016-09-06:
Please disregard my previous message: I've resolved my particular issue. Sorry for the noise.

The issue I was facing is that I had, in the interests of reducing microSD card wear, mounted /var/run as tmpfs. By removing the line in /etc/fstab that mounted /var/run as tmpfs, everything works fine again.

More details are at https://lists.alioth.debian.org/pipermail/pkg-systemd-maintainers/2014-September/003598.html

efficks, 2016-10-19:
@heypete Thank you that does the thrick.

HOD42, 2017-12-08:
@heypete Thanks for posting this, I had done similar and couldn't fathom why reboot stopped working.