raspberrypi/linux #1215: Could not load fbtft module on boot

incyi, 2015-12-06:
It didn't worked to load the module fbtft from notro on boot. Everytime when the Raspberry Pi Zero is started I run this command to load it:
sudo modprobe fbtft_device custom name=fb_ili9341  gpios=reset:25,dc:24,led:18 speed=16000000 rotate=90 bgr=1

And then run to show the console:
con2fbmap 1 1

notro, 2015-12-06:
Have you tried this: https://github.com/notro/fbtft/wiki#make-it-permanent-debian-jessie-arch-linux-systemd

incyi, 2015-12-06:
This worked ! Thanks. Now I need to figure out to load the console on the tft.

dlech, 2015-12-06:
See https://github.com/notro/fbtft/wiki/Boot-console

incyi, 2015-12-06:
Yes that did the work !

Problem solved. I have a working lcd now which is autostarting from boot ! Great !

popcornmix, 2015-12-07:
Okay to close?

incyi, 2015-12-07:
Yes. Okay to close.

closed 2015-12-07 by incyi