raspberrypi/linux #1217: rpi2 running omxplayer repeatedly ok on linux-4.2.6 but hangs on linux-4.4.0-rc3

jeffersonchua, 2015-12-07:
I'm running omxplayer on Raspberry Pi 2 and it worked fine on Linux-4.2.6.

But, on Linux-4.4.0-rc3, running omxplayer repeatedly will eventually hangs the video.

I've attached  my .configs. It's basically the same config that works on 4.2.6 copied over to 4.4.0-rc3.

Any pointers?

Thanks,
Jeff.

[config-4.2.6.TXT](https://github.com/raspberrypi/linux/files/54298/config-4.2.6.TXT)
[config-4.4.0-rc3.TXT](https://github.com/raspberrypi/linux/files/54300/config-4.4.0-rc3.TXT)

popcornmix, 2015-12-07:
I think you'll have to start with bcm2709_defconfig and test if that works.
Need to determine first if it's the 4.4 tree or your config.

Quick guess - CONFIG_CPU_SW_DOMAIN_PAN must not be set.

jeffersonchua, 2015-12-08:
On Tue, Dec 8, 2015 at 2:21 AM, popcornmix <notifications@github.com> wrote:

> I think you'll have to start with bcm2709_defconfig and test if that works.
> Need to determine first if it's the 4.4 tree or your config.
>
> Quick guess - CONFIG_CPU_SW_DOMAIN_PAN must not be set.
>

Amazing! omxplayer works now with CONFIG_CPU_SW_DOMAIN_PAN unset. That was
the cause of the problem. This was introduced with 4.3 ... and that
explains why omxplayer hangs after 4.3

Thank you so much! I'm back on trace using the latest kernel!

Jeff.

closed 2015-12-08 by popcornmix