raspberrypi/linux #1219: system freeze while unplug r8152  usb2network card

huangjiansword, 2015-12-08:
hi there,
  I have a raspberry pi model A running linux 3.18.7+, I found a issue about r8152 driver in the image. 

steps below:
1, attach a usb hub on the pi's usb port 
2, attach a usb2network card (chip ID is 0bda:8152 Realtek Semiconductor Corp.) device on the usb hub
3, connect network wire to r8152 network card and a router
4, run ifconfig to monitor eth0's status
5, when eth0's status changed to running or got ip address, unplug the r8152 device, the raspberry pi has a large probability to stop response
you may try serveral times to reappear this issue.

I try to use another raspberry image(4.1,released 2015-09-25) and r8152, the same issue remains.

I use several usb2network cards(chip ID is 0b95:772b ASIX Electronics Corp. and  ), the same operation, It works.

BTW, I try the latest driver 2.05.0 from realteck website, problem remains.

My power supply is 5V 5A( for it will also supply for my own customized usb hub) , the power is good enough for raspberry pi.

I don't think the freeze is related with the usb hub, I tried many hub chip such as 05e3:0608(GL850G) or 05e3:0610(GL852G) from Genesys logic or  1a40:0101 from Terminus Technology Inc.

And I can not get any related log from dmesg or  other log.

If attached the r8152 directly with the Pi's USB port, no such issue.

I have contacted with the engineer of REALTEK,  the conclusion from REALTEK is:
"If hub is involved to the issue, it could be a timing compatibility issue between usb host controller and genesis chip."

How can I fix the issue?
Thank you in advance!

huangjiansword, 2015-12-08:
I can reappear the issue with Raspberry A+ and Raspberry B with image 4.1.7 and 3.18.7
Linux raspberrypi 4.1.7+ #817 PREEMPT Sat Sep 19 15:25:36 BST 2015 armv6l GNU/Li nux

Linux raspberrypi 3.18.7+ #4 PREEMPT Mon May 11 15:39:01 CST 2015 armv6l GNU/Linux

popcornmix, 2015-12-08:
Do you get a kernel panic/backtrace output to display?

huangjiansword, 2015-12-08:
no, I got nothing display and nothing valuable information in demsg or messages or other log, the system just lost response.

popcornmix, 2015-12-08:
Can you explain exactly what you see?
Are you starting from initial console (i.e. not desktop).
Are you interacting with keyboard/display?
After the crash does the display appears exactly as it did before or does it blank?

huangjiansword, 2015-12-08:
this is the log from messages, it records the crash process, but no any information related with the crash reason

Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Initializing cgroup subsys cpuset
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Initializing cgroup subsys cpu
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Initializing cgroup subsys cpuacct
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Linux version 4.1.7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #817 PREEMPT Sat Sep 19 15:25:36 BST 2015
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Machine model: Raspberry Pi Model A Plus Rev 1.1
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] cma: Reserved 8 MiB at 0x0b800000
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Memory policy: Data cache writeback
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 48720
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708.boardrev=0x12 bcm2708.serial=0xf992ca04 smsc95xx.macaddr=B8:27:EB:92:CA:04 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Memory: 176864K/196608K available (6132K kernel code, 490K rwdata, 1780K rodata, 348K init, 720K bss, 11552K reserved, 8192K cma-reserved)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Virtual kernel memory layout:
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]     vmalloc : 0xcc800000 - 0xff000000   ( 808 MB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]     lowmem  : 0xc0000000 - 0xcc000000   ( 192 MB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]       .text : 0xc0008000 - 0xc07c23ec   (7913 kB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]       .init : 0xc07c3000 - 0xc081a000   ( 348 kB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]       .data : 0xc081a000 - 0xc0894824   ( 491 kB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]        .bss : 0xc0894824 - 0xc09489b0   ( 721 kB)
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] Preemptible hierarchical RCU implementation.
Dec  8 07:17:32 raspberrypi kernel: [    0.000000]      Additional per-CPU info printed with stalls.
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] NR_IRQS:522
Dec  8 07:17:32 raspberrypi kernel: [    0.000000] clocksource stc: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
Dec  8 07:17:32 raspberrypi kernel: [    0.000014] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
Dec  8 07:17:32 raspberrypi kernel: [    0.000056] Switching to timer-based delay loop, resolution 1000ns
Dec  8 07:17:32 raspberrypi kernel: [    0.000337] Console: colour dummy device 80x30
Dec  8 07:17:32 raspberrypi kernel: [    0.001762] console [tty1] enabled
Dec  8 07:17:32 raspberrypi kernel: [    0.001828] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
Dec  8 07:17:32 raspberrypi kernel: [    0.001927] pid_max: default: 32768 minimum: 301
Dec  8 07:17:32 raspberrypi kernel: [    0.002288] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Dec  8 07:17:32 raspberrypi kernel: [    0.002369] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Dec  8 07:17:32 raspberrypi kernel: [    0.003448] Initializing cgroup subsys blkio
Dec  8 07:17:32 raspberrypi kernel: [    0.003547] Initializing cgroup subsys memory
Dec  8 07:17:32 raspberrypi kernel: [    0.003623] Initializing cgroup subsys devices
Dec  8 07:17:32 raspberrypi kernel: [    0.003690] Initializing cgroup subsys freezer
Dec  8 07:17:32 raspberrypi kernel: [    0.003755] Initializing cgroup subsys net_cls
Dec  8 07:17:32 raspberrypi kernel: [    0.003878] CPU: Testing write buffer coherency: ok
Dec  8 07:17:32 raspberrypi kernel: [    0.004005] ftrace: allocating 19773 entries in 58 pages
Dec  8 07:17:32 raspberrypi kernel: [    0.108885] Setting up static identity map for 0x81c0 - 0x81f8
[... cut ...]
Dec  8 12:11:04 raspberrypi kernel: [   47.838937] cfg80211: Calling CRDA to update world regulatory domain
Dec  8 12:11:07 raspberrypi kernel: [   50.998975] cfg80211: Calling CRDA to update world regulatory domain
Dec  8 12:11:10 raspberrypi kernel: [   54.159074] cfg80211: Calling CRDA to update world regulatory domain
Dec  8 12:11:13 raspberrypi kernel: [   57.318918] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
Dec  8 12:11:17 raspberrypi kernel: [   61.111305] usb 1-1.4.1: USB disconnect, device number 4

popcornmix, 2015-12-08:
Does the pi hang and need rebooting before you see the messages? How are you viewing them?

huangjiansword, 2015-12-08:
Are you starting from initial console (i.e. not desktop).    
From desktop, but it hand also starting from console

Are you interacting with keyboard/display?    
Yes, with hdmi display and keyboard

After the crash does the display appears exactly as it did before or does it blank?  
Appears exactly as it did

Does the pi hang and need rebooting before you see the messages? 
Yes,  the pi hang and need rebooting, then I got the messages from /var/log/messages

How are you viewing them?
tail /var/log/message -n 300

popcornmix, 2015-12-08:
Can you add ``dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0`` to end of cmdline.txt and report if any change in behaviour?

huangjiansword, 2015-12-08:
I just tried it, but there is no use with dwc_otg.fiq_enable=0 dwc_otg.fiq_fsm_enable=0.

Also, I tried to add max_usb_current=1 to /boot/config.txt,  issue remains.

popcornmix, 2015-12-08:
Just to confirm, you should see:
```
[    1.882870] dwc_otg: FIQ disabled
[    1.882888] dwc_otg: FIQ split-transaction FSM disabled
```
in log.

huangjiansword, 2015-12-08:
Yes, confirmed

pi@raspberrypi ~ $ dmesg | grep FIQ
[    1.635657] dwc_otg: FIQ disabled
[    1.635698] dwc_otg: FIQ split-transaction FSM disabled

popcornmix, 2015-12-08:
@P33M any ideas?

P33M, 2015-12-08:
Where can I buy one of these devices?

huangjiansword, 2015-12-09:
could you give me your address, so I can mail you some r8152s, my mailbox is huangjiansword@hotmail.com

huangjiansword, 2015-12-10:
Another finding, I try the same device with RPI2, it works no hang no crash!

P33M, 2016-01-05:
I can replicate the issue. 1 in 5 hot-unplugs on a Pi 1 B+ cause a crash. So far 20+ hot-unplugs on a Pi 2B = no crash.

huangjiansword, 2016-01-05:
Yes, the same phenomena as me.

huangjiansword, 2016-01-11:
@P33M @popcornmix, any suggestion about this issue? I think maybe 8152 driver should be fixed, but  the guys from realtek said it's not 8152 issue but host controller's issue. but I can not reproduce the issue if I unload the 8152 driver.

P33M, 2016-01-11:
What's curious is the fact that it crashes on a Pi 1 (single-processor) but apparently doesn't on Pi 2 (multiprocessor). The Pi just hard-freezes without printing a stacktrace which is most unusual.

huangjiansword, 2016-01-12:
What's the difference between Pi1 and Pi2 concerning USB controller?On what condition will the core print no any stacktrace or dmesg message?

huangjiansword, 2016-03-07:
On 3.12.20+, unplug 8152 will not freeze system, but on 3.18.7+, the issue appears, I think maybe some hub or host controller's modification cause this issue.

huangjiansword, 2016-03-24:
@popcornmix @P33M  I also tried with Jessie and r8152, the same issue remains.

mlord, 2016-10-26:
The r8152 driver has been broken since (approx) 3.6.16,
when support was added for hardware rx checksum on newer chip versions.
Symptoms include random segfaults and silent data corruption over NFS.

This does not work on the VER_02 dongle I have here
when used with a slow embedded system CPU.

So, disable hardware rx checksum for VER_02, and fix
an obvious coding error for IPV6 checksums in the same function.

Patch submitted upstream for this today:  https://lkml.org/lkml/2016/10/26/1167

popcornmix, 2016-10-27:
@mlord thanks for info. Let us know when the patch is merged upstream and we can cherry pick it.

P33M, 2017-05-04:
Fixed upstream in -stable: https://lkml.org/lkml/2017/1/11/759

closed 2017-05-04 by P33M