raspberrypi/linux #1221: No Multicore Scheduler support?

robingroppe, 2015-12-08:
Just checked the kernel config again and realized that no Multicore Scheduler Support is in use.
Is there a specific reason for this?

P33M, 2015-12-08:
Are you talking about CONFIG_SCHED_SMT?

```
pi@raspberrypi:~$ zgrep SCHED /proc/config.gz
CONFIG_GENERIC_SCHED_CLOCK=y
CONFIG_CGROUP_SCHED=y
CONFIG_FAIR_GROUP_SCHED=y
# CONFIG_RT_GROUP_SCHED is not set
CONFIG_SCHED_AUTOGROUP=y
CONFIG_IOSCHED_NOOP=y
CONFIG_IOSCHED_DEADLINE=y
CONFIG_IOSCHED_CFQ=y
CONFIG_CFQ_GROUP_IOSCHED=y
CONFIG_DEFAULT_IOSCHED="cfq"
# CONFIG_SCHED_MC is not set
# CONFIG_SCHED_SMT is not set
CONFIG_SCHED_HRTICK=y
CONFIG_NET_SCHED=y
CONFIG_SCHED_DEBUG=y
# CONFIG_SCHEDSTATS is not set
# CONFIG_SCHED_STACK_END_CHECK is not set
CONFIG_SCHED_TRACER=y
```

popcornmix, 2015-12-08:
It's CONFIG_SCHED_MC. The help text says:

> Multi-core scheduler support improves the CPU scheduler's decision making when dealing with multi-core CPU chips at a cost of slightly increased overhead in some places. If unsure say N here.

Someone sees a 10% reduction in performance here:
https://www.redhat.com/archives/dm-devel/2010-August/msg00164.html

So, again I think need careful analysis before enabling.

robingroppe, 2015-12-08:
The question would be if it is in use at all when there is only one cpu in the system like the Pi 1 or Zero.

P33M, 2015-12-08:
We have two different configs for 2835 and 2836 (bcmrpi_defconfig and bcm2709_defconfig).

I don't think Sched_MC will have any positive benefit. It makes the kernel aware of CPU locality (cache hierarchy), but all our A7s share an L2 cache.

https://wiki.linaro.org/WorkingGroups/PowerManagement/Archives/SchedMc

robingroppe, 2015-12-08:
Okay. Thank you for making that clear.

closed 2015-12-08 by robingroppe