raspberrypi/linux #1222: bcm2835-thermal on 4.4: warning: initialization from incompatible pointer type

notro, 2015-12-08:
```
drivers/thermal/bcm2835-thermal.c:80:2: warning: initialization from incompatible pointer type [enabled by default]
  .get_temp = bcm2835_thermal_get_temp,
  ^
drivers/thermal/bcm2835-thermal.c:80:2: warning: (near initialization for ‘ops.get_temp’) [enabled by default]
drivers/thermal/bcm2835-thermal.c:81:2: warning: initialization from incompatible pointer type [enabled by default]
  .get_trip_temp = bcm2835_thermal_get_max_temp,
  ^
drivers/thermal/bcm2835-thermal.c:81:2: warning: (near initialization for ‘ops.get_trip_temp’) [enabled by default]
```
Temperature is now _int_ instead of _unsigned long_: 17e8351a77397e8a83727eb17e3a3e9b8ab5257a

I'm still tracking down the lockup/pagefault problem so I can't make a PR.

popcornmix, 2015-12-08:
https://github.com/raspberrypi/linux/commit/b3f74d7bc6c0e3ce01beac5a034fa32521a77509

notro, 2015-12-08:
I wrote 4.4 but I'm actually on 4.3.
Since 4.4 is the next LTS, will that be the next version bump?

popcornmix, 2015-12-08:
We are certainly aiming for 4.4. We might possibly bump to 4.2 and 4.3 briefly on rpi-update as they may be useful data points for regression testing.
I can cherry-pick this to 4.3.

closed 2015-12-08 by notro