raspberrypi/linux #1223: RPi2 hangs, possible lan/usb driver issue

mobitinker, 2015-12-09:
We are currently developing an environmental monitoring hub using the RPi 2. We’ve installed at an alpha site and are having lots of the devices, 10 of 20, showing network lockups. The device intermittently locks and the network lights go out. The only remedy is power cycling. When looking at syslog after a hard reset it appears that CPU dies as well.

From our googling it appears this may a known problem in the LAN/USB chip caused by low voltage events below 4.75v or data overruns. Though the problem seems common the information available is mostly anecdotal about things to try rather than characterizing the causes of the problem.

We’ve  put a multimeter on the POE splitter we use at each connection point. The testing showed one connector that can be finicky. When all connections are solid, we don’t see voltages below 4.85 or so. We’ve not yet traced a crash to a voltage drop.

Our issue looks similar to this reported one that seems to have been closed without a fix. https://github.com/raspberrypi/firmware/issues/247.

We configure our devices with Raspbian Wheezy. We have a static ip address and use the network connector. We have a TPLink POE splitter at the Pi to separate out power and network. One USB port is used for a BlueTooth dongle. It is at the end of a 6” extension cable. The device is typically in a pretty tight space in electrical junction box in the wall. We’ve seen the problem in that case but also when the devices are sitting on the bench. We’re not overclocking.

Steps we are taking:

- Monitoring tcp traffic with tcpflow to see if certain packets might be linked to the lock up. Wireshark does not show anything anomalous at the time of crash.
- Testing all cables for quality. A 30% sample of our runs from switch to Pi show good cables.
- Monitoring power supply with a scope. We’ve not caught the problem occurring yet while monitoring.
- Creating a watchdog circuit that will cause the pi to power cycle itself if the network can’t be accessed.

The last item will treat the symptom but is less than ideal because of downtime during reboots.

We’d really like to understand the precise cause(s) so that we can reproduce the lockup in testing our watchdog, and hopefully, decrease the times it is needed to keep our units alive. We’re happy to collaborate with others in finding the cause.

P33M, 2015-12-09:
Do your PoE splitters pass the common-mode voltage through to the Pi's ethernet port? I.e. is there no isolation between splitter input/ethernet output?

mobitinker, 2015-12-09:
@P33M, thanks! That turned out to be an interesting question. We are using the TP-Link TL-POE10R PoE device, non-isolated, described here:
http://blog.zencoffee.org/2015/11/tp-link-tl-poe10r-poe-splitter-ground-loop-issues and in this video: https://www.youtube.com/watch?v=TYd3RtooFxY.
Note that we observe our hangs after the device has been running some time without any handling or movement near them. We'll replace them but I'm wondering if the problem will go away.

P33M, 2015-12-09:
There's a product information notice associated with the LAN controller as used on all B models. SMSC have noted issues when PoE-enabled devices have been used in conjunction with their chips.

http://ww1.microchip.com/downloads/en/AppNotes/en562768.pdf

In short, transients associated with hot plug/unplug to PoE-enabled ports can degrade and eventually cause malfunction of the ethernet port.

I'd be interested to know if the faults follow the Pis that exhibit them - does a site/installation with a faulty Pi immediately become trouble-free if the device is replaced?

mobitinker, 2015-12-12:
Hi and thanks! In our case we are not hot-plugging the ethernet cable more than once or twice before leaving the device to run headless. We can see if problem follows devices. 

I could not see in the app note where it said what the effects of cable transients would be on the lan/usb chip, or that it would lead to the hang behavior we are seeing. Do you think it could? 

If transients are the problem, could they arise without hot plugging? Some of the POE switches live next to circuit breaker boxes.

Apologies if any of this sounds naive. I'm a software guy!

P33M, 2015-12-14:
Unless there's a particularly bad arrangement of power wiring run parallel next to the Cat.5 Ethernet cable, I wouldn't think that breaker switching would have this effect

Ruffio, 2016-08-17:
@mobitinker has your issue been resolved? If so, please close this issue. Thanks.

closed 2016-08-17 by mobitinker