raspberrypi/linux #1225: rootless access to gpiomem?

sinadogru, 2015-12-11:
According to issue #1112, I suppose to access gpiomem without root access. But when I open(2) it returns -1 and perror prints, "Permission denied". Also,

pi@raspberrypi ~ $ ls -l /dev/gpiomem 
crw------- 1 root root 244, 0 Jan  1  1970 /dev/gpiomem
pi@raspberrypi ~ $ ls -l /dev/mem
crw-r----T 1 root kmem 1, 1 Jan  1  1970 /dev/mem

I use recently updated Rasbian-Wheezy on Rpi 2 model B,

Here is some information,

pi@raspberrypi ~ $ uname -a
Linux raspberrypi 4.1.13-v7+ #827 SMP PREEMPT Tue Dec 8 20:29:06 GMT 2015 armv7l GNU/Linux

popcornmix, 2015-12-11:
This is really a distribution question rather than a kernel question.
It is supported with Jessie, but I'm not sure it is for an older wheezy install.

On Jessie there is a udev rule:
```
$ cat /etc/udev/rules.d/99-com.rules
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c 'chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio; chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio; chown -R root:gpio /sys/devices/platform/soc/*.gpio/gpio && chmod -R 770 /sys/devices/platform/soc/*.gpio/gpio'"
SUBSYSTEM=="input", GROUP="input", MODE="0660"
SUBSYSTEM=="i2c-dev", GROUP="i2c", MODE="0660"
SUBSYSTEM=="spidev", GROUP="spi", MODE="0660"
SUBSYSTEM=="bcm2835-gpiomem", GROUP="gpio", MODE="0660"
```
which you may be able to manually add.

Ruffio, 2016-08-17:
@sinadogru has your issue been resolved? If so, please close this issue. Thanks.

sinadogru, 2016-08-18:
I am sorry, yes with Jessie it is solved. Thanks for helping!

closed 2016-08-18 by sinadogru

jlinkels, 2017-03-29:
I am a bit lost now. I just dist-upgraded my PI to Jessie to get /dev/gpiomem. However the permissions on /dev/gpiomem were incorrect. The 99-com rules have a different contents from the contents showed by **popcornmix**.
After adding/changing the rules as stated by popcornmix, the permissions of /dev/gpiomem **are** correct. By changing the contents of 99-com.rules the problem is solved in Jessie.
So what I am actually wondering, why is the contents of 99-com.rules in Jessie different? This is quite an old post and I did the dist-upgrade to Jessie on 2017-03-27. Is this a regression error?

popcornmix, 2017-03-29:
Can you post the contents of the non-working ``/etc/udev/rules.d/99-com.rules``?

jlinkels, 2017-03-29:
Hi popcornmix,
This is the 99-com.rules as installed by Jessie:
#SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c 'chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio; chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio'"
SUBSYSTEM=="input", GROUP="input", MODE="0660"
SUBSYSTEM=="i2c-dev", GROUP="i2c", MODE="0660"
SUBSYSTEM=="spidev", GROUP="spi", MODE="0660"
`
Now there could be also a problem in the naming of the folders in /sys/devices/virtual/. As you see in the udev rules, the rules expect the directory  /sys/devices/virtual/gpio. But that directory does not exist. This is the (partial contents of the folder:
`jlinkels@power-rn:~$ ll /sys/devices/virtual/
total 0
drwxr-xr-x  3 root root 0 Mar 29 01:26 bcm2708_vcio
drwxr-xr-x  3 root root 0 Mar 29 01:26 **bcm2835-gpiomem**
drwxr-xr-x 27 root root 0 Mar 29 01:26 bdi
drwxr-xr-x 26 root root 0 Mar 29 01:26 block
drwxr-xr-x  3 root root 0 Mar 29 01:26 graphics
drwxr-xr-x  3 root root 0 Mar 29 01:26 input
`
In the udev rules that you posted you correctly refer to the directory bcm2835-gpiomem.

If it makes any difference, I did a dist-upgrade from Wheezy to Jessie on the PI, not a fresh Jessie install.

**Update**: Before submitting the post I checked the 99-com.rules on a Wheezy installation. That file is identical to the contents I posted above. So it seems this file is not updated during the dist-upgrade. Maybe it is as simple as that. The thought that this could be happening did not even cross my mind as Wheezy did not have /dev/gpiomen yet.

popcornmix, 2017-03-29:
I'm afraid updating wheezy to jessie isn't really supported. See [here](https://www.raspberrypi.org/blog/raspbian-jessie-is-here/)
Do you have the problem with a direct jessie install?
Assuming that is fine you have the choice of starting with jessie and copying your own files from the wheezy install, or trying to update the wheezy image yourself when you find things that don't work correctly.

XECDesign, 2017-03-29:
When upgrading, did dpkg ask you whether you wanted to keep the old file or install the new one?

jlinkels, 2017-03-29:
@popcornmix: OK. This was maybe not the smartest thing to do. The reason for **Jessie** was to get the non-root /dev/gpiomem. And the reason for **dist-upgrade** was to see if a dist-upgrade could succesfully be done. And some lazyness as my system is somewhat customized with cron scripts and startup scripts.

@XECDesign: I have kept all the configurations as per default choice of dpkg. Either I have overlooked the question for udev, or I have assumed that  the rules files would have been updated as well.

Let me see if I have another memory card around so I can do a fresh Jessie install. And see if the rules file is correct.

XECDesign, 2017-03-29:
Files in /etc/ are considered user-configurable conffiles. dpkg avoids modifying any conffiles that have been modified by the user. At some point in the past, that file was not managed by a package at all and was added in the image creation process. It's likely that even if you have not touched the file, dpkg considered it a modified file because it was there before the package which installs it was added.

popcornmix, 2017-03-29:
@jlinkels you seem to have a comment character ``#`` at the start if the SUBSYSTEM="gpio" line.
I'm pretty sure that wasn't in the standard wheezy or jessie version so I guess that is a local edit.
That would stop the rule working.

jlinkels, 2017-03-29:
The comment character is my fault. It is not on the Wheezy files. I commented the line before pasting in your line. I corrected it in the post.

jlinkels, 2017-03-29:
A fresh installation of Jessie is correct and sets the correct permissions on gpiomem. I hope other users benefit from this thread.
For the record, this is the contents of 99-com.rules on Jessie:

`SUBSYSTEM=="input", GROUP="input", MODE="0660"
SUBSYSTEM=="i2c-dev", GROUP="i2c", MODE="0660"
SUBSYSTEM=="spidev", GROUP="spi", MODE="0660"
SUBSYSTEM=="bcm2835-gpiomem", GROUP="gpio", MODE="0660"

SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c '\
	chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio;\
	chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio;\
	chown -R root:gpio /sys$devpath && chmod -R 770 /sys$devpath\
'"

KERNEL=="ttyAMA[01]", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $ALIASES/uart0 $ALIASES/serial0; then \
		echo 0;\
	elif cmp -s $ALIASES/uart0 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi\
'", SYMLINK+="serial%c"

KERNEL=="ttyS0", PROGRAM="/bin/sh -c '\
	ALIASES=/proc/device-tree/aliases; \
	if cmp -s $ALIASES/uart1 $ALIASES/serial0; then \
		echo 0; \
	elif cmp -s $ALIASES/uart1 $ALIASES/serial1; then \
		echo 1; \
	else \
		exit 1; \
	fi \
'", SYMLINK+="serial%c"
`