raspberrypi/linux #1226: aplay cannot play imediately after another aplay

ppedro74, 2015-12-12:
Hi,

when running aplay after another aplay the second aplay does not play the sound and takes while to terminate.

example:

aplay sound.wav ; aplay sound.wav

the first sound file plays for 1 second, the second play takes 40 seconds and no sound is reproduced, if you try to do another aplay (plays ok), but if you run another aplay 2 seconds later, the aplay will get locked for 30 or more seconds.

current system:

uname -r => 4.1.13-v7+
lsb_release -a =>  Raspbian GNU/Linux 8.0 (jessie)
aplay --version => 1.0.28
sound.wav => Signed 16 bit Little Endian, Rate 16000 Hz, Mono

i checked in another system (no issues):
uname -r => 3.16.0-43-generic
lsb_release -a => Ubuntu 14.04.2 LTS
aplay --version => 1.0.27.2
sound.wav => Signed 16 bit Little Endian, Rate 16000 Hz, Mono

both systems are up to date.
i've tried different files e.g. /usr/share/scratch/Media/Sounds/Vocals/Singer1.wav and the result is the same, second aplay gets stucked and no sound.

Any ideas ?

Thanks,

popcornmix, 2015-12-13:
Not seeing that here:
```
pi@domnfs:~ $ aplay /usr/share/scratch/Media/Sounds/Vocals/Singer1.wav; aplay /usr/share/scratch/Media/Sounds/Vocals/Singer1.wav
Playing WAVE '/usr/share/scratch/Media/Sounds/Vocals/Singer1.wav' : Signed 16 bit Little Endian, Rate 11025 Hz, Mono
Playing WAVE '/usr/share/scratch/Media/Sounds/Vocals/Singer1.wav' : Signed 16 bit Little Endian, Rate 11025 Hz, Mono
pi@domnfs:~ $ 
```
File played twice and behaved as expected. Same with hdmi or analogue output.
Can you test with a clean raspbian jessie install?

ppedro74, 2015-12-16:
Hi,

after a few tests i track down the offending change.

i started with a clean image, everything works well, then i did:

sudo apt-get install gstreamer1.0 

and the problem appeared  then i tried:

sudo apt-get remove --purge gstreamer1.0 
sudo apt-get autoremove

and the problem is still present, so the question is what is left behind or what is changed and not reverted when installing gstreamer1.0 that makes the audio to get broken.

any help...

popcornmix, 2015-12-16:
Do you have pulseaudio installed?

ppedro74, 2015-12-16:
yes

ppedro74, 2015-12-16:
i just removed the pulseaudio, and the aplay is working again.

ppedro74, 2015-12-16:
i just added again and the issue returns. I didn't installed must be gstreamer1.0 requirement, and it was not removed when gstreamer is removed.

so the question now is the pulseaudio broken or is a configuration issue ?

popcornmix, 2015-12-16:
Not sure. pulseaudio is generally undesirable on the pi as it's quite heavyweight.
I'd suggest avoiding it for now, but I'll add it to the list of things to look into.

Ruffio, 2016-08-17:
@ppedro74 has your issue been resolved? If so, please close this issue. Thanks.

closed 2017-05-18 by JamesH65