raspberrypi/linux #1229: Raspbian Jessie (at least) does not appear to auto-detect screen resolution.

semininja, 2015-12-15:
A stock image will result in a glitchy grey wave slowly scanning down the screen ad infinitum when using an 800x480 monitor as found [here](https://www.adafruit.com/products/2232), but it will function correctly on a 1920x1080 monitor. A Pi that does work with the smaller monitor does not appear to recognize the higher resolution of the larger monitor. I do not recall having had to do anything to configure the second Pi to work with the smaller image, but I may have updated it to Jessie from Wheezy using the smaller screen. Is this a known issue?

FishTest, 2015-12-29:
try to add below lines to /boot/config.txt
hdmi_group=2
hdmi_cvt=800 480 60 0 0 0

popcornmix, 2015-12-29:
800x480 is not a standard CEA or DMT hdmi mode so has never been supported by the Pi without custom config.txt settings.
If you believe this has worked in the past then report output of ``tvservice -s`` on both the working and non-working systems.

Ruffio, 2016-08-17:
@semininja has your issue been resolved? If so, please close this issue. Thanks.

semininja, 2016-08-17:
I haven't been working with my RPi much lately, so I'd forgotten about this thread. It occurs to me to ask: what resolutions are natively supported, and how likely is it that I've changed some config file somewhere on the working Pi which might cause it to not recognize a standard 1920x1080 display?

popcornmix, 2016-08-17:
We support the standard CEA and DMT modes. See: https://www.raspberrypi.org/documentation/configuration/config-txt.md
Report output of:
 ```
tvservice -s
tvservice -m CEA
tvservice -m DMT
```
(e.g. though ssh) with the monitor attached to see what is going on.
If you do want 800x480 resolution you need a custom mode. See:
https://www.raspberrypi.org/forums/viewtopic.php?t=24679

JamesH65, 2017-05-18:
Closing this issue as questions answered/resolved.

closed 2017-05-18 by JamesH65