raspberrypi/linux #1230: Feature request : Script to set up bit-banged TTY in raspi-config

coreyfro, 2015-12-16:
Hey all,

The RPi has precious few UARTS (and by that, I mean one) and plenty of CPU power. A pretty common first step in working with the RPi is adding an additional TTY, by hook or by crook. It seems to me that an additional option in the raspi-config under advanced for bit-banged TTY would be a good solution for this. While it won't be as nice as an extra UART, it is certainly better than spending money on additional USB FTDI adapters, and with the introduction of the Zero, minimizing USB devices is attractive and important.

It would also be great if there was a "Blessed" way to do this. Easier to support, I am sure.

I ask because I am presently working on an IoT course for complete newbs (hoping to get it simplified down to a 7th grade level) and while I will be spending a good amount of time going over basic Unix Skills, certain tasks, which are common but cumbersome, I'd like to avoid and would suggest remedying in a global fashions since I feel it is a common enough process.

Thank you.

pelwell, 2015-12-17:
1) This is the RPi Linux kernel repo. raspi-config is a Raspbian tool, so such a request should be aimed at the Raspbian maintainers [here](https://www.raspbian.org/RaspbianBugs).
2) What is the interface to this extra UART? Does it appear as a real UART with a /dev/ entry etc? Does a driver exist, and if so where is it?

It is unclear from your message whether you have a tried-and-tested mechanism that you would like to be adopted as a standard, or whether you are hoping that such a mechanism might exist and, if not, that we could create one. If the former, consider distributing a setup script to your newbs - I'd be surprised if this was the only site-specific configuration required. Also, please include all details (including the script) in any further correspondence to the Raspbian devs. If the latter, that sounds like a big ask...

XECDesign, 2015-12-17:
Those raspbian bugs go to plugwash, but raspi-config is our tool.
The issue should be directed here https://github.com/RPi-Distro/raspi-config/issues (with the requested information)

lurch, 2016-01-18:
http://raspberrypi.stackexchange.com/questions/1987/can-raspberry-pi-reliably-bit-bang-a-9600-baud-serial-and-is-there-example-code suggests that even at 9600 baud a bit-banged UART simply isn't possible.
Some interesting alternatives are discussed here https://www.raspberrypi.org/forums/viewtopic.php?f=9&t=2006

popcornmix, 2016-01-18:
Not the right place for this issue.

closed 2016-01-18 by popcornmix