raspberrypi/linux #1234: Official wi-fi dongle extremely unstable in recent kernels.

ali1234, 2015-12-21:
To reproduce:

1. Use a model A+ or Zero, official power supply, official wi-fi dongle connected without hub, and a brand new SanDisk 8GB uSD card that definitely isn't fake and definitely doesn't have any errors.

2. Write a pristine copy of 2015-11-21-raspbian-jessie-lite.img to the uSD card.

3. Boot up, log in over serial, and configure wpa-supplicant.conf to connect to a network.

4. Ping the Pi from another machine on the network.

5. Touch ground anywhere on the Pi, but especially the shielding of the USB port.

Result: endless kernel oops freeze the system.

This is a regression which cannot be reproduced with wheezy on the exact same hardware. Sometimes wth Jessie, merely putting your hand close to the Pi can cause the crash. Using a foot-long USB extension cable can cause the crash without even getting close. The crash happens with both the official wi-fi dongles that I own. It does not happen with any other USB devices, and switching the power supply does not make any difference.

Therefore I suspect that this problem involves an unhandled error on USB which should be silently ignored, but is instead crashing the whole system.

Kernel log:

```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.13+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model A Plus Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x0b800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 48720
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0x12 bcm2708.serial=0xdd2f2419 smsc95xx.macaddr=B8:27:EB:2F:24:19 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 176848K/196608K available (6137K kernel code, 489K rwdata, 1788K rodata, 352K init, 720K bss, 11568K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xcc800000 - 0xff000000   ( 808 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xcc000000   ( 192 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc07c5630   (7926 kB)
[    0.000000]       .init : 0xc07c6000 - 0xc081e000   ( 352 kB)
[    0.000000]       .data : 0xc081e000 - 0xc08987e4   ( 490 kB)
[    0.000000]        .bss : 0xc08987e4 - 0xc094c980   ( 721 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:522
[    0.000000] clocksource stc: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000014] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000056] Switching to timer-based delay loop, resolution 1000ns
[    0.000339] Console: colour dummy device 80x30
[    0.001753] console [tty1] enabled
[    0.001816] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001918] pid_max: default: 32768 minimum: 301
[    0.002275] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002357] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.003441] Initializing cgroup subsys blkio
[    0.003539] Initializing cgroup subsys memory
[    0.003616] Initializing cgroup subsys devices
[    0.003685] Initializing cgroup subsys freezer
[    0.003754] Initializing cgroup subsys net_cls
[    0.003879] CPU: Testing write buffer coherency: ok
[    0.004011] ftrace: allocating 19796 entries in 59 pages
[    0.111328] Setting up static identity map for 0x81c0 - 0x81f8
[    0.114165] devtmpfs: initialized
[... 611 lines cut ...]
[   34.387307] [<c0013ccc>] (die) from [<c0583ccc>] (__do_kernel_fault.part.0+0x64/0x84)
[   34.399457] [<c0583ccc>] (__do_kernel_fault.part.0) from [<c058de70>] (do_page_fault+0x26c/0x3f4)
[   34.412666] [<c058de70>] (do_page_fault) from [<c0009250>] (do_DataAbort+0x44/0xc4)
[   34.424664] [<c0009250>] (do_DataAbort) from [<c058d4dc>] (__dabt_svc+0x3c/0x60)
[   34.436448] Exception stack(0xc90ed9c0 to 0xc90eda08)
[   34.443727] d9c0: cb336040 00000000 00000800 00000000 00000000 cb0c0000 cb336328 c90eda98
[   34.456262] d9e0: cb336040 00000020 c082c138 c90eda14 c90eda18 c90eda08 c003ce08 c0041d10
[   34.468857] da00: 20000093 ffffffff
[   34.474553] [<c058d4dc>] (__dabt_svc) from [<c0041d10>] (kthread_data+0x18/0x20)
[   34.486279] [<c0041d10>] (kthread_data) from [<c003ce08>] (wq_worker_sleeping+0x1c/0xcc)
[   34.498721] [<c003ce08>] (wq_worker_sleeping) from [<c05892ac>] (__schedule+0x480/0x670)
[   34.511239] [<c05892ac>] (__schedule) from [<c05894e8>] (schedule+0x4c/0xa4)
[   34.520606] [<c05894e8>] (schedule) from [<c00267f8>] (do_exit+0x6c4/0xa94)
[   34.529873] [<c00267f8>] (do_exit) from [<c0013ccc>] (die+0x184/0x374)
[   34.538672] [<c0013ccc>] (die) from [<c0583ccc>] (__do_kernel_fault.part.0+0x64/0x84)
[   34.550839] [<c0583ccc>] (__do_kernel_fault.part.0) from [<c058de70>] (do_page_fault+0x26c/0x3f4)
[   34.564073] [<c058de70>] (do_page_fault) from [<c058e0a8>] (do_translation_fault+0xb0/0xb8)
[   34.576800] [<c058e0a8>] (do_translation_fault) from [<c0009250>] (do_DataAbort+0x44/0xc4)
[   34.589475] [<c0009250>] (do_DataAbort) from [<c058d4dc>] (__dabt_svc+0x3c/0x60)
[   34.601221] Exception stack(0xc90edc90 to 0xc90edcd8)
[   34.608506] dc80:                                     0000000d cb38f084 0000015f 00000000
[   34.621026] dca0: 0000014d 0000000a 0000015f c9080160 00000000 00000000 00000064 c90edcf4
[   34.633528] dcc0: cb38f096 c90edcd8 00000003 bf0ec4c0 60000013 ffffffff
```
This OOPS spam continues seemingly forever.

ali1234, 2015-12-21:
This is also reproducible in a rpi-update kernel:
```
Uncompressing Linux... done, booting the kernel.
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.15+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #830 Tue Dec 15 16:58:28 GMT 2015
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] Machine model: Raspberry Pi Model A Plus Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x0b800000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 48720
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708.boardrev=0x12 bcm2708.serial=0xdd2f2419 smsc95xx.macaddr=B8:27:EB:2F:24:19 bcm2708_fb.fbswap=1 bcm2708.disk_led_gpio=47 bcm2708.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Memory: 177232K/196608K available (5762K kernel code, 488K rwdata, 1780K rodata, 352K init, 720K bss, 11184K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xcc800000 - 0xff000000   ( 808 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xcc000000   ( 192 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0765d98   (7544 kB)
[    0.000000]       .init : 0xc0766000 - 0xc07be000   ( 352 kB)
[    0.000000]       .data : 0xc07be000 - 0xc0838218   ( 489 kB)
[    0.000000]        .bss : 0xc0838218 - 0xc08ec380   ( 721 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:522
[    0.000000] clocksource stc: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000014] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000053] Switching to timer-based delay loop, resolution 1000ns
[    0.000336] Console: colour dummy device 80x30
[    0.001272] console [tty1] enabled
[    0.001328] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[    0.001407] pid_max: default: 32768 minimum: 301
[    0.001761] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001827] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.002860] Initializing cgroup subsys blkio
[    0.002947] Initializing cgroup subsys memory
[    0.003011] Initializing cgroup subsys devices
[    0.003069] Initializing cgroup subsys freezer
[    0.003125] Initializing cgroup subsys net_cls
[    0.003234] CPU: Testing write buffer coherency: ok
[    0.003344] ftrace: allocating 19694 entries in 58 pages
[    0.105991] Setting up static identity map for 0x81c0 - 0x81f8
[    0.108136] devtmpfs: initialized
[    0.118888] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.119349] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.121261] pinctrl core: initialized pinctrl subsystem
[    0.122136] NET: Registered protocol family 16
[    0.127639] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[    0.129142] bcm2708.uart_clock = 3000000
[    0.133730] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[    0.133806] hw-breakpoint: maximum watchpoint size is 4 bytes.
[    0.133990] Serial: AMBA PL011 UART driver
[    0.134281] 20201000.uart: ttyAMA0 at MMIO 0x20201000 (irq = 83, base_baud = 0) is a PL011 rev2
[    0.511446] console [ttyAMA0] enabled
[    0.515833] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.560928] bcm2708-dmaengine 20007000.dma: DMA legacy API manager at f2007000, dmachans=0xf35
[    0.569775] bcm2708-dmaengine 20007000.dma: Initialized 7 DMA channels (+ 1 legacy)
[    0.578264] bcm2708-dmaengine 20007000.dma: Load BCM2835 DMA engine driver
[    0.585241] bcm2708-dmaengine 20007000.dma: dma_debug:0
[    0.591379] SCSI subsystem initialized
[    0.595560] usbcore: registered new interface driver usbfs
[    0.601206] usbcore: registered new interface driver hub
[    0.606760] usbcore: registered new device driver usb
[    0.612776] raspberrypi-firmware soc:firmware: Attached to firmware from 2015-12-08 14:38
[... 570 lines cut ...]
[   23.763874] [<c0013a64>] (die) from [<c0528120>] (__do_kernel_fault.part.0+0x64/0x84)
[   23.776340] [<c0528120>] (__do_kernel_fault.part.0) from [<c0530ca4>] (do_page_fault+0x220/0x3b0)
[   23.789839] [<c0530ca4>] (do_page_fault) from [<c0009204>] (do_DataAbort+0x44/0xc4)
[... cut ...]
[   23.960148] [<c0009204>] (do_DataAbort) from [<c053039c>] (__dabt_svc+0x3c/0x60)
[   23.972077] Exception stack(0xc909f9c8 to 0xc909fa10)
[   23.979389] f9c0:                   ca741b40 00000000 00000400 00000000 00000000 cb0c0000
[   23.991976] f9e0: ca741e10 c909faa0 ca741b40 00000020 c07cafc8 c909fa1c c909fa20 c909fa10
[   24.004539] fa00: c003992c c003e2ec 20000093 ffffffff
[   24.011819] [<c053039c>] (__dabt_svc) from [<c003e2ec>] (kthread_data+0x18/0x20)
[   24.023562] [<c003e2ec>] (kthread_data) from [<c003992c>] (wq_worker_sleeping+0x1c/0xcc)
```

popcornmix, 2015-12-21:
It would be useful to be really sure that wheezy doesn't have this problem. It sounds more like a hardware than a software issue.
Testing if this affects B+/Pi2 could be interesting.
If you believe an older kernel didn't have this issue, can you identify the point where it occurred?
You can downgrade to older kernel/firmware with ``sudo rpi-update <hash>``
Use hashes from https://github.com/Hexxeh/rpi-firmware/commits/master (the hash is the end of the URL when you click on a link).
Identify the exact build that introduced the issue and we can look into the changes. Best to test this on the A+ as the Pi0 only has recent support.

ali1234, 2015-12-21:
Bisected with model A+ and a USB extension cable. Good kernels were completely stable and able to download the next kernel for testing. Bad kernels crashed during boot 100% of the time.

```
# first bad commit: [6d25078b7cb361052ae280d552561d7250d33f5b] kernel: Bump to 4.0.5
```

ali1234, 2016-05-17:
When I reported this one I had two access points with the same SSID. I think what happened is one of them was on channel 13, and touching ground was affecting the signal strengths, causing a roam event. It then roamed on to the channel 13 AP and crashed.

Therefore, I think this one is also fixed in 4.4 kernel.

closed 2016-05-17 by ali1234